ros2/rclcpp #1597: Node parameter callback function never called

Omega-Phone, 2021-03-26:
## Feature request

#### Feature description
Hi, I'm trying to add a parameter callback function on a node, but I can't find how to make it work properly. 

If I take the same node code as in https://github.com/ros2/rclcpp/issues/1566 things work as expected and I'm able to called the callback function even with a custom parameter (other than use_sim_tim). But I'm looking to put all this code into my node class, keeping the main() shortest as possible. I've tried with the following code:
```
#include "rclcpp/rclcpp.hpp"
#include "rcl_interfaces/msg/set_parameters_result.hpp"

class TestParams : public rclcpp::Node
{
public:
    rcl_interfaces::msg::SetParametersResult parametersCallback(const std::vector<rclcpp::Parameter> &parameters)
    {
        rcl_interfaces::msg::SetParametersResult result;
        result.successful = true;
        result.reason = "success";
        for (const auto &parameter : parameters)
        {
            if (parameter.get_name() == "my_str" && parameter.get_type() == rclcpp::ParameterType::PARAMETER_STRING)
            {
                my_str_ = parameter.as_string();
                RCLCPP_INFO(this->get_logger(), "Parameter 'my_str' changed: %s", my_str_.c_str());
            }
        }
        return result;
    }

    TestParams() : Node("test_params_rclcpp")
    {
        this->declare_parameter("my_str", "default value");
        const std::vector<rclcpp::Parameter> parameters = this->get_parameters({"my_str"});

        this->add_on_set_parameters_callback(std::bind(&TestParams::parametersCallback, this, parameters));
    }

private:
    std::string my_str_;
};

int main(int argc, char **argv)
{
    rclcpp::init(argc, argv);
    auto node = std::make_shared<TestParams>();
    rclcpp::spin(node);
    rclcpp::shutdown();
    return 0;
}
```
But when I'm setting "my_str" parameters with `ros2 param set /test_params_rclcpp my_str some_value`, the callback function is never called. I'm missing something but I can't find what is it.

Thanks in advance.

fujitatomoya, 2021-03-28:
> But when I'm setting "my_str" parameters with ros2 param set /test_params_rclcpp my_str some_value, the callback function is never called.

because `rclcpp::node_interfaces::OnSetParametersCallbackHandle` is already gone when parameter gets updated.

actually there should be compile error something like this.

```
/root/ros2_ws/colcon_ws/src/ros2/ros2_test_prover/src/rclcpp_1597.cpp:28:107: error: ignoring return value of ‘rclcpp::node_interfaces::OnSetParametersCallbackHandle::SharedPtr rclcpp::Node::add_on_set_parameters_callback(rclcpp::Node::OnParametersSetCallbackType)’, declared with attribute warn_unused_result [-Werror=unused-result]
   28 |         this->add_on_set_parameters_callback(std::bind(&TestParams::parametersCallback, this, parameters));
```

you can take a look at my sample based on yours, it works okay.

https://github.com/fujitatomoya/ros2_test_prover/blob/0e666c9ab7e60e45295b77e57164dfb39f79c760/src/rclcpp_1597.cpp#L27

fujitatomoya, 2021-03-28:
since this is not bug for ros2, i am just go ahead to close. feel free to re-open if you have anything.

closed 2021-03-28 by fujitatomoya