ros2/rclcpp #1599: node crashes while received a cancel request if the state is SUCCEEDED

KavenYau, 2021-03-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries or source 
- Version or commit hash:
  - 47f21dab3d2c01710c8ea04784e970eec4a92e15
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  - rclcpp (rclcpp_action)

#### Steps to reproduce issue

- I commented the code `std::lock_guard<std::recursive_mutex> lock(update_mutex_);` in this file https://github.com/ros-planning/navigation2/blob/main/nav2_util/include/nav2_util/simple_action_server.hpp#L125, because it would cause deadlock,  although this issue #1518 claims that it is fixed. ( due to there are two threads tring to acquire the the mutex `update_mutex_`, https://github.com/ros-planning/navigation2/blob/main/nav2_util/include/nav2_util/simple_action_server.hpp#L384 )
```
  rclcpp_action::CancelResponse handle_cancel(
    const std::shared_ptr<rclcpp_action::ServerGoalHandle<ActionT>>/*handle*/)
  {
    // std::lock_guard<std::recursive_mutex> lock(update_mutex_);
    debug_msg("Received request for goal cancellation");
    return rclcpp_action::CancelResponse::ACCEPT;
  }
```
- then run navigation2 as usual, send goal and cancel goal
- then sometimes will crush and print this infomation
```
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  goal_handle attempted invalid transition from state SUCCEEDED with event CANCEL_GOAL, at /tmp/binarydeb/ros-foxy-rcl-action-1.1.10/src/rcl_action/goal_handle.c:95
```

So far as I am aware, https://github.com/ros2/rcl/blob/foxy/rcl_action/src/rcl_action/goal_state_machine.c#L82 shows that the state SUCCEEDED can not be transformed to any state.  https://github.com/ros2/rclcpp/blob/foxy/rclcpp_action/src/server_goal_handle.cpp#L81 the `_cancel_goal()` function will be called internally only. so if the `handle_cancel_` user callback return `rclcpp_action::CancelResponse::ACCEPT` https://github.com/ros2/rclcpp/blob/foxy/rclcpp_action/include/rclcpp_action/server.hpp#L363, and it crushes. 
I can't catch the exception because it occurs internally.

So I think this exception should be catched internally, or do not throw exception, just handle it. Make sure it will not crush.

Of course, I think it can be avoid in the action server implementation too. Because the action server should know it is going to publish succeeded state, and the  the `handle_cancel_` user callback return `rclcpp_action::CancelResponse::REJECT`, so it would not to call `goal_handle->_cancel_goal();`.

But the second solution demand that I have to know more about the implementation of rclcpp and rcl.

So do you think it's the issue of rclcpp?
Or do you have any good suggestions?

fujitatomoya, 2021-03-28:
@KavenYau thanks for opening issue.

i might be mistaken but, i am not really sure what exactly the problem is.

> I can't catch the exception because it occurs internally.

why? could you elaborate here? exception eventually goes to the application, and the process gets crashed since application does not catch it.

> So I think this exception should be catched internally, or do not throw exception, just handle it. Make sure it will not crush.

i do not think this could be a justification...exception is meant to be sent to user application, because this needs to be notify the user application. (with exception, user WILL get that exception.)

> Of course, I think it can be avoid in the action server implementation too. Because the action server should know it is going to publish succeeded state, and the the handle_cancel_ user callback [...]

i believe that the way it is. application is the one to control this, so that application can decide what should be cancelled or rejected. i mean in other words, i do not know how system knows what should be canceled or not canceled?

> So do you think it's the issue of rclcpp?

so far, i would say no.

> Or do you have any good suggestions?

how about checking the goal state to see if it is succeeded or not in `handle_cancel`? (more like checking that goal can be cancelled or not, the return either `REJECT` or `ACCEPT`)

KavenYau, 2021-03-30:
@fujitatomoya thanks for your reply.

Here is the backtrace. https://gist.github.com/KavenYau/c32ef04f62d9894ed1edeab2ef65b9e2

> why? could you elaborate here? exception eventually goes to the application, and the process gets crashed since application does not catch it.

I think I can catch the exception here(https://github.com/ros-planning/navigation2/blob/41996f409fcd7b005cf779e0376273cee3de6d76/nav2_util/src/node_thread.cpp#L29).
Although the crashing problem is solved, it is alreadly outside of the node, like this code. And the node will be removed.
```c++
 thread_ = std::make_unique<std::thread>(
    [&]()
    {
      executor_.add_node(node_);
      try{
        executor_.spin();
      } catch (...) {
      }
      executor_.remove_node(node_);
    });
```
The node is created here (https://github.com/ros-planning/navigation2/blob/41996f409fcd7b005cf779e0376273cee3de6d76/nav2_util/src/lifecycle_node.cpp#L56),
and action server is created here (https://github.com/ros-planning/navigation2/blob/main/nav2_util/include/nav2_util/simple_action_server.hpp#L85). 
Is there a better place to catch the exception?

> i mean in other words, i do not know how system knows what should be canceled or not canceled?

I think you already know, because the transition from state SUCCEEDED to CANCELED is illegal.

> how about checking the goal state to see if it is succeeded or not in handle_cancel? (more like checking that goal can be cancelled or not, the return either REJECT or ACCEPT)

Yes, it should be controlled by user application too.

By the way, I found a similar issue #1569.

fujitatomoya, 2021-03-30:
thanks for sharing additional information.

> I think you already know, because the transition from state SUCCEEDED to CANCELED is illegal.

just to confirm, cz i am confused a bit.
so you saying that application callback returns `CancelResponse::ACCEPT` but system processes it with `CancelResponse::REJECTED` if exception happens? in application perspective, it is really strange behavior for me...if i am not mistaken.

i think current behavior is better. application is responsible to tell what is cancelled or rejected.

KavenYau, 2021-04-06:
> so you saying that application callback returns CancelResponse::ACCEPT but system processes it with CancelResponse::REJECTED if exception happens?

Emm... 
In this perspective. it's really strange. I'm sorry about misleading you.

> i think current behavior is better. application is responsible to tell what is cancelled or rejected.

In some cases, it's difficult to control it.

1. If now it is not publishing succeeded state, but handling cancel callback. so returns `ACCEPT`.
2. Once the cancel callback has returned. `ServerGoalHandleBase::_succeed` is called, and the goal_state has transformed to `SUCCEEDED`. Then waiting to erase the `goal_handle`,
3. Due to cancel callback returns `ACCEPT`, `ServerGoalHandleBase::_cancel_goal` is called. 
4. Exception throws.

It seems that  some mutex (or other synchronization mechanisms) are lacking between goal handle methods and goal callbacks

fujitatomoya, 2021-04-22:
@KavenYau 

see https://github.com/ros2/rclcpp/pull/1635#discussion_r618062059, now i see that not throwing exception is the way to do according to the action design. (my bad, sorry about that 😢 ) anyway, we could keep this open for discussion.

KavenYau, 2021-04-23:
Thanks!! :smile: 

Just to be sure, is there someone handling this issue now? If not, I could make a PR for it.

fujitatomoya, 2021-04-23:
> Just to be sure, is there someone handling this issue now? If not, I could make a PR for it.

i do not think so. you can just go ahead for it 👍

KavenYau, 2021-04-30:
With both #1635 and #1641 were merged, this issue could be closed

fujitatomoya, 2021-05-01:
@KavenYau thanks for your patience and contribution!

closed 2021-05-01 by fujitatomoya