ros2/rclcpp #1601: What is "https://github.com/ros2/rclcpp"

BriceRenaudeau, 2021-03-29:
## Bug report

**Required Info:**

- Operating System:
Ubuntu
- Installation type:
Binary
- Version or commit hash:
master
- DDS implementation:
Cyclone
- Client library (if applicable):
lifecycle_node_interface

#### Steps to reproduce the issue
When killing my lifecycle node "assisted_teleop" this message appears:

```
[assisted_teleop-28] oh yeah destructuro
[assisted_teleop-28] oh yeah end destructuro
[assisted_teleop-28] oh yeah end destructuro222
[assisted_teleop-28] oh yeah destructuro
[assisted_teleop-28] oh yeah end destructuro
[assisted_teleop-28] oh yeah end destructuro222
```

#### Expected behavior

#### Actual behavior

#### Additional information

christophebedard, 2021-03-29:
It was introduced by https://github.com/ros2/rclcpp/pull/1507. Seems like it was just left behind and no one caught it.

closed 2021-03-29 by ivanpauno

Karsten1987, 2021-03-29:
lol. Now y'all know my advanced debugging techniques ;-)

Thanks @ivanpauno for taking care of it!

fujitatomoya, 2021-03-29:
lol 😆