ros2/rclcpp #1630: rclcpp::lifecycle cannot be used on windows

flynneva, 2021-04-14:
## Bug report

while trying to see if we could build the [`ros-drivers/transport_drivers`](https://github.com/ros-drivers/transport_drivers) packages on windows I ran into this problem which seems to be a bug with `rclcpp::lifecycles`.

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - [binaries](https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html)
- Version or commit hash: 
  - not sure how to check this on windows but LMK if you know how
- DDS implementation:
  - default (FastDDS i think is what it is now)
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
- install ros2 foxy binaries on windows 10 using instructions linked above
- clone my  `build-on-windows` branch from [`flynneva/transport_drivers`](https://github.com/flynneva/transport_drivers/tree/build-on-windows) to your workspace
- you will also need to clone the [`udp_msgs` dependency](https://github.com/flynneva/udp_msgs.git) to your workspace
- run `colcon build` and you should get the same error
#### Expected behavior
rclcpp::lifecycle does not throw an error on windows

#### Actual behavior
rclcpp::lifecycle causes the build to fail on windows

#### Additional information
I'm not 100% sure this is an issue with the lifecycle package, so feel free to tell me its not if you know what is causing it 😄

flynneva, 2021-04-14:
[stdout.log](https://github.com/ros2/rclcpp/files/6312361/stdout.log)

here is the `stdout.log` from the build which might help with debugging

clalancette, 2021-04-14:
For what it is worth, there are definitely demos and examples that use lifecycle nodes on Windows in the Foxy core.  One example is https://github.com/ros2/demos/blob/master/lifecycle/src/lifecycle_talker.cpp .  So while there might be a problem, it doesn't seem like it is completely broken.

Given the errors in the log there, this very much feels like some kind of missing command-line arguments on Windows.  It would help to get detailed information about what the compile command looks like.

flynneva, 2021-04-14:
i was just using the simple `colcon build` commmand. for compiling on windows does there need to be specific arguments passed in? @clalancette

clalancette, 2021-04-14:
> i was just using the simple `colcon build` commmand. for compiling on windows does there need to be specific arguments passed in? @clalancette

Not usually, no.  Sometimes there needs to be something in the `CMakeLists.txt`, depending on what features you are using, but that isn't really that common either.  Getting the verbose compile-line output will be helpful here.

flynneva, 2021-04-14:
`colcon build`:
[command.log](https://github.com/ros2/rclcpp/files/6312805/command.log)

`colcon build --symlink-install`:
[command.log](https://github.com/ros2/rclcpp/files/6312814/command.log)

christophebedard, 2021-04-14:
> i was just using the simple colcon build commmand. for compiling on windows does there need to be specific arguments passed in?

just a random shot in the dark, but did you install Visual Studio (2019)?

flynneva, 2021-04-14:
@christophebedard yes. and i can confirm I can compile on windows other ROS2 packages that are not lifecycle nodes.

flynneva, 2021-04-14:
i think im going to go ahead and try to install ROS2 myself by downloading it rather than using the "choco install" approach.  thanks for the tips, ill go ahead and close this.

closed 2021-04-14 by flynneva

flynneva, 2021-04-16:
in case anyone from google ends up here.....I figured out my issue.

I had a bunch of typos in my code: `cylce` instead of `cycle`, mis-spelled class methods, etc.

In short: double check you've implemented all the lifecycle methods and spelled everything correctly....that should fix your issue.