ros2/rclcpp #1633: cppcheck test times out on OSX

sloretz, 2021-04-15:
## Bug report

**Required Info:**

- Operating System:
  -  OSX
- Installation type:
  -  source
- Version or commit hash:
  -  ec70642c553cce307e099af7802ae7cbe509bf1b
- DDS implementation:
  -  N/A
- Client library (if applicable):
  -  rclcpp

#### Steps to reproduce issue

Run this job https://ci.ros2.org/job/nightly_osx_release/2029/

#### Expected behavior

#### Actual behavior

```
    Start 2: cppcheck

2: Test command: /Users/osrf/jenkins-agent/workspace/nightly_osx_release/venv/bin/python3 "-u" "/Users/osrf/jenkins-agent/workspace/nightly_osx_release/ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/Users/osrf/jenkins-agent/workspace/nightly_osx_release/ws/build/rclcpp/test_results/rclcpp/cppcheck.xunit.xml" "--package-name" "rclcpp" "--output-file" "/Users/osrf/jenkins-agent/workspace/nightly_osx_release/ws/build/rclcpp/ament_cppcheck/cppcheck.txt" "--command" "/Users/osrf/jenkins-agent/workspace/nightly_osx_release/ws/install/ament_cppcheck/bin/ament_cppcheck" "--xunit-file" "/Users/osrf/jenkins-agent/workspace/nightly_osx_release/ws/build/rclcpp/test_results/rclcpp/cppcheck.xunit.xml" "--include_dirs" "/Users/osrf/jenkins-agent/workspace/nightly_osx_release/ws/src/ros2/rclcpp/rclcpp/include"
2: Test timeout computed to be: 300
2: -- run_test.py: invoking following command in '/Users/osrf/jenkins-agent/workspace/nightly_osx_release/ws/src/ros2/rclcpp/rclcpp':
2:  - /Users/osrf/jenkins-agent/workspace/nightly_osx_release/ws/install/ament_cppcheck/bin/ament_cppcheck --xunit-file /Users/osrf/jenkins-agent/workspace/nightly_osx_release/ws/build/rclcpp/test_results/rclcpp/cppcheck.xunit.xml --include_dirs /Users/osrf/jenkins-agent/workspace/nightly_osx_release/ws/src/ros2/rclcpp/rclcpp/include
1/1 Test #2: cppcheck .........................***Timeout 300.13 sec
```

#### Additional information

It was pretty close to failing the previous job

```
1/1 Test #2: cppcheck .........................   Passed  295.42 sec
```

Maybe the test timeout needs to be extended?

wjwwood, 2021-04-15:
> Maybe the test timeout needs to be extended?

I think so. This isn't the first time we've had to raise it. `cppcheck` is just slow.

ivanpauno, 2021-04-15:
We could make it faster though, the main reason it's so slow is that we're not passing preprocessor definitions (see https://github.com/ament/ament_lint/issues/270).

PR #1634 (2021-04-21): Increase cppcheck timeout to 500s
Resolves #1633 by increasing the timeout
closed 2021-04-21 by sloretz