ros2/rclcpp #1652: rclcpp IPC does not raise exception with TRANSIENT_LOCAL durability

alsora, 2021-04-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Source
- Version or commit hash:
  - ROS 2 master and Foxy
- DDS implementation:
  - Cyclone DDS and Fast-DDS (but irrelevant)
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

This code provides a small variation to the MinimalPublisher ros2 example.

Differences are:
 - we enable IPC optimization as part of the node options
 - we change the publisher QOS to use transient local

```
#include <chrono>
#include <memory>
#include <string>

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"

using namespace std::chrono_literals;

class MinimalPublisher : public rclcpp::Node
{
public:
  MinimalPublisher()
  : Node("minimal_publisher", "", rclcpp::NodeOptions().use_intra_process_comms(true)), count_(0)
  {
    publisher_ = this->create_publisher<std_msgs::msg::String>("topic", rmw_qos_durability_policy_t::RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL);
    auto timer_callback =
      [this]() -> void {
        auto message = std_msgs::msg::String();
        message.data = "Hello, world! " + std::to_string(this->count_++);
        RCLCPP_INFO(this->get_logger(), "Publishing: '%s'", message.data.c_str());
        this->publisher_->publish(message);
      };
    timer_ = this->create_wall_timer(500ms, timer_callback);
  }

private:
  rclcpp::TimerBase::SharedPtr timer_;
  rclcpp::Publisher<std_msgs::msg::String>::SharedPtr publisher_;
  size_t count_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<MinimalPublisher>());
  rclcpp::shutdown();
  return 0;
}

```

#### Expected behavior

Running the application should raise the following exception
```
terminate called after throwing an instance of 'std::invalid_argument'
  what():  intraprocess communication allowed only with volatile durability
Aborted (core dumped)
```

because IPC does not support transient_local durability

#### Actual behavior

Creating the publisher as 
```
publisher_ = this->create_publisher<std_msgs::msg::String>("topic", rmw_qos_durability_policy_t::RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL);
```
does not cause any error.

On the other hand, the error is correctly raised if we create the publisher as 
```
    publisher_ = this->create_publisher<std_msgs::msg::String>("topic", rclcpp::SensorDataQoS().durability(rmw_qos_durability_policy_t::RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL));
```

clalancette, 2021-05-01:
> publisher_ = this->create_publisher<std_msgs::msg::String>("topic", rmw_qos_durability_policy_t::RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL);

I think what is happening here is that the compiler is assuming that `rmw_qos_durability_policy_t::RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL` is an integer (which it is), and thus it is constructing `rclcpp::QoS` with this as the *depth*, not the durability policy.  You can confirm that by firing up your example and then run `ros2 topic info -v` on the topic; I suspect it will be volatile durability, not transient_local.

If that is the case, I'm not sure there is much we can do about it.

wjwwood, 2021-05-01:
Exactly what @clalancette said, you're using an rmw type on an interface that expects rclcpp types or a `size_t`.

wjwwood, 2021-05-01:
I think we should close this as won't fix, simply because I cannot think of a way to catch this specific issue and warn the user. Removing the signature that accepts a `size_t` isn't really an option, given how widely it is used.

If you guys disagree we can continue the discussion here and I can reopen it.

closed 2021-05-01 by wjwwood