ros2/rclcpp #1655: rebind is deprecated in C++17 and removed in C++20

SquirrelCZE, 2021-05-02:
https://github.com/ros2/rclcpp/blob/d051b8aa20fa263b1480945aebd1a8e5df315b67/rclcpp/include/rclcpp/allocator/allocator_deleter.hpp#L98

Because of this line, the package is not usable with C++20 code as `template bind` is deprecated now. (Similar issued solved in message generation for C++ in PR https://github.com/ros2/rosidl/pull/564 )

I kind of want to write PR for this for rclcpp, but I am not sure I understand what is going on here.
Why is it  typename std::allocator<void>::template rebind<T>::other in ther first place? Seems like using `std::allocator< T >` is what is actually wanted here (which corresponds to actual implementation of std:: I have - rebind is implemented just as a way to get std::allocator<T>)

I believe I am missing something here? If not I will try to make PR assap

John

devrite, 2021-05-06:
To me it looks also the  same as just putting in `std::allocator<T>`. Rebind was for things when you where given the allocator as template arg already and you would want to use it for another type. But since the original allocator is hard coded here rebind will just result to `std::allocator<T>`  if I am not missing anything either.

wjwwood, 2021-05-07:
Good to know, thanks for pointing it out.

> Why is it `typename std::allocator::template rebind::other` in ther first place? Seems like using `std::allocator< T >` is what is actually wanted here (which corresponds to actual implementation of std:: I have - rebind is implemented just as a way to get std::allocator)

Seems reasonable to me.

pettni, 2021-05-14:
I also ran into this issue and would like to see a fix.

For reference it was fixed in ```rosidl``` here by switching to ```allocator_traits<T>::rebind_alloc```: https://github.com/ros2/rosidl/pull/564. When I grep in my ```/opt/ros/foxy/include``` this is the only other instance of ```rebind```, so it would be great to get this fix into galactic to make it C++20-compatible.

@SquirrelCZE Want to do the PR? If not I'd be happy to.

I agree however that putting just ```std::allocator<T>``` would have the same result here.

SquirrelCZE, 2021-05-14:
@pettni This is still in my TODO list, but I have a ton of projects to mark just now, so please feel free to do it :)

wjwwood, 2021-05-27:
Closed with https://github.com/ros2/rclcpp/pull/1678, thanks everyone!

closed 2021-06-04 by sloretz