ros2/rclcpp #1656: Node::get_effective_namespace return invaid namespace if sub-namespace is empty

devrite, 2021-05-05:
Hi everyone

I tested this on foxy but as you can see from: https://github.com/ros2/rclcpp/blob/1fc2d58799a6d4530522be5c236c70be53455e60/rclcpp/src/rclcpp/node.cpp#L81-L95

it returns an namespace with a slash at the end in both cases.  `rclcpp::expand_topic_or_service_name` will then throw an exception it represents an invalid namespace format (e.g., '/' at the end are not allowed).

Is this still the case? Or rclcpp behaving wrong in this case. But since we usually specify namespaces without slashes almost everywhere I can recall using it I think even if it were permitted we should return the namespace without the slash.

BR, Markus

fujitatomoya, 2021-05-06:
i think that question is `subordinate node can be created w/o subordinate namespace`? i guess no...

that's why it generates exception `rclcpp::exceptions::InvalidNamespaceError` via `create_sub_node("")` as following.

```
terminate called after throwing an instance of 'rclcpp::exceptions::InvalidNamespaceError'
  what():  Invalid namespace: namespace must not end with a '/', unless only a '/':
  '/ns/'
      ^

Thread 1 "rclcpp_1656" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff725e859 in __GI_abort () at abort.c:79
#2  0x00007ffff74e4951 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff74f047c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff74f04e7 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff74f0799 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7d23bd9 in rclcpp::Node::Node (this=0x5555555e6c70, other=..., sub_namespace="")
    at /root/ros2_ws/colcon_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/node.cpp:246
#7  0x00007ffff7d25334 in rclcpp::Node::create_sub_node (this=0x55555557f380, sub_namespace="")
    at /root/ros2_ws/colcon_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/node.cpp:600
#8  0x000055555555b709 in main (argc=1, argv=0x7fffffff1638)
    at /root/ros2_ws/colcon_ws/src/ros2/ros2_test_prover/prover_rclcpp/src/rclcpp_1656.cpp:8
```

i guess current behavior makes sense to me. but it would be nice to add test on this case.

CC: @ivanpauno @wjwwood what do you think :question:

devrite, 2021-05-06:
Yep I know,  but this https://github.com/ros2/rclcpp/blob/0659d829cec5a39c6ad861d597b4aa9aee224a25/rclcpp/include/rclcpp/node.hpp#L1204-L1226

reads to me as the effective namespace used for creating any entities (e.g., if it is a sub_node with the sub_namespace, if it is not just its original namespace), while get_namespace will only give you the original namespace should you be a sub_node (usefully if sub_nodes and nodes communicate on some entities with one another). Especially since it is part of any node's interface it should not return invalid namespace, but rather throw an exception itself it is not intended to be called on a normal node. In this case, to avoid boiler plate code in user code, we either adapt the function or provide a new interface for this.

Hence for me calling `get_effectinve_namespace` on a standard node should return a well formed namespace (i.e., the original ns for normal nodes). Which as of now it does not. If you try to create a sub_node with no sub_namespace it already throws an exception and does not allow creating such a node.

devrite, 2021-05-06:
Also for a reference you may refer to https://github.com/ros2/rclcpp/blob/0659d829cec5a39c6ad861d597b4aa9aee224a25/rclcpp/src/rclcpp/node.cpp#L193-L196

Which is part of https://github.com/ros2/rclcpp/blob/0659d829cec5a39c6ad861d597b4aa9aee224a25/rclcpp/src/rclcpp/node.cpp#L143-L146

the normal nodes interface and creates an invalid effective_namespace_.

fujitatomoya, 2021-05-06:
@devrite thanks for the explanation, i see that by itself as effective namespace, it should not include `/` slash at the end.

devrite, 2021-05-06:
@fujitatomoya  Your welcome. Took me quite some time to identify the issue, as I was originally faced by it when trying to patch `image_transport`. And thought I was using the interface wrong (one of the tests failed there, because of it.

I already linked a PR to it, which should avoid this issue. But your comment remined me to check if an empty sub_node would now be allowed due to this change. I'll go check it.

devrite, 2021-05-06:
@fujitatomoya Could you add a call to `get_effective_namespace()` to both node and sub node in your prove example? This would already satisfy the requesst of @wjwwood for a regression test I think.

fujitatomoya, 2021-05-07:
### Reproducible Sample

https://github.com/fujitatomoya/ros2_test_prover/blob/master/prover_rclcpp/src/rclcpp_1656.cpp

```bash
root@134f29e8f25f:~/ros2_ws/colcon_ws# ros2 run prover_rclcpp rclcpp_1656
[INFO] [1620346207.926351517] [ns.my_node]: Parent Node's effective name: /ns/
[INFO] [1620346207.926427278] [ns.my_node]: Child Node's effective name: /ns/sub_ns
```

`Parent Node's` cannot have effective name.

fujitatomoya, 2021-05-07:
@devrite i think you can add above test case around

https://github.com/ros2/rclcpp/blob/0659d829cec5a39c6ad861d597b4aa9aee224a25/rclcpp/test/rclcpp/test_node.cpp#L164

devrite, 2021-05-07:
Oh thanks, was not sure where I would add it.

devrite, 2021-06-22:
@wjwwood Hi if there is no conclusion yet, I would just change MR to use the old behaviour (no empty sub-namespace) but fix the problem for now. If we want to implement the new behaviour later (empty sub-ns allowed), we can open a new issue for that?

commit 3cddb4ed (2021-08-20): Fix returning invalid namespace if sub_namespace is empty (#1658)

* Create valid effective namespace when sub-namespace is empty

Fix #1656.

* Add regression test for effective namespace and empty sub-namespace

Adds regression test for #1656.
commit a58f8c1d (2021-11-11): Fix returning invalid namespace if sub_namespace is empty (#1658) (#1811)

* Create valid effective namespace when sub-namespace is empty

Fix #1656.

* Add regression test for effective namespace and empty sub-namespace

Adds regression test for #1656.

(cherry picked from commit 3cddb4edab317758dc8a8cac94b90794641c7488)
closed 2021-08-04 by fujitatomoya