ros2/rclcpp #1664: Integrate Type Adaptation into Intra-process Manager

audrow, 2021-05-11:
Currently, when using a custom type, it is always converted to a ROS message type, which requires a few copy operations. These copy operations should be able to be avoided through intra-process communication.

After this is implemented, the following tutorial should work for adapted types: https://docs.ros.org/en/rolling/Tutorials/Intra-Process-Communication.html

Also, when this is done we should update the last section of the tutorial: https://docs.ros.org/en/rolling/Tutorials/Intra-Process-Communication.html#type-masquerading

<hr>

I've put this issue on the H-turtle board for consideration.

clalancette, 2022-03-28:
This was already done, so closing this out.

ottojo, 2022-03-29:
Hi! The second link in the issue description (https://docs.ros.org/en/rolling/Tutorials/Intra-Process-Communication.html#type-masquerading) does not seem to link to an existing section in the docs, is there documentation somewhere already on how intra-process communication interacts with the new type-adapters in humble?

clalancette, 2022-03-29:
> is there documentation somewhere already on how intra-process communication interacts with the new type-adapters in humble?

I don't think so at the moment.  This is on our wish-list to do for the Humble release.  If you'd like to open an issue over at https://github.com/ros2/ros2_documentation, that would help us track it.

ottojo, 2022-03-29:
> I don't think so at the moment. This is on our wish-list to do for the Humble release. If you'd like to open an issue over at https://github.com/ros2/ros2_documentation, that would help us track it.

I created an issue here: https://github.com/ros2/ros2_documentation/issues/2377