ros2/rclcpp #1698: rclcpp_components_register_nodes() fails if not in the project's root CMakeLists.txt

stonier, 2021-06-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04, Dashing
- Installation type:
  - Binaries and source
- Version or commit hash:
  - Released binary and branch
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

* Create a nested `CMakeLists.txt` in the [demos/composition](https://github.com/ros2/demos/tree/dashing/composition) package
* Move the `talker_component` logic into the nested CMakeLists.txt (be sure to include the registering macro)
* Build and install

#### Expected behavior

The talker component will be present in `share/ament_index/resource_index/rclcpp_components/composition`

#### Actual behavior

The talker component is missing.

iuhilnehc-ynos, 2021-06-22:
> * Create a nested `CMakeLists.txt` in the [demos/composition](https://github.com/ros2/demos/tree/dashing/composition) package
> * Move the `talker_component` logic into the nested CMakeLists.txt (be sure to include the registering macro)

Hi, @stonier 

I suppose you want to use the subdirectory to organize the source code. I think you could refer to this [link](https://answers.ros.org/question/361289/ros2-components-registration-from-subdirectory-cmakelists-file/?answer=361439#post-id-361439) firstly.
(FYI: I have tried the workaround (the method might be a little different from the above link) for the [demo](https://github.com/ros2/demos/commit/e9d69ddcd29732c483386abbcb3856d304a71a11) and it seems working.)

wjwwood, 2021-06-22:
I think it should be technically possible to support what @stonier is doing, and probably a good idea to do so as well. @sloretz's workaround will work, but is somewhat fragile because it depends on a cmake variable name that we otherwise don't expose and could be changed in the future. I don't have time to work on this at the moment, but we could consider changing how we set/get that variable to work around the scoping, or maybe we could add a macro that must be called in the child scope to make this work (a less-fragile form of the `PARENT_SCOPE` workaround).

iuhilnehc-ynos, 2021-06-25:
> we could add a macro that must be called in the child scope to make this work

Yeah, this could be a method.
But I think users want to use `rclcpp_components_register_nodes` in subdirectories in the same way as the root directory.

Patches that I am not sure if it's the right way.
<details><summary> rclcpp </summary>
<p>

```shell

diff --git a/rclcpp_components/cmake/rclcpp_components_register_nodes.cmake b/rclcpp_components/cmake/rclcpp_components_register_nodes.cmake
index e80550a3..74045f99 100644
--- a/rclcpp_components/cmake/rclcpp_components_register_nodes.cmake
+++ b/rclcpp_components/cmake/rclcpp_components_register_nodes.cmake
@@ -67,6 +67,15 @@ macro(rclcpp_components_register_nodes target)
         "${_RCLCPP_COMPONENTS_${resource_index}__NODES}${_arg};${_path}/$<TARGET_FILE_NAME:${target}>\n")
       list(APPEND _RCLCPP_COMPONENTS_PACKAGE_RESOURCE_INDICES ${resource_index})
     endforeach()
-  endif()
-endmacro()
 
+    # Check if current scope is subdirectory or not
+    if(NOT CMAKE_SOURCE_DIR STREQUAL CMAKE_CURRENT_SOURCE_DIR)
+      set(_RCLCPP_COMPONENTS_${resource_index}__NODES
+        "${_RCLCPP_COMPONENTS_${resource_index}__NODES}"
+        PARENT_SCOPE)
+      set(_RCLCPP_COMPONENTS_PACKAGE_RESOURCE_INDICES
+        ${_RCLCPP_COMPONENTS_PACKAGE_RESOURCE_INDICES}
+        PARENT_SCOPE)
+    endif()
+  endif()
+endmacro()
\ No newline at end of file
diff --git a/rclcpp_components/rclcpp_components-extras.cmake.in b/rclcpp_components/rclcpp_components-extras.cmake.in
index 45a4e5ac..5dc33d9b 100644
--- a/rclcpp_components/rclcpp_components-extras.cmake.in
+++ b/rclcpp_components/rclcpp_components-extras.cmake.in
@@ -17,7 +17,11 @@
 # register ament_package() hook for node plugins once.
 macro(_rclcpp_components_register_package_hook)
   if(NOT DEFINED _RCLCPP_COMPONENTS_PACKAGE_HOOK_REGISTERED)
-    set(_RCLCPP_COMPONENTS_PACKAGE_HOOK_REGISTERED TRUE)
+    if(CMAKE_SOURCE_DIR STREQUAL CMAKE_CURRENT_SOURCE_DIR)
+      set(_RCLCPP_COMPONENTS_PACKAGE_HOOK_REGISTERED TRUE)
+    else()
+      set(_RCLCPP_COMPONENTS_PACKAGE_HOOK_REGISTERED TRUE PARENT_SCOPE)
+    endif()
 
     find_package(ament_cmake_core QUIET REQUIRED)
     ament_register_extension("ament_package" "rclcpp_components"
```

</p>
</details>

<details><summary> ament_cmake_core </summary>
<p>

```shell
diff --git a/ament_cmake_core/cmake/core/ament_register_extension.cmake b/ament_cmake_core/cmake/core/ament_register_extension.cmake
index 45effb2..b4118a4 100644
--- a/ament_cmake_core/cmake/core/ament_register_extension.cmake
+++ b/ament_cmake_core/cmake/core/ament_register_extension.cmake
@@ -30,4 +30,10 @@
 macro(ament_register_extension extension_point package_name cmake_filename)
   list(APPEND AMENT_EXTENSIONS_${extension_point}
     "${package_name}:${cmake_filename}")
+
+  if(NOT CMAKE_SOURCE_DIR STREQUAL CMAKE_CURRENT_SOURCE_DIR)
+    set(AMENT_EXTENSIONS_${extension_point}
+      ${AMENT_EXTENSIONS_${extension_point}}
+      PARENT_SCOPE)
+  endif()
 endmacro()

```

</p>
</details>

HighPriest, 2021-12-17:
I think I am having the same issue, where my component is closely based on the [DEMO](https://github.com/ros2/demos/commit/e9d69ddcd29732c483386abbcb3856d304a71a11), but the only thing which is returned by the `ros2 pkg executables` command and is linked to the _project_  is a `.so` file with the name of the library created in CMakeLists.txt.

In my case I have two levels of `add_subdirectory`. One for the `project` level, which adds the `src` level, which adds all the `components`.