ros2/rclcpp #1701: Prefer lambda to std::bind

iuhilnehc-ynos, 2021-06-25:
## Feature request

#### Feature description

Replace all `std::bind` with `lambda`. Is it worth doing it for now?

https://github.com/isocpp/CppCoreGuidelines/issues/1796

#### Implementation considerations

e.g.

Replace https://github.com/ros2/rclcpp/blob/e9e398d2d49de2ecc3148d425f8659d9db74918c/rclcpp/src/rclcpp/time_source.cpp#L130-L133 with
```cpp
  parameter_subscription_ = rclcpp::AsyncParametersClient::on_parameter_event(
    node_topics_,
    [this](std::shared_ptr<const rcl_interfaces::msg::ParameterEvent> event) {
        return on_parameter_event(event);
      }
    );
```

wjwwood, 2021-06-25:
I'm not opposed to it, but I'm also not sure if it is a really important thing to do either.

A few somewhat unimportant things to consider:

- using `std::bind` in `rclcpp` gives us implicit coverage on that use case, which is nice but not required as we have demos that cover that case too
- in the example you gave, `std::bind` is slightly nicer because we don't need to include the type of `event`, and don't think `auto` would work in this case due to how the template deduction works on that

But I don't think those points push me one way or the other that much.

oysstu, 2025-12-16:
When rclcpp targets C++20, std::bind_front gives the best of both worlds. The function traits in rclcpp needs to be updated to handle this for various compilers though.