ros2/rclcpp #1703: Intra-process services and actions

alsora, 2021-06-30:
## Feature request

#### Feature description

Currently the intra-process manager only handles publishers and subscriptions.
It should be extended to all the communication modes (services and actions).

#### Implementation considerations

I wonder if the same approach used for publishers/subscriptions (e.g. create intra-process client and intra-process service that derive from waitable) is enough or if a different solution should be adopted.

wjwwood, 2021-07-01:
Actions will probably come for "free" if the intra-process option is turned on, since actions are built entirely on topics and services.

For now, building services out like subscriptions have been done is fine. We might want to change the architecture in the future, but better to move forward now as-is.

irobot-ros, 2021-07-01:
Ok good. We will have bandwidth to work on this and other Intra-Process improvements after the RMW listener APIs are merged, unless someone else pick this up.