ros2/rclcpp #1704: Shutdown cleanly on SIGTERM

emersonknapp, 2021-06-30:
## Feature request

#### Feature description

The `rclcpp` Context shuts down gracefully on a SIGINT, which is generally received via a Ctrl+C interaction. However, another reasonable way on Linux to try to shut down processes is via a SIGTERM - according to [GNU Documentation](https://www.gnu.org/software/libc/manual/html_node/Termination-Signals.html):

> The SIGTERM signal is a generic signal used to cause program termination. Unlike SIGKILL, this signal can be blocked, handled, and ignored. It is the normal way to politely ask a program to terminate.

It seems reasonable to me to have `rclcpp` generate a clean shutdown on SIGTERM by default. This could be another `InitOptions` member like `shutdown_on_sigint`, defaulting to `true`.

#### Implementation considerations

Should be pretty easy, install signal handlers the same way we do for SIGINT.

wjwwood, 2021-07-01:
Seems reasonable to me, but it is a significant change in behavior which might affect how things with with say launch right now. So it would need to be thoroughly tested.

@ros2/team for feedback.

emersonknapp, 2021-07-02:
>  significant change in behavior

We could also consider default to false, in which case the default behavior would not change. Though I think I'd prefer to default to true.

clalancette, 2021-07-06:
:+1: to handling SIGTERM as well, though I agree with William that it needs thorough testing.  Messing with the signal handlers is always fragile.