ros2/rclcpp #1707: Service server: Extend API to be able to respond to requests asynchronously

ivanpauno, 2021-07-01:
## Feature request

#### Feature description

Currently, when creating a service one must provide a callback that takes a request and returns a response.
In some cases, the response cannot be calculated immediately.
It would be useful to also be able to create a service server only providing a callback that takes a request (the request message and the request id), but defers to send a response.
In that case, user code is responsible of sending a response later.

#### Implementation considerations

I'm currently using a [hack](https://github.com/ros2/domain_bridge/blob/ab5d3ab13e009eacd98ad11c848da639e06fc59a/include/domain_bridge/service_bridge_impl.inc#L84-L121) in the domain bridge to work around the issue.
The use case is bridging a service between two domains.

In a service callback we want to send the request to another server in a different domain.
A response cannot immediately be sent, because we cannot block in the service callback waiting for the other service to respond.
If we could defer to send a response, we would only need to capture the "request id" in an "async_send_request" callback to the original server, and use in that callback that "request id" and response to complete the bridge.
See [here](https://github.com/ros2/domain_bridge/blob/ab5d3ab13e009eacd98ad11c848da639e06fc59a/include/domain_bridge/service_bridge_impl.inc#L193-L211).

wjwwood, 2021-07-01:
Is this the same as https://github.com/ros2/rclcpp/issues/491?

ivanpauno, 2021-07-02:
> Is this the same as #491?

Yes, it's the same.
My idea was to extend `Service` constructor so that it also accepts a callback with signature

```cpp
void (
      const std::shared_ptr<rmw_request_id_t>,
      const std::shared_ptr<typename ServiceT::Request>);
```

instead of adding an `AsyncService` class.

@wjwwood what do you think?

wjwwood, 2021-07-12:
> Yes, it's the same.

Let's close one or the other then.

wjwwood, 2021-07-12:
> instead of adding an AsyncService class.

Yes, I think this is preferable.

closed 2021-07-12 by ivanpauno