ros2/rclcpp #1710: intra-process communication with serialized message

Jackey-Huo, 2021-07-12:
Hi all,

I'm work on a embedded linux system with 4-core A53 platform, and I'm planning to use ros2 to implement the intra-process communication.

For some reason, I'm not willing to use the default ros msg generation mechanism, but some other IDL language generator (like protobuf, flatsbuffer and so on), so I'd like to pass a serialized message directly to to ros publish function. But then, in the source code, I found the serialized message publish for intra-process is not implemented yet
https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/publisher.hpp#L466

So is there any schedule to implement the serialized publish recently? or is there any alternative method to achieve my goal?

wjwwood, 2021-07-12:
There's no plan to implement this for serialized messages at this time, but it might get done along with other related work to the intra process logic, but again that's not got a concrete schedule.

You could publish inside of a ROS message that just contains binary data.

----

What you probably want is to pass custom objects like protobuf's structure through intra-process (not serialized), in which case that's a feature that we want to implement but have not, see: https://github.com/ros2/rclcpp/issues/1664

wjwwood, 2021-07-12:
There is actually a pull request to do what you want, but it is very out of date, but if you need something like this, it might be a starting place:

https://github.com/ros2/rclcpp/pull/973