ros2/rclcpp #1718: How do I get all the subsriber's information from publisher side

crazyhank, 2021-07-16:
I don't know whether it is ok to ask this question here. I checked the source code of ROS2 framework and fouund no related API to do that.  Maybe it can be done from bottom dds layer, but that is not what I want.

wjwwood, 2021-07-16:
Usually you should ask a question like this on answers.ros.org, see: https://docs.ros.org/en/rolling/Contact.html#help

But I can try to answer it here. Though it would be best if you moved it to answers.ros.org and linked to that question here.

----

The answer is that it depends on what you need and when.

You can list subscriptions on a topic with this API at any time:

https://docs.ros2.org/foxy/api/rclcpp/classrclcpp_1_1Node.html#afba993c401901433b3c8647ef716c7e4

But... this API isn't guaranteed to return any useful information, and it depends on what the middleware underneath has available to give you. You can at least get some information most of the time, like the node name, topic type, and QoS:

https://docs.ros2.org/foxy/api/rclcpp/classrclcpp_1_1TopicEndpointInfo.html

If you want to know how many subscriptions have been "matched", and therefore can communicate, with your publisher, then you can use this:

https://docs.ros2.org/foxy/api/rclcpp/classrclcpp_1_1PublisherBase.html#a60c84260b2ff0fb364fad688ed91234f

but you need to have created a publisher first, and it may take some small amount of time for them to find each other and connect.

----

Hopefully that gets you in the right direction. I'm going to close this for now, but we can continue to discuss here.

closed 2021-07-16 by wjwwood

crazyhank, 2021-07-21:
Thanks for you info.
Now I want to ask one more question: is it possible to restrict the max number of subscribers for a publisher?

fujitatomoya, 2021-07-21:
@crazyhank 

> Usually you should ask a question like this on answers.ros.org, see: https://docs.ros.org/en/rolling/Contact.html#help

questions are welcome but could you post that https://answers.ros.org/questions/ instead?

thanks