ros2/rclcpp #1724: Template parameter deduction failure in Node::create_subscription (works on rolling but not on galactic)

tylerjw, 2021-07-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries

#### Steps to reproduce issue
I have a PR that builds and works on rolling but not on galactic.  It has something to do with the template parameter deduction in the function `Node::create_subscription`.  Here is the most informative output I have:

```c++
no known conversion for argument 1 from ‘std::shared_ptr<rclcpp::Subscription<std::shared_ptr<const sensor_msgs::msg::JointState_<std::allocator<void> > >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<std::shared_ptr<const sensor_msgs::msg::JointState_<std::allocator<void> > >, std::allocator<void> > > >’ to ‘std::shared_ptr<rclcpp::Subscription<sensor_msgs::msg::JointState_<std::allocator<void> > > >&&’
```

Here is the bit of code that is the problem:

```c++
using JointStateUpdateCallback = boost::function<void(const sensor_msgs::msg::JointState::ConstSharedPtr&)>;

...

rclcpp::Subscription<sensor_msgs::msg::JointState>::SharedPtr joint_state_subscription_;
void CurrentStateMonitorRclInterface::createJointStateSubscription(const std::string& topic,
                                                                   JointStateUpdateCallback callback)
{
  joint_state_subscription_ =
      node_->create_subscription<sensor_msgs::msg::JointState>(topic, 25, callback);
}

...

rcl_interface_->createJointStateSubscription(
          joint_states_topic, std::bind(&CurrentStateMonitor::jointStateCallback, this, std::placeholders::_1));
```

Please let me know if I can provide more information.  I'm also interested if there is some way I can just work around this for galactic.  Here is the PR that has this problem: https://github.com/ros-planning/moveit2/pull/562

tylerjw, 2021-07-22:
I suspect it has something to do with https://github.com/ros2/rclcpp/commit/1037822a63330495dcf0de5e8f20544375a5f116#diff-7c9ef81e95ef36be3289b391002ac5e81a3545293a88f9f53c11f5cd73e6280c

Is there someway I can get this to work on galactic without this commit.  Also, I don't really understand the commit message about the callback type, should I use a different callback type?  Is there something wrong with the one I'm using?

tylerjw, 2021-07-22:
I got this working with these changes on Galactic.  https://github.com/ros-planning/moveit2/pull/562/commits/8a69cf8cc60871f717d86e4ebafed968e6d56f01  I'll leave this here in case it is interesting to anyone.  Close if you'd like.

fujitatomoya, 2021-07-23:
I will go ahead to close this.

closed 2021-07-23 by fujitatomoya