ros2/rclcpp #1735: Segfault from within rclcpp::Node destructor

Zero00INF, 2021-08-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 Focal Fossa ( probably all OS-es as related issues/PRs will indicate )
- Installation type:
  - Binaries via pkg manager ( ros-base only)
- Version or commit hash:
  - Foxy, latest ( as of 2021.08.02 )
- DDS implementation:
  - Fast-RTPS based on callstack
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Hello everyone in the rclcpp developer team!
We are using the latest ros2 foxy (base) package for ubuntu 20.04 installed via
```
sudo apt install ros-foxy-ros-base
```
Mentioning this only because it might be relevant later.

We are getting a similar behaviour and a segfault from Node::~Node() destructor(-chain) that #1468 describes. 
(Below I'll link a callstack if this is somehow different from that)
As the old bug report says it is hard to reliably reproduce these kind of issues on anything else that the actual ( and complex ) code base / environment.
That bug report was filed for foxy/latest - at that time - and already had its related pull-request (#1469 ) merged back in March which should've solved the issue ( at long as ours was caused by the same thing ofc ). 

Checking the  _foxy_ branch in the repo, however, we couldn't find the effects of said pull request, checking the version of the rclcpp which ros-foxy-* by-default installs we get:
```
  Installed: 2.3.1-1focal.20210513.202828
  Candidate: 2.3.1-1focal.20210513.202828
```

The 2.3.1 tag (assuming it is the correct one for the above lines ) in the repository also directs to a code state where the destructors are empty for Node::~Node() & LifecycleNode::~LifecycleNode() which the merged PR should've fixed. 

I do not know if this was a conscious decision, we are missing something that's in plain sight, or it is only a master->foxy cherry-pick away. Do tell us if the issue is on our end, or it should be long resolved by now, however:

### Actual Problem 
Our actual problem is that we need to get over this hurdle quite fast, which isn't realistic considering your time, plus how long it would take to get the administrative things done for a new foxy package ( if at all this small change could warrant it.. ). So as an alternative if you can tell us how we could get an rclcpp package containing that PR that does not break anything in our/the latest foxy version, that would be a lifesaver too.

#### Expected behavior
No crashes during shutdown 

#### Actual behavior
On shutdown when the ROS Node is destructed we get the following :
Error message:
Segmentation fault      (core dumped)

Relevant callstack:
```
?? (+0x204a9c) from /opt/ros/foxy/lib/libfastrtps.so.2
in eprosima::fastdds::rtps::SharedMemManager::Listener::~Listener() (+0x58d930) from /opt/ros/foxy/lib/libfastrtps.so.2
in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (+0x8c4c8) from ../../out_x86_64-ubuntu-2004_Release/bin/aimsim
in eprosima::fastdds::rtps::SharedMemTransport::delete_input_channel(eprosima::fastdds::rtps::SharedMemChannelResource*) (+0x5a6dcb) from /opt/ros/foxy/lib/libfastrtps.so.2
in eprosima::fastdds::rtps::SharedMemTransport::CloseInputChannel(eprosima::fastrtps::rtps::Locator_t const&) (+0x5a70b3) from /opt/ros/foxy/lib/libfastrtps.so.2
in eprosima::fastrtps::rtps::RTPSParticipantImpl::disable() (+0x2f0ca3) from /opt/ros/foxy/lib/libfastrtps.so.2
in eprosima::fastrtps::rtps::RTPSDomain::removeRTPSParticipant(eprosima::fastrtps::rtps::RTPSParticipant*) (+0x2f6cdd) from /opt/ros/foxy/lib/libfastrtps.so.2
in eprosima::fastrtps::ParticipantImpl::~ParticipantImpl() (+0x303cec) from /opt/ros/foxy/lib/libfastrtps.so.2
in eprosima::fastrtps::ParticipantImpl::~ParticipantImpl() (+0x303f8d) from /opt/ros/foxy/lib/libfastrtps.so.2
in eprosima::fastrtps::Domain::removeParticipant(eprosima::fastrtps::Participant*) (+0x2f9f4c) from /opt/ros/foxy/lib/libfastrtps.so.2
in rmw_fastrtps_shared_cpp::destroy_participant(CustomParticipantInfo*) (+0x26f75) from /opt/ros/foxy/lib/librmw_fastrtps_shared_cpp.so
in rmw_fastrtps_shared_cpp::decrement_context_impl_ref_count(rmw_context_t*) (+0x26098) from /opt/ros/foxy/lib/librmw_fastrtps_shared_cpp.so
in rcl_node_fini (+0x19ee4) from /opt/ros/foxy/lib/librcl.so
in ?? (+0xfc44e) from /opt/ros/foxy/lib/librclcpp.so
in ?? (+0xfc5f5) from /opt/ros/foxy/lib/librclcpp.so
in rclcpp::node_interfaces::NodeBase::~NodeBase() (+0xfcb8c) from /opt/ros/foxy/lib/librclcpp.so
in rclcpp::node_interfaces::NodeBase::~NodeBase() (+0xfccdd) from /opt/ros/foxy/lib/librclcpp.so
in rclcpp::Node::~Node() (+0xf4580) from /opt/ros/foxy/lib/librclcpp.so
```

wjwwood, 2021-08-02:
Without some information on how you're using the API it's going to be difficult to tell you what the solution is. These problems typically stem from users having globals in their code and using globals in ROS's library at the same time (or in supporting libraries like FastDDS, which seems to be the case here).

So if you're using global references to nodes, try to avoid that.

We also ask for a minimal, workable example that demonstrates the problem. I understand that it's difficult to do that, as pointed out by you and the previous issue you linked to, but that doesn't make it any easier for us to work on it. We either need a reproducible example or we need someone to reason about the code and suggest where the problem could be and how to fix it, which is what happened in the other issue #1468.

---

> Our actual problem is that we need to get over this hurdle quite fast

A bug fix into Foxy will take time, a month or longer most likely, even if someone proposes a fix in a pull request for us.

>  So as an alternative if anyone can tell us how we could get an rclcpp package containing that PR that does not break anything in our/the latest foxy version, that would be a lifesaver too.

Well first we need a solution, which requires understanding the problem. Unfortunately the callstack isn't much to go on in this case. I'm not even sure this is a problem with `rclcpp` and not `rmw_fastrtps_cpp`.

But once we have a solution, you can build everything from `rclcpp` (or `rmw_fastrtps_cpp` or whatever package ends up changing) up to your code, from source. That's really the only option. You can still install `rclcpp`'s dependencies from a binary source like apt-get.

wjwwood, 2021-08-02:
> Checking the foxy branch in the repo, however, we couldn't find the effects of said pull request

I think you're right though, #1469 has not been backported to foxy. It might be possible to do that, but it would be helpful to know if that fixes your problem before we do that. Can you test that patch against your application?

wjwwood, 2021-08-02:
In fact there's already https://github.com/ros2/rclcpp/pull/1736, but we still need to see if that's a solution for your issue.

Zero00INF, 2021-08-02:
Right, I expect us to get back to you within a day with at least a result and possibly a better understanding of what goes on (timezones, meetings etc). Thanks for the fast reply & investigation!

clalancette, 2021-08-19:
We believe this was fixed by #1736 .  Can you please confirm?

clalancette, 2021-09-02:
Since we believe this was fixed and released by #1736, I'm going to close this.  Feel free to reopen if you are still having the issue.

closed 2021-09-02 by clalancette

Zero00INF, 2021-09-03:
Sorry for not getting back earlier, sure you can close it, if we find anything further we'll start with a minimal example ready + after debugging plenty beforehand