ros2/rclcpp #1742: QNX support: can_be_nullptr causes build failure with packages using create_service()

asobhy-qnx, 2021-08-04:
can_be_nullptr causes build failure with packages using create_service()

After commit ros2/rclcpp@0750dc4, Rolling QNX build https://gitlab.com/qnx/ros2/ros2_qnx/-/jobs/1443549330 was broken due to the following error in examples_rclcpp_minimal_services:

```
Starting >>> examples_rclcpp_minimal_service
--- stderr: examples_rclcpp_minimal_service                             
In file included from /home/builder/ros2_rolling/install/aarch64le/include/rclcpp/service.hpp:28,
                 from /home/builder/ros2_rolling/install/aarch64le/include/rclcpp/callback_group.hpp:25,
                 from /home/builder/ros2_rolling/install/aarch64le/include/rclcpp/any_executable.hpp:20,
                 from /home/builder/ros2_rolling/install/aarch64le/include/rclcpp/memory_strategy.hpp:25,
                 from /home/builder/ros2_rolling/install/aarch64le/include/rclcpp/memory_strategies.hpp:18,
                 from /home/builder/ros2_rolling/install/aarch64le/include/rclcpp/executor_options.hpp:20,
                 from /home/builder/ros2_rolling/install/aarch64le/include/rclcpp/executor.hpp:37,
                 from /home/builder/ros2_rolling/install/aarch64le/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /home/builder/ros2_rolling/install/aarch64le/include/rclcpp/executors.hpp:21,
                 from /home/builder/ros2_rolling/install/aarch64le/include/rclcpp/rclcpp.hpp:155,
                 from /home/builder/ros2_rolling/src/ros2/examples/rclcpp/services/minimal_service/main.cpp:19:
/home/builder/ros2_rolling/install/aarch64le/include/rclcpp/any_service_callback.hpp: In instantiation of 'struct rclcpp::detail::can_be_nullptr<void (&)(std::__1::shared_ptr<rmw_request_id_t>, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Request_<std::__1::allocator<void> > >, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::__1::allocator<void> > >), void>':
/home/builder/ros2_rolling/install/aarch64le/include/rclcpp/any_service_callback.hpp:104:80:   required by substitution of 'template<class CallbackT, typename std::__1::enable_if<rclcpp::detail::can_be_nullptr<CallbackT, void>::value, int>::type <anonymous> > void rclcpp::AnyServiceCallback<example_interfaces::srv::AddTwoInts>::set<CallbackT, <enumerator> >(CallbackT&&) [with CallbackT = void (&)(std::__1::shared_ptr<rmw_request_id_t>, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Request_<std::__1::allocator<void> > >, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::__1::allocator<void> > >); typename std::__1::enable_if<rclcpp::detail::can_be_nullptr<CallbackT, void>::value, int>::type <anonymous> = <missing>]'
/home/builder/ros2_rolling/install/aarch64le/include/rclcpp/create_service.hpp:43:3:   required from 'typename rclcpp::Service<ServiceT>::SharedPtr rclcpp::create_service(std::__1::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::__1::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, const string&, CallbackT&&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = example_interfaces::srv::AddTwoInts; CallbackT = void (&)(std::__1::shared_ptr<rmw_request_id_t>, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Request_<std::__1::allocator<void> > >, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::__1::allocator<void> > >); typename rclcpp::Service<ServiceT>::SharedPtr = std::__1::shared_ptr<rclcpp::Service<example_interfaces::srv::AddTwoInts> >; std::__1::string = std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >; rmw_qos_profile_t = rmw_qos_profile_t; rclcpp::CallbackGroup::SharedPtr = std::__1::shared_ptr<rclcpp::CallbackGroup>]'
/home/builder/ros2_rolling/install/aarch64le/include/rclcpp/node_impl.hpp:147:53:   required from 'typename rclcpp::Service<ServiceT>::SharedPtr rclcpp::Node::create_service(const string&, CallbackT&&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = example_interfaces::srv::AddTwoInts; CallbackT = void (&)(std::__1::shared_ptr<rmw_request_id_t>, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Request_<std::__1::allocator<void> > >, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::__1::allocator<void> > >); typename rclcpp::Service<ServiceT>::SharedPtr = std::__1::shared_ptr<rclcpp::Service<example_interfaces::srv::AddTwoInts> >; std::__1::string = std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >; rmw_qos_profile_t = rmw_qos_profile_t; rclcpp::CallbackGroup::SharedPtr = std::__1::shared_ptr<rclcpp::CallbackGroup>]'
/home/builder/ros2_rolling/src/ros2/examples/rclcpp/services/minimal_service/main.cpp:40:82:   required from here
/home/builder/ros2_rolling/install/aarch64le/include/rclcpp/any_service_callback.hpp:44:74: error: assignment of read-only location 'std::__1::declval<void (&)(std::__1::shared_ptr<rmw_request_id_t>, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Request_<std::__1::allocator<void> > >, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::__1::allocator<void> > >)>()'
     decltype(std::declval<T>() == nullptr), decltype(std::declval<T &>() = nullptr)>>
                                                      ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
/home/builder/ros2_rolling/install/aarch64le/include/rclcpp/any_service_callback.hpp:44:74: error: assignment of read-only location 'std::__1::declval<void (&)(std::__1::shared_ptr<rmw_request_id_t>, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Request_<std::__1::allocator<void> > >, std::__1::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::__1::allocator<void> > >)>()'
```
It seems that the compiler is refusing to accept the assignment ```decltype(std::declval<T &>() = nullptr```

QNX uses g++ v8.3.0 with LLVM libc++

After removing ```decltype(std::declval<T &>() = nullptr``` and using can_be_nullptr as below, building examples_rclcpp_minimal_service succeeds.
 
```
template<typename T>
struct can_be_nullptr<T, std::void_t<
    decltype(std::declval<T>() == nullptr)>>
  : std::true_type {};
}
```
Is it okay for QNX implementation to test only if the callback function is equal to nullptr?

MostafaGomaa, 2021-08-16:
Hello,
I am wondering whats the status of this, I am getting the same issue trying to build rclcpp/rclcpp/test/benchmark:
```
In file included from /home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/service.hpp:28,
                 from /home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/callback_group.hpp:25,
                 from /home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/any_executable.hpp:20,
                 from /home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/memory_strategy.hpp:25,
                 from /home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/memory_strategies.hpp:18,
                 from /home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/executor_options.hpp:20,
                 from /home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/executor.hpp:37,
                 from /home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/executors.hpp:21,
                 from /home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/rclcpp.hpp:155,
                 from /home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/test/benchmark/benchmark_client.cpp:19:
/home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/any_service_callback.hpp: In instantiation of ‘struct rclcpp::detail::can_be_nullptr<ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>&, void>’:
/home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/any_service_callback.hpp:104:80:   required by substitution of ‘template<class CallbackT, typename std::enable_if<rclcpp::detail::can_be_nullptr<CallbackT, void>::value, int>::type <anonymous> > void rclcpp::AnyServiceCallback<test_msgs::srv::Empty>::set<CallbackT, <enumerator> >(CallbackT&&) [with CallbackT = ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>&; typename std::enable_if<rclcpp::detail::can_be_nullptr<CallbackT, void>::value, int>::type <anonymous> = <missing>]’
/home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/create_service.hpp:43:3:   required from ‘typename rclcpp::Service<ServiceT>::SharedPtr rclcpp::create_service(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, const string&, CallbackT&&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = test_msgs::srv::Empty; CallbackT = ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>&; typename rclcpp::Service<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Service<test_msgs::srv::Empty> >; std::__cxx11::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_t; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
/home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/node_impl.hpp:147:53:   required from ‘typename rclcpp::Service<ServiceT>::SharedPtr rclcpp::Node::create_service(const string&, CallbackT&&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = test_msgs::srv::Empty; CallbackT = ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>&; typename rclcpp::Service<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Service<test_msgs::srv::Empty> >; std::__cxx11::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_t; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
/home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/test/benchmark/benchmark_client.cpp:40:93:   required from here
/home/localstorage/workspace/ros2/rolling/distro/src/ros2/rclcpp/rclcpp/include/rclcpp/any_service_callback.hpp:44:74: error: no match for ‘operator=’ (operand types are ‘ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>’ and ‘std::nullptr_t’)
     decltype(std::declval<T>() == nullptr), decltype(std::declval<T &>() = nullptr)>>
```

Is the fix mentioned by  @asobhy-qnx  safe, does it break anything?

kb0n, 2021-08-17:
Proper fix is actually simply replacing the `=` assignment by a `==` comparison
```diff
--- a/rclcpp/include/rclcpp/any_service_callback.hpp
+++ b/rclcpp/include/rclcpp/any_service_callback.hpp
@@ -41,7 +41,7 @@ struct can_be_nullptr : std::false_type {};
 // We also test if `T &` can be assigned to `nullptr` to avoid the issue.
 template<typename T>
 struct can_be_nullptr<T, std::void_t<
-    decltype(std::declval<T>() == nullptr), decltype(std::declval<T &>() = nullptr)>>
+    decltype(std::declval<T>() == nullptr), decltype(std::declval<T &>() == nullptr)>>
   : std::true_type {};
 }  // namespace detail

```

asobhy-qnx, 2021-08-17:
@kb0n that does make sense. What is confusing me is the comment above the line 
‘’’
// We also test if `T &` can be assigned to `nullptr` to avoid the issue.
‘’’
Also there is a CI for Rolling so the builds for Linux, Windows and Mac should have failed when building examples_rclcpp_minimal_service but they did not, so the compilers accepted the assignment.

Technically speaking declval returns an rvalue reference and in this case when we know the type is a function pointer then the value should be a memory address, so we should be checking to see if this memory address is equal to null or not using the == operator.

kb0n, 2021-08-17:
I'm not that familiar with the code here to see where and how `can_be_nullptr` is being used. So the whole comment-block is confusing to me :shrug:  
The metaprogramming part with `decltype(std::declval<T &>() == nullptr)` is valid code though.

As for the CI: Maybe it only tests packages up to a certain point? So `rclcpp` (or anything else) should compile fine if `can_be_nullptr` is not being used.  
But this is just a guess, I didn't look into the CI jobs yet.

wjwwood, 2021-08-19:
@ivanpauno do you have some time to look at this? (I see you assigned yourself)

ivanpauno, 2021-08-20:
> Is it okay for QNX implementation to test only if the callback function is equal to nullptr?

That sounds fine to me only for qnx, I added the check for the availability of the assignment operator because it was generating a warning otherwise (see comment [here](https://github.com/ros2/rclcpp/blob/0750dc418ada283d0248e7bb2e0a79ec6dbe428a/rclcpp/include/rclcpp/any_service_callback.hpp#L39-L41)).
If there's another way to avoid the warning it seems okay to me to also completely remove that check from all the platforms.

> Proper fix is actually simply replacing the = assignment by a == comparison

That doesn't seem to be the case, why would you test `== nullptr` for both `T` and `T &`?

---

Alternatively, we can delete the `can_be_nullptr` trait and check for nullptr `callback_` after calling `callback_.emplace()` instead of checking that in the passed argument.
That will simplify the code quite a bit. 

---

@MostafaGomaa @asobhy-qnx A PR patching this in any of the ways mentioned before is really welcomed!

kb0n, 2021-08-24:
@ivanpauno Yes you're right, I just read `error: assignment of read-only location` and jumped to conclusions.

However, the error occurs all over the place. I`m running debian-10 and get
```bash
                 from ~/install/ros2_rolling/src/ros2/rclcpp/rclcpp/test/benchmark/benchmark_client.cpp:19:
~/install/ros2_rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/any_service_callback.hpp: In instantiation of ‘struct rclcpp::detail::can_be_nullptr<ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>&, void>’:
~/install/ros2_rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/any_service_callback.hpp:109:80:   required by substitution of ‘template<class CallbackT, typename std::enable_if<rclcpp::detail::can_be_nullptr<CallbackT, void>::value, int>::type <anonymous> > void rclcpp::AnyServiceCallback<test_msgs::srv::Empty>::set<CallbackT, <enumerator> >(CallbackT&&) [with CallbackT = ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>&; typename std::enable_if<rclcpp::detail::can_be_nullptr<CallbackT, void>::value, int>::type <anonymous> = <missing>]’
~/install/ros2_rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/create_service.hpp:43:3:   required from ‘typename rclcpp::Service<ServiceT>::SharedPtr rclcpp::create_service(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, const string&, CallbackT&&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = test_msgs::srv::Empty; CallbackT = ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>&; typename rclcpp::Service<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Service<test_msgs::srv::Empty> >; std::__cxx11::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_t; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
~/install/ros2_rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/node_impl.hpp:147:53:   required from ‘typename rclcpp::Service<ServiceT>::SharedPtr rclcpp::Node::create_service(const string&, CallbackT&&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = test_msgs::srv::Empty; CallbackT = ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>&; typename rclcpp::Service<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Service<test_msgs::srv::Empty> >; std::__cxx11::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_t; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
~/install/ros2_rolling/src/ros2/rclcpp/rclcpp/test/benchmark/benchmark_client.cpp:40:93:   required from here
~/install/ros2_rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/any_service_callback.hpp:48:74: error: no match for ‘operator=’ (operand types are ‘ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>’ and ‘std::nullptr_t’)
     decltype(std::declval<T>() == nullptr), decltype(std::declval<T &>() = nullptr)>>
                                                      ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
~/install/ros2_rolling/src/ros2/rclcpp/rclcpp/test/benchmark/benchmark_client.cpp:36:8: note: candidate: ‘ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>& ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>::operator=(const ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>&)’ <deleted>
       [](
        ^
~/install/ros2_rolling/src/ros2/rclcpp/rclcpp/test/benchmark/benchmark_client.cpp:36:8: note:   no known conversion for argument 1 from ‘std::nullptr_t’ to ‘const ClientPerformanceTest::SetUp(benchmark::State&)::<lambda(test_msgs::srv::Empty_Request_<std::allocator<void> >::SharedPtr, test_msgs::srv::Empty_Response_<std::allocator<void> >::SharedPtr)>&’
```

or

```bash
                 from ~/install/ros2_rolling/src/ros2/examples/rclcpp/services/minimal_service/main.cpp:19:
~/install/ros2_rolling/install/rclcpp/include/rclcpp/any_service_callback.hpp: In instantiation of ‘struct rclcpp::detail::can_be_nullptr<void (&)(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<example_interfaces::srv::AddTwoInts_Request_<std::allocator<void> > >, std::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::allocator<void> > >), void>’:
~/install/ros2_rolling/install/rclcpp/include/rclcpp/any_service_callback.hpp:109:80:   required by substitution of ‘template<class CallbackT, typename std::enable_if<rclcpp::detail::can_be_nullptr<CallbackT, void>::value, int>::type <anonymous> > void r [... cut ...]
~/install/ros2_rolling/install/rclcpp/include/rclcpp/any_service_callback.hpp:48:74: error: assignment of read-only location ‘std::declval<void (&)(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<example_interfaces::srv::AddTwoInts_Request_<std::allocator<void> > >, std::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::allocator<void> > >)>()’
```

I think you need a different approach for what you are trying to achieve.

> [...] I added the check for the availability of the assignment operator [...]

Not sure if actually **assigning** something to `nullptr` is the proper way of checking if it **can** be assigned to it.

ivanpauno, 2021-08-24:
> Not sure if actually assigning something to nullptr is the proper way of checking if it can be assigned to it. (sth. like std::is_convertible?)

is_convertible doesn't seem to fit well, because that's only about implict conversions.
https://en.cppreference.com/w/cpp/types/is_assignable seems better, @kb0n could you open a PR trying that.

ivanpauno, 2021-08-24:
Reopening because the last PR fixed the issue for QNX, but it also happens in other OSs (debian).

acochrane, 2023-03-16:
Having this issue in debian 10. I spent a little time playing with std::is_assignable but not enough to get a PR together. Anyway, compilation succeeds with the __QNXNTO__ solution. Makes me curious what kind of problems I might run into in linux.

PR #1752 (2021-08-23): remove can_be_nullptr assignment check for QNX case
Fixes #1742 
@ivanpauno