ros2/rclcpp #1743: Function template version of declare_parameter does not work with uint32_t and uint64_t

Bi0T1N, 2021-08-04:
## Bug report

**Required Info:**

- Operating System:
  - Linux Mint 20.1 (based on Ubuntu 20.04)
- Installation type:
  - ROS 2 Galactic Geochelone Debian packages for Ubuntu Focal
- Version or commit hash:
  - Latest Galactic Geochelone
- DDS implementation:
  - default one shipped with the Debian package -> Cyclone DDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
The minimal (non)working example is based on the tutorial for [Using parameters in a class (C++)](https://docs.ros.org/en/galactic/Tutorials/Using-Parameters-In-A-Class-CPP.html):
```c++
#include <rclcpp/rclcpp.hpp>
#include <chrono>
#include <string>
#include <functional>

using namespace std::chrono_literals;

class ParametersClass: public rclcpp::Node
{
  public:
    ParametersClass()
      : Node("parameter_node")
    {
      this->declare_parameter<int8_t>("this_int_works1", 1);
      this->declare_parameter<int16_t>("this_int_works2", 1);
      this->declare_parameter<int32_t>("this_int_works3", 1);
      this->declare_parameter<int64_t>("this_int_works4", 1);

      this->declare_parameter<uint8_t>("this_uint_works1", 1);
      this->declare_parameter<uint16_t>("this_uint_works1", 1);
      this->declare_parameter<uint32_t>("this_uint_does_not_work1", 1);
      this->declare_parameter<uint64_t>("this_uint_does_not_work2", 1);
    }
};

int main(int argc, char** argv)
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<ParametersClass>());
  rclcpp::shutdown();
  return 0;
}
```

```cmake
cmake_minimum_required(VERSION 3.8)
project(cpp_parameters)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)

add_executable(parameter_node src/cpp_parameters_node.cpp)
ament_target_dependencies(parameter_node rclcpp)

install(TARGETS
  parameter_node
  DESTINATION lib/${PROJECT_NAME}
)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # uncomment the line when a copyright and license is not present in all source files
  #set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # uncomment the line when this package is not in a git repo
  #set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
```

#### Expected behavior
I'd expect that all of the above version of `declare_parameter` would work.
#### Actual behavior
All versions except the ones for `uint32_t` and `uint64_t` are working. It says `call of overloaded ‘ParameterValue(const unsigned int&)’ is ambiguous`:
```
In file included from /opt/ros/galactic/include/rclcpp/node.hpp:1315,
                 from /opt/ros/galactic/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/galactic/include/rclcpp/executors.hpp:22,
                 from /opt/ros/galactic/include/rclcpp/rclcpp.hpp:156,
                 from /<path>/param_test/cpp_parameters/src/cpp_parameters_node.cpp:1:
/opt/ros/galactic/include/rclcpp/node_impl.hpp: In instantiation of ‘auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool) [with ParameterT = unsigned int; std::string = std::__cxx11::basic_string<char>; rcl_interfaces::msg::ParameterDescriptor = rcl_interfaces::msg::ParameterDescriptor_<std::allocator<void> >]’:
/<path>/param_test/cpp_parameters/src/cpp_parameters_node.cpp:21:70:   required from here
/opt/ros/galactic/include/rclcpp/node_impl.hpp:205:15: error: call of overloaded ‘ParameterValue(const unsigned int&)’ is ambiguous
  205 |       rclcpp::ParameterValue(default_value),
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/galactic/include/rclcpp/parameter.hpp:26,
                 from /opt/ros/galactic/include/rclcpp/node_interfaces/node_parameters_interface.hpp:28,
                 from /opt/ros/galactic/include/rclcpp/node.hpp:54,
                 from /opt/ros/galactic/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/galactic/include/rclcpp/executors.hpp:22,
                 from /opt/ros/galactic/include/rclcpp/rclcpp.hpp:156,
                 from /<path>/param_test/cpp_parameters/src/cpp_parameters_node.cpp:1:
/opt/ros/galactic/include/rclcpp/parameter_value.hpp:94:12: note: candidate: ‘rclcpp::ParameterValue::ParameterValue(double)’
   94 |   explicit ParameterValue(const double double_value);
      |            ^~~~~~~~~~~~~~
/opt/ros/galactic/include/rclcpp/parameter_value.hpp:91:12: note: candidate: ‘rclcpp::ParameterValue::ParameterValue(float)’
   91 |   explicit ParameterValue(const float double_value);
      |            ^~~~~~~~~~~~~~
/opt/ros/galactic/include/rclcpp/parameter_value.hpp:88:12: note: candidate: ‘rclcpp::ParameterValue::ParameterValue(int64_t)’
   88 |   explicit ParameterValue(const int64_t int_value);
      |            ^~~~~~~~~~~~~~
/opt/ros/galactic/include/rclcpp/parameter_value.hpp:85:12: note: candidate: ‘rclcpp::ParameterValue::ParameterValue(int)’
   85 |   explicit ParameterValue(const int int_value);
      |            ^~~~~~~~~~~~~~
/opt/ros/galactic/include/rclcpp/parameter_value.hpp:82:12: note: candidate: ‘rclcpp::ParameterValue::ParameterValue(bool)’
   82 |   explicit ParameterValue(const bool bool_value);
      |            ^~~~~~~~~~~~~~
/opt/ros/galactic/include/rclcpp/parameter_value.hpp:71:7: note: candidate: ‘rclcpp::ParameterValue::ParameterValue(const rclcpp::ParameterValue&)’
   71 | class ParameterValue
      |       ^~~~~~~~~~~~~~
/opt/ros/galactic/include/rclcpp/parameter_value.hpp:71:7: note: candidate: ‘rclcpp::ParameterValue::ParameterValue(rclcpp::ParameterValue&&)’
```
#### Additional information
The same occurs with a `rclcpp_lifecycle::LifecycleNode`.  
It might be useful to test all template variants through the unit tests. It seems only `std::string` and `int` is tested.

wjwwood, 2021-08-04:
Integers for parameters are always stored as `int64_t`, so in the case of `uint64_t` it cannot safely work because there might be loss of data converting it `int64_t` as the max for `uint64_t` is larger than the max for `int64_t`.

I'm not sure why `uint32_t` doesn't work. Likely the ones that work are selecting the `explicit ParameterValue(const int int_value);` candidate? `uint32_t` may or may not work for that depending on the system.

I would say the only ones sure to work are `this->declare_parameter<int>` and `this->declare_parameter<int64_t>`, and that's expected. If we want more to work, then we could support them all except `uint64_t` I think.

I'd mark this as expected behavior at the moment to be honest. @ivanpauno what do you think?

ivanpauno, 2021-08-04:
ParameterValue() has a constructor taking an `int` and another one taking an `int64_t`.

According to the standard, unsigned char -> int and unsigned short -> int is a valid integer promotion, but unsigned char -> int64_t and unsigned short -> int64_t are conversions.
There's no ambiguity in that case, as integer promotion has higher priority that integer conversion.

unsigned int -> int is a conversion, and also unsigned int -> int64_t.
In that case there's ambiguity.
(what is an integer promotion and what not seems to be a bit arbitrary tbh)

See https://en.cppreference.com/w/cpp/language/implicit_conversion.

---

> I'd mark this as expected behavior at the moment to be honest. @ivanpauno what do you think?

Agreed, I think we should only allow the <int64_t> option and not the others, to be more explicit about what's the actualy type of the parameter.

Bi0T1N, 2021-08-05:
> I would say the only ones sure to work are `this->declare_parameter<int>` and `this->declare_parameter<int64_t>`, and that's expected.

I think for consistency reasons it should only allow `int64_t` then as the width of `int` is not defined. That's also what is written in the [Parameter API design in ROS](https://design.ros2.org/articles/ros_parameters.html#supported-datatypes). Actually I found that it's also documented in the normal documentation [here](https://docs.ros.org/en/galactic/Concepts/About-ROS-2-Parameters.html).
In contrast, there is no consistency with the interface definition since the fixed width types are allowed there. [[1](https://design.ros2.org/articles/interface_definition.html), [2](https://design.ros2.org/articles/legacy_interface_definition.html), [3](https://design.ros2.org/articles/idl_interface_definition.html)]

However, if it's expected behavior it should be covered by the code so that the compiler throws an useful error messages which says that this data type is not supported for parameters.

roncapat, 2023-07-13:
Sorry for reviving this one, but what about uint64_t? May I ask why it is unsupported?