ros2/rclcpp #1746: rclcpp_lifecycle::LifecycleNode::get_current_state is not thread safe

igorbanfi, 2021-08-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.4
- Installation type:
  - binaries
- Version or commit hash:
  - foxy
- DDS implementation:
  - fastDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

In code below get_current_state is called in 2 threads and after random number of iterations ``` Error in state! Internal state_handle is NULL. ``` is thrown when calling id() state.

The problem disappears if only one thread calls get_current_state function.

```
#include <chrono>
#include <iostream>
#include <memory>
#include <thread>
#include "lifecycle_msgs/msg/transition.hpp"
#include "rclcpp/rclcpp.hpp"
#include "rclcpp/publisher.hpp"
#include "rclcpp_lifecycle/lifecycle_node.hpp"
#include "rcutils/logging_macros.h"

/// Get current node state in a loop
void checkState(std::shared_ptr<rclcpp_lifecycle::LifecycleNode> node) {

   // infinite loop
   while (1) {
      // This call fails
      node->get_current_state().id();
   }
}

int main(int argc, char * argv[])
{
   setvbuf(stdout, NULL, _IONBF, BUFSIZ);

   rclcpp::init(argc, argv);

   std::shared_ptr<rclcpp_lifecycle::LifecycleNode> lc_node =
           std::make_shared<rclcpp_lifecycle::LifecycleNode>("lc_talker");

   lc_node->trigger_transition(lifecycle_msgs::msg::Transition::TRANSITION_CONFIGURE);

   // Created multiple threads
   std::thread thread_object_1(checkState, lc_node);
   std::thread thread_object_2(checkState, lc_node);

   // wait
   thread_object_1.join();

   rclcpp::shutdown();

   return 0;
}
```

devrite, 2021-08-09:
This is true. Current states queries from impl, sets to local var and returns it.

https://github.com/ros2/rclcpp/blob/b918bd4c25c2ec59bf6ef58eff04612902eb9330/rclcpp_lifecycle/src/lifecycle_node_interface_impl.hpp#L335-L340

Solution (if it is safe to query the internal state from multiple threads): Do not set the local var, just return the constructed value.

fujitatomoya, 2021-08-09:
This is a bug, confirmed that core crash happens on mainline.

> Do not set the local var, just return the constructed value.

for doing this, I think we need to change the API to return the object.

> But since a state may change parallel (outside service call proc by some cb) of others threads querying it there probably should be locks probably.

right. I believe that `state_machine_` needs to be protected by mutex lock. (note that we need to release the lock when calling the user callback.)

devrite, 2021-08-11:
Seems reasonable to unlock while doing user cb. Requires, pre, cb, post not to be an atomic operation. If that is the case an unlock can be done. E.g. if no other changing/writing calls from other threads are allowed during the release of pre and reacquiring of cb or post, we can only unlock if other writing API calls recognize an ongoing atomic transaction (due to code in pre). This would just abort any further transaction with ongoing transaction error or maybe a possibility (conditionvar) to wait.

The other option is to provide an API impl to the callback that does not need to lock, but this would be a breaking change I guess.

alsora, 2021-09-09:
+1 on this.
Thank you @fujitatomoya for the work!

I also noticed the crash when concurrently invoking a lifecycle node state transition while at the same time the node was checking the current state.

closed 2022-10-28 by fujitatomoya