ros2/rclcpp #1749: Parameter redeclaration

livanov93, 2021-08-17:
## Bug report

**Required Info:**
`automatically_declare_parameters_from_overrides` method on the `NodeOptions` redeclares already declared paramters.

- Operating System:
  - Ubuntu 20.04.2 LTS
- Installation type:
  - binaries
- Version or commit hash:
  - foxy
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Test package can be found [here](https://github.com/livanov93/test_params_pkg). Put the package in your ~/ros2_ws/src and build it from ~/ros2_ws/
```
colcon build --packages-select test_params_pkg
source install/setup.bash
ros2 run test_params_pkg test_params_node --ros-args -p declared_parameter:="Totally"

```

#### Expected behavior
```
declared_param: Totally
```
#### Actual behavior
```
terminate called after throwing an instance of 'rclcpp::exceptions::ParameterAlreadyDeclaredException'
  what():  parameter 'declared_parameter' has already been declared
```
#### Additional information

wjwwood, 2021-08-17:
I think that if you trace the source of the exception, it will be this line:

https://github.com/livanov93/test_params_pkg/blob/6c856e28ec9b69850517172c5b25572f5e515d60/src/test_params_node.cpp#L12

So, it's actually the user code that is re-declaring the parameter, not this setting. But you have a point that it might not be possible to declare a parameter before the logic for `automatically_declare_parameters_from_overrides` manages to declare them.

So we might want to adjust the documentation, but this is expected behavior as far as I can see here.

You can see where the logic for `automatically_declare_parameters_from_overrides` avoids re-declaring here:

https://github.com/ros2/rclcpp/blob/1b28f389c28c7cd832c17599353808ac019f409d/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp#L103-L115

livanov93, 2021-08-18:
> I think that if you trace the source of the exception, it will be this line:
>
> https://github.com/livanov93/test_params_pkg/blob/6c856e28ec9b69850517172c5b25572f5e515d60/src/test_params_node.cpp#L12 [...]

I have found the same lines regarding `node_parameters`, however I expected that the parameters declared within the node constructor should be also avoided while the `automatically_declare_parameters_from_overrides` is true. The misunderstanding came to me from the documentation - `Already declared parameters will not be re-declared`. Usually additional declaration is done within the node itself (usually you don't declare parameters of one node within the other one...) so I immediately concluded it should take care about redeclaration of parameters within the Node's constructor.

Maybe it would be nice to add in the documentation that declaration of the overrides parameters is done before any other declaration within the node so the node creator should be the one to check if the parameter is already declared. 

@wjwwood  Thank you for clarification!

wjwwood, 2021-08-18:
> I expected that the parameters declared within the node constructor should be also avoided while the automatically_declare_parameters_from_overrides is true. 

Yes, I can see why you would think that, but unfortunately it would be kind of messy to implement this, as then we'd need to know why/how a parameter was declared and make declare parameter work differently depending on how the already declared parameter was declared.

Alternatively, the logic to implement `automatically_declare_parameters_from_overrides` is relatively trivial and could also be put into a method on the node class, so that rather than setting `automatically_declare_parameters_from_overrides` to `true` you could do all your declarations in your constructor and at the end call this method to declare the rest of the parameters automatically. We can't do that via a node option just because we have no "tail" call for constructor. The user's derived class constructor is always the last code run.

wjwwood, 2021-08-18:
If you would be kind enough to propose some changes to the documentation, that would be appreciated. Otherwise I'll leave this issue open (or maybe open a new one that's more specific) and put the "help wanted" label on it.

Either way, this issue will hopefully help others that run into this issue in the meantime. Thanks for pointing this out!

wjwwood, 2021-09-14:
I'm gonna close this as "fixed with docs" in https://github.com/ros2/rclcpp/pull/1751.

closed 2021-09-14 by wjwwood