ros2/rclcpp #1762: Allow to undeclare statically typed or read-only parameters from host node

bpwilcox, 2021-09-02:
I had been following the discussions and eventual implementation of the design for enabling statically typed parameters. While I am glad to see this enhancement, one area I see lacking sufficient discussion in the design conversation and the current documentation is the behavior of undeclaring parameters. Let me first start by illustrating a simple user scenario:

There is a node which dynamically loads and unloads plugins, each of which declares its own parameters on the host node. These parameters would appropriately be either read-only or statically typed in order to protect from any undesired modifications. Those parameters will be accessible by other nodes in the system, and when the plugins are dynamically unloaded, we expect that clients of this node should have awareness that the plugins and hence their parameters are gone (e.g. it is not fetching values from a no longer active plugin).  

Currently, a parameter which is either read-only or is statically typed (the default) [cannot be undeclared](https://github.com/ros2/rclcpp/blob/ecb81ef2c35a3e605b7c501ace4ef1baf4755bf9/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp#L555). My initial impression was that this behavior was uninituitive, because to undeclare a parameter seems a different use case than attempting to modify it. Even further, setting a parameter to `rclcpp::PARAMETER_NOT_SET` seems like an _additional_ use case, especially now that [uninitialized parameters are allowed](https://github.com/ros2/rclcpp/pull/1673) again. The main counterpoint I can see is that a parameter could be undeclared and subsequently declared again with a different type or different value (in case of read-only).

**My current proposal** is to allow only the host node to undeclare parameters. In https://github.com/ros2/rclcpp/issues/807, the idea was discussed to allow only the owner node to make modifications to a read-only parameter. Without going so far, I think it would be fair to at least give the host node the authority to undeclare any parameter (whether statically typed or read-only), while blocking remote nodes or command line from undeclaring these parameters (those that are read only or statically typed) via a set parameter request. I see an argument for removing the logic to allow the set_parameter feature to undeclare a parameter as well. Ideally I see a situation where, in case of the above example, a node plugin can declare its parameters on the host node when it is loaded, and undeclare its parameters when it is dynamically unloaded, while still offering its protection against statically typed modifications or read-only changes. The host node is already the arbiter for when a parameter is declared, so it makes sense to me that it should have the same capacity to undeclare them. This may be a particularly prevalent design pattern in lifecycle nodes, where system cycles may be regularly performed, and it may make more sense to declare/undeclare parameters than to continuosly check if a parameter has already been declared in order to avoid an error. 

I appreciate any thoughts or other perspectives on this issue as well. I simpy have seen practical use cases where it would be preferred behavior to undeclare parameters that may still be read only or statically typed.

ivanpauno, 2021-09-17:
> My current proposal is to allow only the host node to undeclare parameters. In #807, the idea was discussed to allow only the owner node to make modifications to a read-only parameter. Without going [...]

That makes sense to me, maybe we could have a "forced" set and undeclare methods.
The parameter server would use the non-forced ones (i.e. you can only call the "forced" methods locally).

@wjwwood what do you think?

alsora, 2021-11-04:
Hi, having "forced" methods seems a very easy to implement solution that would solve the problem.
@ivanpauno @wjwwood any objection if we start developing this feature?

ivanpauno, 2021-11-04:
sounds good to me

wjwwood, 2021-11-29:
That's good with me, the having methods that are only called locally to remove them.

wjwwood, 2021-11-29:
As I commented on the pull request, I think having a local undeclare that work for read-only is fine, but I think setting them is problematic. If we want forced set methods, I think we need a new concept like `read_only` vs `immutable` or something like that. But those are close to synonyms.

alsora, 2022-02-17:
@wjwwood could you give an estimate of the complexity of implementing the `read_only` vs `immutable` concepts?