ros2/rclcpp #1769: Support loaned message passed by intraprocess

mauropasse, 2021-09-09:
Currently RCLCPP intra-process communication can't handle Loan Messages:
https://github.com/ros2/rclcpp/blob/665e37784a6f1772e0efd2685c10fb0c17fdac96/rclcpp/include/rclcpp/publisher.hpp#L397-L400

After some CPU profiling done around Loaned Messages shown in https://github.com/ros2/rclcpp/issues/1642#issuecomment-916078459 I can tell that is not trivial to improve significantly the performances (CPU, latency). For small message passing, RCLCPP intra-process still performs at least twice as better. This leads me to the question of how much we could improve the use of Loaned Messages if they were supported by the intra-process manager, which will bring probably some benefits.

@Karsten1987 is there any updates on a plan to support Loaned Messages with intra-process communication? 
@alsora @wjwwood

wjwwood, 2021-11-29:
There is no update.

esteve, 2022-11-28:
@wjwwood is this a limitation by design or just that it hasn't been implemented yet? If the latter, I can look into it, any guidance? Thanks.

esteve, 2022-12-01:
I'm working on this one.

gilduz, 2022-12-06:
I've just received the message: ""storing loaned messages in intra process is not supported yet"
For my application, it's mandatory to maintain some nodes in the same process, while other in separate processes. Support to intraprocess loaning would be great.

SebastianMauchly, 2023-05-25:
DDS vendors stepped up to design systems that make it possible to share huge messages between nodes in a true zero-copy way: first CycloneDDS via Iceoryx and then FastDDS. Then, ROS2 maintainers stepped up to create the loaning API so that this functionality could be used by ROS developers with minimum change to their publisher / subscriber callsites, and without breaking the encapsulation of which DDS vendor they were using. And in the background, we had ROS maintainers that developed the notion of Components so that the choice of whether your nodes run in the same process or separate ones is a runtime decision. Put together, this makes it so that ROS users can finally (finally!) not care about their message sizes and assume they will be moved in the most efficient way possible. Many of us in industry have been waiting with bated breath for all of those things to come together, because the sum is far greater than its parts.

... except, you fumbled at the goal line, and the loaning API doesn't support intraprocess. Intraprocess! The easiest of all cases, where a simple pointer can just be shared to the other node directly! I'm feeling bitten here, because I've moved mountains within my company to push us towards leveraging ROS2. The promise of totally fungible process layout and worry-free low-cost transport was one of the primary carrots I held up to motivate that transition. This humdinger of a missing feature has made a liar out of me, and I've seen nobody outside of this quiet github issue mention it.

Is there something I'm missing? All of the hard work done to make the process layout and transport mechanism a runtime policy decision is ruined without this feature, no? If this has been the case since 2021, how is it that OSRF didn't identify it and prioritize it for Iron? Is nobody actually going to use the loaning API? @Karsten1987 ? Bueller?

CursedRock17, 2023-07-28:
It seems that issue #2203 explains why loaned messages aren't efficient when used in an intraprocess. If there are DDS vendors that already support the sharing of large messages in zero-copy, why add the extra overhead of the LoanedMessage API?