ros2/rclcpp #1772: Support a bunch of SingleThreadedExecutor with dedicated thread for each component nodes

gezp, 2021-09-14:
## Feature request

#### Feature description

Hi, guys

I'm working on project [Reduce ROS2 Nodes and Determinism](https://navigation.ros.org/2021summerOfCode/projects/spinners.html) of OSPP2021 (under the mentor of Steve Macenski).

Currently, i'm trying to support composed bringup for Nav2 stack, and i have done some works about manually composed bringup (Compile-time composition) of Nav2, which performs better than Normal bringup , in addition, i find a problem that  `a large multi-threaded executor` consumes higher cpu(increase 30%-50%) than `a bunch of single-threaded executors` , you can find some details here: https://discourse.ros.org/t/nav2-composition/22175. 

for `Run-time composition` , i notice that `rclcpp_components` only provides `component_container.cpp` which use `SingleThreadedExecutor` for all component nodes in container and `component_container_mt.cpp` which use `MultiThreadedExecutor`.  

how about supporting a bunch of `SinglethreadedExecutor` for component container?  in other word, create a `SingleThreadedExecutor` with dedicated thread for each component nodes.

#### Implementation considerations

in my opinion, we should spin container node(`ComponentManager`) and components nodes separately, in other word, we need create another thread and another executor to spin components nodes in `ComponentManager`. then we could provide a general approach so that users could choose different executor mode according to ros parameters of container 

* `use_multi_executors`
  * if true, use a bunch of SingleThreadedExecutor with dedicated thread for each component nodes.
  * if false, use a single Executor for component nodes.
* `use_multi_threads`  (when use a single Executor)
  * if true, use a SingleThreadedExecutor  for component nodes.
  * if false, use a MultiThreadedExecutor for component nodes.

any discussion is welcome.

SteveMacenski, 2021-09-15:
@gezp please propose the simplified design we chatted about to the maintainers. I'm sure just the parameter + simplified needs we directly have can be accommodated without much concern.

gezp, 2021-09-15:
@SteveMacenski , i have cleaned this ticket, and open a new ticket for another topic: https://github.com/ros2/rclcpp/issues/1774.

SteveMacenski, 2021-09-15:
Should this be closed then?

closed 2021-09-16 by gezp