ros2/rclcpp #1774: use dedicated executor for component nodes

gezp, 2021-09-15:
## Feature request

#### Feature description

currently, `component_container` in `rclcpp_components` uses a shared executor for container itself and all of component nodes.

in my opinion, it's clearly and better that component nodes use a dedicated executor with dedicated thread, and the executor in `main()` should be only used by container itself (e.g.  load/unload component service).

what do you think about it ?

#### Implementation considerations

in order to backward compatible, we could add a parameter `dedicated_executor` in `ComponentManager` .

* If false (by default) does the current behavior that they all share one. 
* If true, then each node instantiated should have its own executor to use.

gezp, 2021-09-15:
with this feature, we can merge `component_container` and `component_container_mt`,  and we can select `SinglethreadedExecutor` or`MultiThreadedExecutor`  for component nodes by parameters in `ComponentManager`

SteveMacenski, 2021-09-15:
This should be a parameterized option with the defaults for the current behavior of a single single-threaded executor. If a parameter, for instance `dedicated_thread` is set to `true`, when a new component is added, it is added to its own executor instead of a single one for all components. 

I don't think this is clear that that is the objective best solution, but it is a useful option to make available based on the needs and discussions in https://discourse.ros.org/t/nav2-composition/22175. In that ticket, we bring up analysis showing that N-single threaded executors are far more computationally efficient than a single N-multithreaded executor. We'd like to be able to have this N-single threaded container option available to us so we can do dynamic composed bring up in Nav2 rather than having to manually compose them in a process. That's not very flexible and configurable and afterall Nav2 is a framework rather than a standalone solution.

We do not suggest any changes to the multi-threaded executor container, nor are we suggesting to merge them into a single component manager, that would be feature creep and it makes sense for those to be separate executables. 

What we do suggest is the following:
- Add a parameter, such as `dedicated_thread` for the container to load the components into a single executor if false and into individual dedicated executor threads if true. Default to false for current behavior. 
- On the loading of component nodes, use that value to either create a new executor to add the node to or add it to the single main executor
- On unloading of component nodes, if the parameter is true, we will remove that node from the executor and delete it along with the component

fujitatomoya, 2021-09-17:
thanks for the information 👍 

sorry for answering questions with question, https://discourse.ros.org/t/nav2-composition/22175 `start navigation` is result based on the same workload in the same time window with the same number of threads? (for `MultiThreadedExecutor` and `Multiple SingleThreadedExecutor`)

for example, application has a workload to finish.

| Executor | # of threads | CPU | Elapsed Time |
|---|---|---|---|
| MultiThreadedExecutor | 12 | 43% | 0.75T seconds |
| Multiple SingleThreadedExecutor | 9 | 34% | T seconds |

- 12 comes from `std::thread::hardware_concurrency` in the test environment described in https://discourse.ros.org/t/nav2-composition/22175.
- 9 comes from https://github.com/gezp/navigation2/blob/49bfd34c985a2a0928a87d91cd01cc28fba44038/nav2_composition/src/main.cpp#L81-L89

if the result is something like above, i do not see the difference in performance. for me, it is just trading-off.
in this case, maybe we want to have an ability to configure the number of threads for `MultiThreadedExecutor`? (something like https://github.com/ros2/rclcpp/pull/1708?)

what do you think? if i am missing anything, please let me know!

gezp, 2021-09-18:
i make a [simple package](https://github.com/gezp/ros2_topic_performance) to test.  i create a publisher and subscriber on a topic with message `geometry_msgs::msg::PolygonStamped` (the num of points is: 10000, publishing rate is 100hz) for three cases, and get [result here](https://github.com/gezp/ros2_topic_performance/blob/main/test_result/test_use_dedicated_executors.txt): 
```
case 1: standalone publisher and standalone subscriber
cpu: 7.4 memory: 0.09968215868648357
case 2: composed publisher and composed subscriber with a multi-threaded-executor
cpu: 7.8 memory: 0.0470663710230197
case 3: composed publisher and composed subscriber with multiple single-threaded-executor
cpu: 6.2 memory: 0.04695660234221972
```
*  the thread num of multi-threaded-executor equals to the num of nodes, in this case, thread num = 2

you could reproduce similar result by running `python scripts/test_use_dedicated_executors.py` in [this package](https://github.com/gezp/ros2_topic_performance) 

composition with a multi-threaded-executor (thread num=2) consumes higher cpu than normal standalone case, which seems like a bit strange.
and composition with multiple single-threaded-executor performs better than normal standalone case.

fujitatomoya, 2021-09-19:
@gezp here is what i got on my local environment.

```
case 1: standalone publisher and standalone subscriber
[psutil.Process(pid=1275415, name='standalone_publisher', started='23:09:04')]
[psutil.Process(pid=1275417, name='standalone_subscriber', started='23:09:04')]
cpu: 25.0 memory: 0.1808959626374977
case 2: composed publisher and composed subscriber with a multi-threaded-executor
[psutil.Process(pid=1275442, name='composed_npub_nsub', started='23:09:25')]
cpu: 27.9 memory: 0.09321206907146806
case 3: composed publisher and composed subscriber with multiple single-threaded-executor
[psutil.Process(pid=1275465, name='composed_npub_nsub', started='23:09:36')]
cpu: 21.0 memory: 0.09055736496036379
```

I also confirmed that all messages from publisher are received on subscription. after all, i think `MultiThreadedExecutor` is not performative compared to `Multiple SingleThreadedExecutor` in this case.

btw, https://github.com/gezp/ros2_topic_performance needs to be updated as below to build.

```
diff --git a/package.xml b/package.xml
index 687383c..4250df7 100644
--- a/package.xml
+++ b/package.xml
@@ -10,14 +10,18 @@

   <build_depend>example_interfaces</build_depend>
   <build_depend>rclcpp</build_depend>
+  <build_depend>rclcpp_components</build_depend>
   <build_depend>rcutils</build_depend>
   <build_depend>std_msgs</build_depend>
+  <build_depend>geometry_msgs</build_depend>

   <exec_depend>example_interfaces</exec_depend>
   <exec_depend>launch_ros</exec_depend>
   <exec_depend>rclcpp</exec_depend>
+  <exec_depend>rclcpp_components</exec_depend>
   <exec_depend>rcutils</exec_depend>
   <exec_depend>std_msgs</exec_depend>
+  <exec_depend>geometry_msgs</exec_depend>

   <test_depend>ament_cmake_pytest</test_depend>
   <test_depend>ament_lint_auto</test_depend>
```

I think that having option `dedicated executor` makes sense for now. (personally i would like to figure out and improve `MultiThreadedExecutor` in the future.) @ivanpauno @wjwwood @clalancette any opinion?

SteveMacenski, 2021-09-20:
I wouldn’t argue with a better multithreaded executor as an alternative, but what’s elegant about the changes requested is that they would impact both single and multi threaded executor containers to offer an “executor per component” behavior for complex systems needing that behavior for single and/or multi threaded needs. 

So I think the best answer is not either-or, but _both_!

fujitatomoya, 2021-09-22:
@SteveMacenski we had a quick discussion on this in MW WG, i think that it is okay to have this option.

gezp, 2021-09-23:
i try to implement this feature here https://github.com/gezp/rclcpp/commit/3d48c0aacdc62c1ab688d1147679157ad578927f, what do you think about it?

fujitatomoya, 2021-09-23:
@gezp thanks for the contribution ! could you make PR, so that we can start review.

closed 2022-05-26 by gezp

ZhenshengLee, 2022-11-08:
Is this feature supported through launch file? 

Thanks.

SteveMacenski, 2022-11-08:
Yes, you can select this contain like any others