ros2/rclcpp #1778: Spinning node does not also spin sub nodes

jlack1987, 2021-09-16:
## Bug report
I'm assuming this is a bug so i'm filing as so but if this is intended behavior then perhaps this is a feature request.
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - foxy
- DDS implementation:
  - whatever is default
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Simply create sub nodes and topics/services with those sub nodes and only spin the parent node.

#### Expected behavior
I expected spinning the top level node would also spin the subordinate nodes

#### Actual behavior
Unless the subordinate nodes are spun individually they do not get spun

jlack1987, 2021-09-16:
Ultimately the use case that i'm trying to understand how to handle is that i'm converting a relatively large chunk of code from ros1 to ros2 it involves lots of classes who own classes who own classes and so on and so on. In ros1, each of these classes had their own node handles which extended the namespace where they all could put their topics/services/params. 

From the main thread I could simply init the node, create the top level class(or pointer to it), give it the top level nodehandle, and send it on its way ticking periodically and the main thread would periodically call `ros::spinOnce()` at whatever rate I want ros stuff to happen.

In attempting to recreate this type of behavior in ros2 it seemed like sub nodes were the way to go and each node could pass a sub node down the branch of classes. But I then found that i'd still have to spin each sub node separately which would mean hundreds of execution threads bc I can't spin nodes from my main RT thread. I then found the Executors but attempting to add sub nodes to an executor results in,

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Node has already been added to an executor.
```

So the only thing I can think of is to have every class manually create a new non-subordinate node and pass a pointer to an executor around and things can add their nodes to that and have the top level spin the executor. I hope this makes sense and please let me know if I have something incorrect

fujitatomoya, 2021-09-20:
as far as i know, subordinate nodes use shared object from 'rclcpp::Node'. So when 'rclcpp::Node' spins, that means subordinate nodes spin. i confirmed with your sample program with 'ros2 topic info -v', 'ros2 topic echo' and 'ros2 service call'.

> terminate called after throwing an instance of 'std::runtime_error'
>   what():  Node has already been added to an executor.

this is expected because subordinate nodes are on the same executor with main 'rclcpp::Node'.

jlack1987, 2021-09-20:
Ok, I guess then I don't understand what's going on in my example then. When I run it I see,

```
✘  16:24 ~/nautilus_ws # ros2 topic list
/parameter_events
/rosout
/top_level/top_level/a
/top_level/top_level/b
```

but I am expecting to see

```
/top_level/one/a
/top_level/one/b
/top_level/two/a
/top_level/two/b
  .
  .
  .
/top_level/ten/a
/top_level/ten/b
```
Are you saying when you run it you see all of the topics?

fujitatomoya, 2021-09-20:
> Are you saying when you run it you see all of the topics?

No, i did not mean that. I don't see those topics either. Just saying that sub-nodes are spinning when main node spins. but i understand what you mean. i am not sure original design about the subordinate nodes. any help?

aprotyas, 2021-09-20:
From https://github.com/ros2/rclcpp/issues/731:
> Currently how parameters and "sub-nodes" interact is not documented and is not consistent or tested. That needs to be documented, audited, and tested.
Looks like there's some WIP but that issue is also old, so I'm not sure if things changed in between.

I agree that the expected behavior should be for sub-nodes to also spin, but I don't find it documented anywhere in [`rclcpp::Node::create_sub_node`](https://docs.ros2.org/galactic/api/rclcpp/classrclcpp_1_1Node.html#a1df8e4507d17a503630317432449d69d) or [`rclcpp::Executor::spin_some`](https://docs.ros2.org/galactic/api/rclcpp/classrclcpp_1_1Executor.html#ae821cd4cf1040fe2da521fb19a458fa1).

Also, it looks like the `spin_some` function only adds the passed-in node (and not any of it's sub-nodes) to an executor. https://github.com/ros2/rclcpp/blob/2801553d61c5a30a0327d5cbc8d28bcd74e9703d/rclcpp/src/rclcpp/executor.cpp#L409-L415

fujitatomoya, 2021-09-20:
@aprotyas nice catch for https://github.com/ros2/rclcpp/issues/731 ❗ thanks!

> I agree that the expected behavior should be for sub-nodes to also spin

they do.

> but I don't find it documented anywhere

👍 for adding document.

> Also, it looks like the spin_some function only adds the passed-in node (and not any of it's sub-nodes) to an executor.

i think `passed-in node` is also sub-node. so that i can spin and take events to execute.

i guess that we might want to change the subject like `it does not extend the namespace of all entities created with subordinate node.`?

aprotyas, 2021-09-20:
> i think `passed-in node` is also sub-node. so that i can spin and take events to execute.

I didn't understand that. Say, from the example code provided by @jlack1987, when `rclcpp::spin_some(node);` is executed, does it also add the sub-nodes to the executor?

fujitatomoya, 2021-09-20:
> I didn't understand that. Say, from the example code provided by @jlack1987, when rclcpp::spin_some(node); is executed, does it also add the sub-nodes to the executor?

yes, it takes events(in this sample service request and send response) and publish messages. you can see service servers are responding (also logging messages) for each sub-nodes once you actually send the requests.

aprotyas, 2021-09-21:
> yes, it takes events(in this sample service request and send response) and publish messages. you can see service servers are responding (also logging messages) for each sub-nodes once you actually send the requests.

Understood!

---

In conclusion, it looks like (quoting @fujitatomoya):

> i guess that we might want to change the subject like `it does not extend the namespace of all entities created with subordinate node.`?

@jlack1987 if you agree with the assessment above, I think the behavior exhibited by `rclcpp` is clarified, and we can close this issue. I will probably submit a PR with some documentation about said spinning behavior.

jlack1987, 2021-09-21:
@aprotyas i'm still not sure I understand. It's still not clear to me why from my example that I don't see the topics from all the sub nodes as I mentioned [here](https://github.com/ros2/rclcpp/issues/1778#issuecomment-923085508). 

It sounds like you all are saying that it does indeed spin the subordinate nodes which is good and in that case I guess my issue is with some other behavior that I haven't yet been able to describe accurately. I'm expecting those topics/services to show up under the sub nodes but they do not. Also if I need to close this one and open a new one with the actual issue better described I can do that as well once I understand what it is :laughing:

aprotyas, 2021-09-21:
@jlack1987 I believe the crux of the issue is captured by https://github.com/ros2/rclcpp/issues/731#issuecomment-640277329 - quoted below:

> So, is this ticket basically tracking that sub nodes do not actually use the sub_namespace in the name of the parameter? For example:
> 
> ```
> auto subnode = parent->create_sub_node("controller_name");
> std::string controller_type = subnode->declare_parameter<std::string>("type", "");
> ```
> 
> I would expect the parameter to be "controller_name/type", but instead "ros2 param list" shows only a parameter named "type"? Is this expected behavior? Is this a bug?

Essentially, sub-nodes do indeed get added to the executor, and they are spinning - so this part matches all of our expected behavior. The behavior that is not consistent with our expectation is that sub-namespaces are not propagated for entities that belong to/were created by the sub-node. That is an acknowledged bug in #731, but I'm not sure if any progress has been made on that front.

jlack1987, 2021-09-21:
oooh ok I see what you mean. Thanks for the clarification. In that case yeah i'll go ahead and close this out and keep an eye on that issue

closed 2021-09-21 by jlack1987