ros2/rclcpp #1780: NodeOptions is not a member of rclcpp

sandhavel, 2021-09-22:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

[image]

ERROR:

"error: ‘NodeOptions’ is not a member of ‘rclcpp’
       rclcpp::NodeOptions().use_intra_process_comms(intra_process_comms))"

clalancette, 2021-09-22:
What steps did you use to checkout ros2, and what branch are you on?  It looks like you are either on the wrong branch, or haven't sourced the underlay with rclcpp properly.

sandhavel, 2021-09-22:
> What steps did you use to checkout ros2, and what branch are you on? It looks like you are either on the wrong branch, or haven't sourced the underlay with rclcpp properly.

wget https://raw.githubusercontent.com/ros2/ros2/foxy/ros2.repos

clalancette, 2021-09-22:
> wget https://raw.githubusercontent.com/ros2/ros2/foxy/ros2.repos

Sorry, you'll need to provide more information.  We know this works for us (we run it nightly in CI), so there is something about your environment that is different.

sandhavel, 2021-09-22:
https://docs.ros.org/en/foxy/Installation/Ubuntu-Development-Setup.html

I follow the steps, which is mentioned in above link, when i run colcon build --symlinnk-install, that error occurs

sandhavel, 2021-09-22:
even i used  https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos ,
the same error occurs...

closed 2021-09-23 by sandhavel