ros2/rclcpp #1786: error: ‘NodeOptions’ is not a member of ‘rclcpp’

sandhavel, 2021-09-27:
[image]

## Bug report
When i run the " colcon build  --symlink-install " command, i face this issue

**Required Info:**

- Operating System:
  OS - Ubuntu 18.04

#### Expected behavior
Successfully build all the packages

#### Actual behavior
Error at  lifecycle package, like mentioned in the screenshot.

Note:
I get the source from this repo,
"https://raw.githubusercontent.com/ros2/ros2/foxy/ros2.repos"

Please help me to solve this issue.....

fujitatomoya, 2021-09-27:
I do not have any problem with Ubuntu 20.04. Ubuntu 18.04 is officially supported platform for Foxy?

https://docs.ros.org/en/foxy/Installation/Ubuntu-Development-Setup.html
Tier 1: Ubuntu Linux - Focal Fossa (20.04) 64-bit

aprotyas, 2021-09-27:
It looks like you're building against a ROS 2 Crystal distribution in `/opt`, rather than the Foxy source in your workspace. `NodeOptions` did not exist till Dashing (?), so you can't find that class.

Make sure that you aren't sourcing `opt/ros/crystal/setup.bash` either in your `bashrc` file or whenever you start up a shell.

sandhavel, 2021-09-28:
> It looks like you're building against a ROS 2 Crystal installation from `/opt`, rather than the Foxy source in your workspace. `NodeOptions` did not exist till Dashing (?), so you can't find that class.
> 
> Make sure that you aren't sourcing `opt/ros/crystal/setup.bash` either in your `bashrc` file or whenever you start up a shell.

Yeah ok thank you @aprotyas

closed 2021-09-28 by aprotyas