ros2/rclcpp #1819: Logging Macros Error with Pointers

DLu, 2021-11-09:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: Binaries
- Version or commit hash: 13.1.0-1focal.20211101.200103
- DDS implementation: default
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
Create a pointer to a `rclcpp::Logger`

    std::shared_ptr<rclcpp::Logger> logger = std::make_shared<rclcpp::Logger>(node->get_logger());

Then try to log something using the dereferenced logger. 

    RCLCPP_INFO_STREAM(*logger_, "Someday we'll find it, the rainbow connection");

#### Expected behavior

Logged string!

#### Actual behavior
Does not compile. 

```
/home/dlu/path/test.cpp: In member function ‘void fake_package::load()’:
/home/dlu/path/test.cpp:132:3: error: ‘class std::shared_ptr<rclcpp::Logger>’ has no member named ‘get_name’
  132 |   RCLCPP_INFO_STREAM(*logger_, "Someday we'll find it, the rainbow connection");
      |   ^~~~~~~~~~~~~~~~~~
/home/dlu/path/test.cpp:132:3: error: ‘class std::shared_ptr<rclcpp::Logger>’ has no member named ‘get_name’
  132 |   RCLCPP_INFO_STREAM(*logger_, "Someday we'll find it, the rainbow connection");
      |   ^~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/fake_package.dir/build.make:101: CMakeFiles/fake_package.dir/src/test.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/fake_package.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

#### Additional information
Surrounding the first argument with parentheses makes it work. 

    RCLCPP_INFO_STREAM((*logger_), "Someday we'll find it, the rainbow connection");

DLu, 2021-11-09:
Note: I use this pattern occasionally when making plugins and you want latent initialization of the logger.

aprotyas, 2021-11-10:
I think this is what's happening here:
- Preprocessor does verbatim text replacement.
- The macro expansion provided in [rclcpp: include/rclcpp/logging.hpp File Reference](https://docs.ros2.org/latest/api/rclcpp/logging_8hpp.html#aa237b2c561a1c71ea7208f35690fbeb3) means the following expression is evaluated:
```cpp
*logger_.get_name()
```
- Member access operator has higher precedence than dereference operator.

The fix you noted with the parentheses around `*logger_` is the way to go, except it should go in the template generating the logging macros, which I've done in https://github.com/ros2/rclcpp/pull/1820.

@DLu can you check if https://github.com/ros2/rclcpp/pull/1820 fixes the problem for you?

aprotyas, 2021-11-11:
Closing this issue since #1820 has merged. Thank you for the report!

closed 2021-11-11 by aprotyas