ros2/rclcpp #1831: Exiting a subscriber causes subsequent subscribers to no longer receive messages

guru-florida, 2021-11-27:
## Bug report
Exiting a subscriber (ex. "ros2 topic echo /tf") causes subsequent subscribers to no longer receive messages. Pre-existing subscribers continue to receive messages ok. 

The behavior is truly strange. In the attached video I show:
1. Running `ros2 topic echo /tf` subscriber
2. Running a c++ node that publishes /tf data
         (data is then seen in [1] subscriber, no issue)
3. Exiting subscriber in [1]   __(problem seems to arise here)__
4. Running subscriber in [1] again and no data is streamed.
5. Showing Rviz is still running and receiving messages since it was a pre-existing subscriber. Also running two `ros2 topic echo /tf` subscribers and exiting only one of them, the other will continue to see data after issue presents.

I also show that:
1. `ros2 node info <tf_node_publisher>` cannot find node after issue presents, but...
    - node is indeed running and clearly still publishing both /tf and on other topics. _This points to a possibly discovery issue?_
    - `/tf` will show 0 publishers, but pre-existing subscribers are still getting published data (huh?!?!)

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries from foxy
- Version or commit hash:
  - latest repo update
  - deb [arch=amd64] http://packages.ros.org/ros2/ubuntu focal main
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I don't yet have enough information to provide a reliable way to reproduce this problem. 

#### Expected behavior
* Order of publisher/subscriber node startup should not matter
* Existing a subscriber (such as ros2 cli echo) should not trash operational state of another node

#### Actual behavior
* Subscribers started after publisher node receive no messages
* Subscribers started before publisher node do continue to receive messages

#### Video demo
See youtube video of behavior here:
https://www.youtube.com/watch?v=6YKKcrbBR8A
- Intro to issue ends @2:00
- Plot thickens @4:00
- Somewhat of a conclusion @9:00
 
#### Additional information
* I also encountered startup issues with another rclpy-based python node in Aug/Sept. This required restarting all nodes, waiting a minute for `ros2 node list` to clear out and relaunching the system. This started after ros2 repo updates. It may be related since it seemed to be a similar node discovery issue.
* It's possible this issue is something in my c++ node, but given the circumstances and evidence so far I suspect a recent code change to rclcpp or FastRTPS library. Rclcpp/FastRTPS has been practically flawless since 2019 when I started using it until recently when I started seeing this weird behavior after some upgrades.
* I am going to try Connext middleware and see if the problem persists

Any other suggestions to investigate further?

guru-florida, 2021-11-29:
update: Moving to galactic works as expected with no code changes. Using cyclonedds on Galactic. 

I tried using rmw-connext-cpp on foxy but I couldn't get it to work. It would return "Failure to create participant" error.

Barry-Xu-2018, 2021-11-29:
@guru-florida 

> update: Moving to galactic works as expected with no code changes. Using cyclonedds on Galactic.  

Could you try to use Fastrtps (fastdds) on galatic since you use fastrtps on foxy ?   
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp and run your test again.

guru-florida, 2021-11-29:
@Barry-Xu-2018 Yes, seems to work fine on galactic with rmw_fastrtps_cpp as well.

fujitatomoya, 2021-11-29:
this seems to be related to rmw_fastrtps (specifically eProsima Fast-RTPS discovery).
i am not really sure which fix ca solve this problem, but probably better to backport?

btw, we could transfer issue to https://github.com/ros2/rmw_fastrtps.

guru-florida, 2021-11-30:
@fujitatomoya I suspect it is in the fastrtps middleware as well, particularly discovery. I thought about posting the issue there but figured I should move down the chain one by one and get input from others before jumping to there. If the fastrtps in Galactic can be backported to foxy I can retest. If you want to transfer the issue I am good with that too.

clalancette, 2022-02-04:
@MiguelCompany Any thoughts here?

MiguelCompany, 2022-02-07:
@guru-florida @clalancette I would first check if this is fixed with ROS 2 Foxy built from sources, as branch 2.1.x has some backports that may affect this (eProsima/Fast-DDS#1864 is the one I think could be solving this issue)