ros2/rclcpp #1837: Support loaned message for serialized data

Barry-Xu-2018, 2021-12-07:
rosbag2 sends serialized data by rclcpp::GenericPublisher.   
Now there is no interface to use loaned message for serialized data (DDS not support it).   
In order to use current loaned message implementation, whether we can add new interface to rclcpp::GenericPublisher ?   

e.g.
```c++  
void GenericPublisher::try_publish_loaned_msg(const rclcpp::SerializedMessage & message)
{
    if (this->can_loan_messages()) {
      rcl_borrow_loaned_message(.., loaned_msg);
      rmw_deserialize(message.get_rcl_serialized_message(), .., loaned_msg);
      rcl_publish_loaned_message(..., loaned_msg, null);
    } else {
      publish(message); // Call original publish()
    }
}
```
Any thoughts?

fujitatomoya, 2021-12-07:
@ivanpauno @wjwwood @jacobperron 
CC: @emersonknapp 

this is `ros2 bag play` to enable `LoanedMessage` for zero copy enhancement. would you share your thoughts about this?

emersonknapp, 2021-12-08:
I agree with the approach you are taking. I think that this new interface would be very useful, and it's good to put it in `rclcpp` so that others could use it too. It will be important to document that deserialization happens within this function, so the user should be aware of that for performance.

fujitatomoya, 2021-12-08:
@emersonknapp thanks for the comment.

> It will be important to document that deserialization happens within this function, so the user should be aware of that for performance.

agree 👍

fujitatomoya, 2021-12-22:
@ivanpauno @wjwwood friendly ping, what would you think about this?

fujitatomoya, 2022-01-05:
@ivanpauno @wjwwood @jacobperron friendly ping.

jacobperron, 2022-01-05:
IMO, I would just throw an error if the publisher doesn't support loaned messages. Otherwise, it's not clear exactly what the call will do unless the user calls `can_loan_messages()`. I imagine usage would look like this:

```c++
if (publisher->can_loan_messages()) {
    publisher->publish_loaned_message(message);
} else {
    publisher->publish(message);
}
```

It just seems more clear from the outside what is happening. Than having the `can_loan_messages()` check inside the publish method.

We could consider naming the method such that it's clear that deserializing is happening. E.g. `publish_deserialized_and_loaned()`

wjwwood, 2022-02-17:
I agree with @jacobperron, have it throw and let callers do the try if they want.

closed 2022-03-28 by fujitatomoya