ros2/rclcpp #1846: Allow LifecycleNode to automatically transition Lifecycle-enabled ROS2 entities

bpwilcox, 2021-12-14:
This issue is a question/enhancement regarding the behavior of lifecycle nodes and their counterpart lifecycle entities (`LifecyclePublisher` and future lifecycle-enabled versions of subscriptions, services, action servers, etc.)

First, from the managed nodes design [doc](https://design.ros2.org/articles/node_lifecycle.html):

> Primary State: Active
> This is the main state of the node’s life cycle. While in this state, the node performs any processing, responds to service requests, reads and processes data, produces output, etc.

Similarly with other state guidelines in the documentation. the design outlines the expected behavior of a node in particular states. For the active state, all ROS communication is expected to be active and processing data. As it relates to ROS2 entities such as publishers, subscriptions, services, etc., this implies that their behavior would be in alignment with the lifecycle state of the node. To date, there only exists one lifecycle entity, the [LifecyclePublisher](https://github.com/ros2/rclcpp/blob/master/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_publisher.hpp), however with the introduction of a `LifecycleService` in #1838 / #1836, and the potential for lifecycle versions of subscriptions, action servers and more, the code is beginning to align better with the design documentation.

As of today, a LifecyclePublisher must be manually managed to activate/deactivate its state in a node. With the addition of new lifecycle entities to match the intended design, this management will only grow in scale/complexity as users will have to micromanage all entities for lifecycle transitions. **My proposal in this issue** is that instead of requiring user management of lifecycle entity activation/deactivation, **allow the lifecycle node to automatically transition the state of its lifecycle entities (e.g. lifecycle publishers, subscriptions, services, etc.).** Ignoring implementation considerations for now, this feature would ensure that **lifecycle entity states (active/inactive) align with the lifecycle node**, which already seems to be the intended design according to the documentation. This distinctly distinguishes lifecycle entities from the standard ones, since developers who don't anticipate to align their entities with lifecycle states would likely be well served by standard ones instead.

### Benefits
- Reduce user micromanaging of lifecycle entities as components scale. 
- Default behavior to that of intended best practice as recommended by managed node design documentation
- Implicit insight into state of lifecycle entities via query of lifecycle node state (e.g. an inactive lifecycle node also implies that lifecycle entities are inactive)

I welcome feedback discussing the merits or counterpoints to having this be the default behavior pattern for lifecycle nodes and its lifecycle entities.

fujitatomoya, 2021-12-14:
@bpwilcox thanks for opening discussion.

> allow the lifecycle node to automatically transition the state of its lifecycle entities (e.g. lifecycle publishers, subscriptions, services, etc.)

as i mentioned https://github.com/ros2/rclcpp/pull/1836#pullrequestreview-829697665, i believe that this is the intension of current design. but the design is considered long time ago 2015, so we might be missing something.

friendly ping: @wjwwood @clalancette @ivanpauno

ivanpauno, 2021-12-15:
:+1: to the proposal, manually doing that doesn't seem to make sense.

If all the entities would use the same interface (instead of one lifecycle interface for each type of entity, like [here](https://github.com/ros2/rclcpp/blob/2e4c97ac5616fb024ff6fc6544742be3ba77c6b1/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_publisher.hpp#L36-L43)), we could store all those entities in a vector and the node will redirect the `on_configure`/etc calls when the transition happens.

ivanpauno, 2021-12-15:
@alsora sorry we run out of time in the middleware working group meeting today.
If you want we can start discussing the topic here.

ivanpauno, 2021-12-15:
The only thing I don't see automatically happening is the `on_configure/on_cleanup` part, that will still need to be written in the node.

See e.g. https://github.com/ros2/demos/blob/master/lifecycle/src/lifecycle_talker.cpp.

ivanpauno, 2021-12-15:
I'm working on implementing managed nodes in rclpy, and I took this feedback into account when implementing it.
See https://github.com/ros2/rclpy/pull/865.

If you're interested to take a look to the approach I have taken and see if it matches the idea proposed here that would be great.

bpwilcox, 2021-12-17:
> The only thing I don't see automatically happening is the `on_configure/on_cleanup` part, that will still need to be written in the node.
> 

I agree. The user may want/need to control the creation/destruction of lifecycle entities in those transitions, however management of the entity state would be automatic (active/inactive).

bpwilcox, 2021-12-17:
> I'm working on implementing managed nodes in rclpy, and I took this feedback into account when implementing it. See [ros2/rclpy#865](https://github.com/ros2/rclpy/pull/865).
> 

I think what you've implemented in the draft is in the right direction of what I was thinking. I also agree to having all the entities share the same interface.

ivanpauno, 2022-01-06:
> I think what you've implemented in the draft is in the right direction of what I was thinking. I also agree to having all the entities share the same interface.

Sounds good, I'm planning to implement something equivalent soon in rclcpp, so the rclpy and rclcpp API are more similar.
I think that I can do that without breaking existing code, the idea is that the default implementation of on_activate()/on_deactivate() will automatically make the transition of managed subentities, but when the methods are overriden no automatic transition happens (of course, with the exception of when the parent class method is explicetly being called),

jacobperron, 2022-01-06:
FWIW, I'm not a user of `rclcpp_lifecycle`, but the proposal sounds good to me :+1:

fujitatomoya, 2022-01-12:
there is a discussion at MW WG meeting. i would like to confirm the behavior of lifecycle managed entities as design 1st.

- `uncofigured` state, all user entities in this lifecycle node does not exist. in other words, other nodes cannot even see those lifecycle entities such publisher, subscription, service so on? once it transits into `configured` state, those user entities are created.
- `inactive` state, all user entities exist but not processing. (i think we have the same understanding on this) but we would want to keep the messages in the entity queue or just discard everything during this state? related to [this comment](https://github.com/ros2/rclcpp/pull/1836#discussion_r779742223). I was inclined to keep all the messages in the queue (just relies on rmw implementation queue), but dispose all messages during `inactive` state also makes sense as expected behavior. probably we can provide configurable setting for this?

i may be missing something...

jacobperron, 2022-01-12:
@fujitatomoya I agree with your interpretation of `unconfigured` and `inactive` states; it seems to agree with the design doc.

It is documented that the main purpose of the `inactive` state is to (re)configure pub/subs, etc:

> The main purpose of this state is to allow a node to be (re-)configured (changing configuration parameters, adding and removing topic publications/subscriptions, etc) without altering its behavior while it is running.

So, I think when we enter/leave the `inactive` state (ie configure or unconfigure) is when we would create/destroy ROS entities like pub/sub.

For the question of what to do with message queues, the design doc does state that the policy will be determined by the QoS:

> Data retention will be subject to the configured QoS policy for the topic.

Though, I think it would be reasonable to also have an explicit option for lifecycle nodes to eliminate the queue or not.

Also, the design documents that service requests should be rejected immediately (which makes sense to me):

> Any managed service requests to a node in the inactive state will not be answered (to the caller, they will fail immediately).

closed 2022-01-14 by ivanpauno