ros2/rclcpp #1852: [enhancement] Add ROS_MULTITHREADING=OFF option

allsey87, 2022-01-03:
## Feature request

#### Feature description
The idea behind this CMake option would be to enable to compilation of rclcpp and rclpy and their dependencies for systems where pthread support is either non-existent or incomplete. In this configuration, only the single threaded executor would be available.

#### Implementation considerations
I am still studying the relevant code, however, I suspect that this could be achieved using conditional compilation in the preprocessor, e.g.,
```c++
#ifdef ROS_MULTITHREADING
   std::mutex some_mutex;
   std::lock_guard lock(some_mutex);
#endif // no need to lock if not multithreading
```
and in CMake to drop the `REQUIRED` from `find_package(Threads REQUIRED)`. In fact, if threads are not found, perhaps such a variable could automatically be set?

fujitatomoya, 2022-01-03:
is this feature supposed to be only for `rclcpp` and `rclpy`?

> systems where pthread support is either non-existent or incomplete.

i think we need to consider if this is supported platform, cz most of libraries cannot be supported especially RMW implementations.
according to https://www.ros.org/reps/rep-2000.html#targeted-platforms, `systems where pthread support is either non-existent or incomplete` is not targeted platform, so why we need to have this compile option?

allsey87, 2022-01-03:
> is this feature supposed to be only for `rclcpp` and `rclpy`?

In my mind, it would be applied to `rclcpp`, `rclpy` and packages they depend on, e.g., `rcutils`, `rcpputils`, `rmw_implementation`, etc.

While it is true that the targeted platforms will almost certainly have pthread available, I would consider such an option as being in the same category as being able to set `BUILD_SHARED_LIB=OFF`. It's not strictly necessary for the targeted platforms, but it does make 3rd party efforts to port and adapt the code to embedded environments easier.

fujitatomoya, 2022-01-03:
thanks for your comments.

> I would consider such an option as being in the same category as being able to set BUILD_SHARED_LIB=OFF

for me, not really see in that way. i think linking and compiling are different process.

> but it does make 3rd party efforts to port and adapt the code to embedded environments easier.

understood.
just fyi, this could be off topic from your original issue but [micro-ROS](https://micro.ros.org/) project could be interesting for you.

clalancette, 2022-01-04:
Yeah, I have to agree that this seems infeasible in rclcpp and rclpy.  Both of them heavily rely on threading, and we'd have to have many hundreds of the `#ifdef` section to make this work.  That makes the code really hard to read.

As @fujitatomoya mentioned, the microROS project (and rclc) may be more of what you are looking for.  If not, maybe you could explain your exact use case and we can see if there are other possible solutions here.

closed 2022-01-04 by allsey87