ros2/rclcpp #1861: Get exception "Resource deadlock avoided" while running TestTimeSource.clock_sleep_until_with_ros_time_basic

Barry-Xu-2018, 2022-01-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- Version or commit hash:
  - 6b321edb4fabcc57f666f77a4dc091e95948070b
- DDS implementation:
  - cyclonedds
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Find this issue in CI execution  https://build.ros2.org/job/Rpr__rclcpp__ubuntu_focal_amd64/1636/testReport/projectroot.test/rclcpp/test_time_source/

Use this way also can reproduce this problem  
```
$ source install/setup.bash
$ build/rclcpp/test/rclcpp/test_time_source --gtest_filter=*clock_sleep_until_with_ros_time_basic --gtest_repeat=10000
```
...
Repeating all tests (iteration 9304) . . .

Note: Google Test filter = *clock_sleep_until_with_ros_time_basic*
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from TestTimeSource
[ RUN      ] TestTimeSource.clock_sleep_until_with_ros_time_basic
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
Aborted (core dumped)
```

#### Expected behavior

No exception

#### Actual behavior

Unexpected exception occurs

fujitatomoya, 2022-01-11:
Does this happen with rmw_fastrtps with mainline?

Barry-Xu-2018, 2022-01-11:
> Does this happen with rmw_fastrtps with mainline?

Also happen after checking.

Barry-Xu-2018, 2022-01-11:
Backstrace info based on coredump file   
[gdb_coredump.log](https://github.com/ros2/rclcpp/files/7844930/gdb_coredump.log)

Barry-Xu-2018, 2022-01-14:
After checking codes, I find the possible cause

In test codes
https://github.com/ros2/rclcpp/blob/80f93d1dbbe8398524ceec637f50ce2e3f1e9103/rclcpp/test/rclcpp/test_time_source.cpp#L751-L755

NodeState will create a thread (new executor) to  run a subscriber to receive clock message. 

https://github.com/ros2/rclcpp/blob/80f93d1dbbe8398524ceec637f50ce2e3f1e9103/rclcpp/src/rclcpp/time_source.cpp#L419-L433

While destroying TimeSource (maybe internal NodeState is being destroyed), above callback is called.   

According to coredump info, the problem occurs while getting shared_ptr of NodeState from the weak_ptr of NodeState (code is `auto state_ptr = state.lock()`).   
It leads to destroy NodeState. I don't understand why lock() can cause this.  If you know it, please share with me. 
```
#17 0x00007f0f319ff6f6 in std::shared_ptr<rclcpp::TimeSource::NodeState>::~shared_ptr (this=0x7f0f167f9e30, __in_chrg=<optimized out>)
    at /usr/include/c++/9/bits/shared_ptr.h:103
#18 0x00007f0f31a008b2 in rclcpp::TimeSource::NodeState::create_clock_sub()::{lambda(std::shared_ptr<rosgraph_msgs::msg::Clock_<std::allocator<void> > const>)#2}::operator()(std::shared_ptr<rosgraph_msgs::msg::Clock_<std::allocator<void> > const>) const (__closure=0x55ea11be9bc0, msg=std::shared_ptr<const struct rosgraph_msgs::msg::Clock_<std::allocator<void> >> (use count 4, weak count 0) = {...}) at /tmp/ros2-test/src/ros2/rclcpp/rclcpp/src/rclcpp/time_source.cpp:426
```

In order to make sure clock subscriber thread is terminated while destroying TimeSource, I modify code as below 
```
TimeSource::~TimeSource()
{
  // Make sure clock subscriber thread is terminated at this time
  node_state_->detachNode();
}
```
https://github.com/Barry-Xu-2018/rclcpp/commit/d00f495c10268b6c9d571024c5aa8852e70885b3   

After modification, I re-run test many times (> 60000). This problem doesn't occur.

Barry-Xu-2018, 2022-01-17:
@clalancette is refactoring relevant codes.  
https://github.com/clalancette/rclcpp/tree/clalancette/time-source-cleanups

It can fix this issue and it is better solution.  So I will close my fixing #1865.

PR #1867 (2022-01-20): Cleanup time source object lifetimes
Previously, we were managing a bunch of internal objects with shared_ptr/weak_ptr.  But this was an unnecessary abstraction, and also lead to flaky tests as in #1861 .

Instead, simplify this quite a lot and just use normal C++ object lifetimes.  This is easier to understand, is less code, and also fixes the flaky test.

Supersedes #1865.

Fixes #1861 

@Barry-Xu-2018 @ralph-lange FYI.
closed 2022-01-20 by clalancette