ros2/rclcpp #1866: Component container fail unexpected

ghuan4, 2022-01-14:
I encounter this problem/bug with the component CLI not sure what cause the issue so the title may not be correct.

## Bug report

- Operating System: 
  - Ubuntu 20.04
- Installation type: 
  - binaries
- Version or commit hash:
  - foxy
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
The issue can be reproduce with the demo code of rclcpp_component. When execute following command:
1st terminal: `ros2 run rclcpp_components component_container`
2nd terminal:
```
ros2 component load /ComponentManager demo_nodes_cpp demo_nodes_cpp::Talker
ros2 component unload /ComponentManager 1
ros2 component load /ComponentManager demo_nodes_cpp demo_nodes_cpp::Talker
ros2 component unload /ComponentManager 2
```

#### Expected behavior
The component should load and unload as normal
#### Actual behavior
From the instance the 2nd component load in the `ComponentManager` seem to just fail and all service is unavailable (e.g component list,, load, unload)
#### Additional information
But if skip the first unload then the `ComponentManager` will kept running without issue and all service exist.
Terminal output of both error and working case:
[image]

fujitatomoya, 2022-01-24:
with https://github.com/ros2/ros2/commit/4a36f319afcc208deae3c373493ef6d9feda1324, i cannot reproduce the issue.

```
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component list
/ComponentManager
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component load /ComponentManager demo_nodes_cpp demo_nodes_cpp::Talker
Loaded component 1 into '/ComponentManager' container node as '/talker'
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component list
/ComponentManager
  1  /talker
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component unload /ComponentManager 1
Unloaded component 1 from '/ComponentManager' container node
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component list
/ComponentManager
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component load /ComponentManager demo_nodes_cpp demo_nodes_cpp::Talker
Loaded component 2 into '/ComponentManager' container node as '/talker'
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component list
/ComponentManager
  2  /talker
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component unload /ComponentManager 2
Unloaded component 2 from '/ComponentManager' container node
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component list
/ComponentManager
```

ComponentManger's log.

```
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run rclcpp_components component_container
[INFO] [1643061216.098328013] [ComponentManager]: Load Library: /root/ros2_ws/colcon_ws/install/demo_nodes_cpp/lib/libtopics_library.so
[INFO] [1643061216.101055329] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::Listener>
[INFO] [1643061216.101130779] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::ListenerBestEffort>
[INFO] [1643061216.101158553] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::LoanedMessageTalker>
[INFO] [1643061216.101181811] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::SerializedMessageListener>
[INFO] [1643061216.101204944] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::SerializedMessageTalker>
[INFO] [1643061216.101227611] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::Talker>
[INFO] [1643061216.101249692] [ComponentManager]: Instantiate class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::Talker>
[INFO] [1643061217.105451413] [talker]: Publishing: 'Hello World: 1'
[INFO] [1643061218.105492334] [talker]: Publishing: 'Hello World: 2'
[INFO] [1643061219.105297218] [talker]: Publishing: 'Hello World: 3'
[INFO] [1643061220.105237753] [talker]: Publishing: 'Hello World: 4'
[INFO] [1643061221.105270877] [talker]: Publishing: 'Hello World: 5'
[INFO] [1643061222.105343690] [talker]: Publishing: 'Hello World: 6'
[INFO] [1643061223.105348407] [talker]: Publishing: 'Hello World: 7'
[INFO] [1643061224.105305346] [talker]: Publishing: 'Hello World: 8'
[INFO] [1643061225.105315648] [talker]: Publishing: 'Hello World: 9'
[INFO] [1643061226.105292112] [talker]: Publishing: 'Hello World: 10'
[INFO] [1643061227.105299805] [talker]: Publishing: 'Hello World: 11'
[INFO] [1643061228.105279175] [talker]: Publishing: 'Hello World: 12'
[INFO] [1643061229.105269158] [talker]: Publishing: 'Hello World: 13'
[INFO] [1643061230.105215072] [talker]: Publishing: 'Hello World: 14'
[INFO] [1643061231.105240447] [talker]: Publishing: 'Hello World: 15'
[INFO] [1643061244.753320664] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::Listener>
[INFO] [1643061244.753365262] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::ListenerBestEffort>
[INFO] [1643061244.753373848] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::LoanedMessageTalker>
[INFO] [1643061244.753386572] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::SerializedMessageListener>
[INFO] [1643061244.753398219] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::SerializedMessageTalker>
[INFO] [1643061244.753409482] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::Talker>
[INFO] [1643061244.753421187] [ComponentManager]: Instantiate class: rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::Talker>
[INFO] [1643061245.757330043] [talker]: Publishing: 'Hello World: 1'
[INFO] [1643061246.757337073] [talker]: Publishing: 'Hello World: 2'
[INFO] [1643061247.757309019] [talker]: Publishing: 'Hello World: 3'
[INFO] [1643061248.757284935] [talker]: Publishing: 'Hello World: 4'

```

fujitatomoya, 2022-01-24:
**this problem happens only on `foxy`, but `master` and `galactic`.** but not sure which commit fixes this problem, any thoughts?

-with `foxy`. (docker image ros:foxy && apt upgrade)

```
root@tomoyafujita:~# source /opt/ros/foxy/setup.bash
root@tomoyafujita:~# ros2 component list
/ComponentManager
root@tomoyafujita:~# ros2 component load /ComponentManager demo_nodes_cpp demo_nodes_cpp::Talker
Loaded component 1 into '/ComponentManager' container node as '/talker'
root@tomoyafujita:~# ros2 component list
/ComponentManager
  1  /talker
root@tomoyafujita:~# ros2 component unload /ComponentManager 1
Unloaded component 1 from '/ComponentManager' container node
root@tomoyafujita:~# ros2 component list
/ComponentManager
root@tomoyafujita:~# ros2 component load /ComponentManager demo_nodes_cpp demo_nodes_cpp::Talker
Loaded component 2 into '/ComponentManager' container node as '/talker'
root@tomoyafujita:~# ros2 component list
/ComponentManager
No 'list_nodes' service found when listing components
root@tomoyafujita:~# ros2 component list
/ComponentManager
No 'list_nodes' service found when listing components
root@tomoyafujita:~# ros2 component unload /ComponentManager 2
No 'unload_node' service found for '/ComponentManager' container
```

-with `galactic`. (docker image ros:galactic && apt upgrade)

```
root@263f0153ac57:~# source /opt/ros/galactic/setup.bashg
bash: /opt/ros/galactic/setup.bashg: No such file or directory
root@263f0153ac57:~# source /opt/ros/galactic/setup.bash
root@263f0153ac57:~# ros2 component list
/ComponentManager
root@263f0153ac57:~# ros2 component load /ComponentManager demo_nodes_cpp demo_nodes_cpp::Talker
Loaded component 1 into '/ComponentManager' container node as '/talker'
root@263f0153ac57:~# ros2 component list
/ComponentManager
  1  /talker
root@263f0153ac57:~# ros2 component unload /ComponentManager 1
Unloaded component 1 from '/ComponentManager' container node
root@263f0153ac57:~# ros2 component list
/ComponentManager
root@263f0153ac57:~# ros2 component load /ComponentManager demo_nodes_cpp demo_nodes_cpp::Talker
Loaded component 2 into '/ComponentManager' container node as '/talker'
root@263f0153ac57:~# ros2 component list
/ComponentManager
  2  /talker
root@263f0153ac57:~# ros2 component unload /ComponentManager 2
Unloaded component 2 from '/ComponentManager' container node
root@263f0153ac57:~# ros2 component list
/ComponentManager
```

clalancette, 2022-01-24:
@fujitatomoya What happens if you use `rmw_cyclonedds_cpp` on Foxy, or alternatively `rmw_fastrtps_cpp` on Galactic or Rolling?

fujitatomoya, 2022-01-24:
@clalancette nice catch, with `foxy` and `rmw_cyclonedds`, it just works.

```
root@tomoyafujita:/# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@tomoyafujita:/# ros2 component list
1643063179.207462 [0]       ros2: using network interface enp4s0 (udp/43.135.133.118) selected arbitrarily from: enp4s0, docker0
/ComponentManager
root@tomoyafujita:/# ros2 component load /ComponentManager demo_nodes_cpp demo_nodes_cpp::Talker
1643063185.635297 [0]       ros2: using network interface enp4s0 (udp/43.135.133.118) selected arbitrarily from: enp4s0, docker0
Loaded component 1 into '/ComponentManager' container node as '/talker'
root@tomoyafujita:/# ros2 component list
1643063190.111820 [0]       ros2: using network interface enp4s0 (udp/43.135.133.118) selected arbitrarily from: enp4s0, docker0
/ComponentManager
  1  /talker
root@tomoyafujita:/# ros2 component unload /ComponentManager 1
1643063195.129578 [0]       ros2: using network interface enp4s0 (udp/43.135.133.118) selected arbitrarily from: enp4s0, docker0
Unloaded component 1 from '/ComponentManager' container node
root@tomoyafujita:/# ros2 component list
1643063198.800767 [0]       ros2: using network interface enp4s0 (udp/43.135.133.118) selected arbitrarily from: enp4s0, docker0
/ComponentManager
root@tomoyafujita:/# ros2 component load /ComponentManager demo_nodes_cpp demo_nodes_cpp::Talker
1643063202.879053 [0]       ros2: using network interface enp4s0 (udp/43.135.133.118) selected arbitrarily from: enp4s0, docker0
Loaded component 2 into '/ComponentManager' container node as '/talker'
root@tomoyafujita:/# ros2 component list
1643063205.983254 [0]       ros2: using network interface enp4s0 (udp/43.135.133.118) selected arbitrarily from: enp4s0, docker0
/ComponentManager
  2  /talker
root@tomoyafujita:/# ros2 component unload /ComponentManager 2
1643063211.861358 [0]       ros2: using network interface enp4s0 (udp/43.135.133.118) selected arbitrarily from: enp4s0, docker0
Unloaded component 2 from '/ComponentManager' container node
root@tomoyafujita:/# ros2 component list
1643063215.302212 [0]       ros2: using network interface enp4s0 (udp/43.135.133.118) selected arbitrarily from: enp4s0, docker0
/ComponentManager
```

fujitatomoya, 2022-01-24:
@MiguelCompany do you have thoughts on this?

jacobperron, 2022-02-28:
I can also reproduce the issue with the latest Fast-DDS `2.1.x` branch (https://github.com/eProsima/Fast-DDS/tree/0670fb96255e6862c62f185c34cfd7551bd00b08). 

@MiguelCompany Can you think of a change between 2.1.x and v2.3.1 that might fix this?

jacobperron, 2022-02-28:
Or I suppose it could be a change in `rmw_fastrtps_cpp` as well.