ros2/rclcpp #1876: no matching function for rclcpp::create_publisher

alsora, 2022-01-23:
## Bug report

Build errors when calling `rclcpp::create_publisher` are not easy to solve looking at the `create_publisher.hpp` file.
Either we should add documentation there, or we should make the functions more flexible in the accepted input arguments.

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - Rolling
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
class MyNode : public rclcpp::Node
{
public:
    MyNode() : rclcpp::Node("my_node")
    {
        auto pub = rclcpp::create_publisher<std_msgs::msg::String>(
            this->get_node_topics_interface(),
            "my_topic",
             rclcpp::SensorDataQoS());
    }
};
```

#### Actual behavior

The above code snippet fails to compile due to no matching function found in `rclcpp::create_publisher`.
Looking at https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/create_publisher.hpp didn't really help to figure out the solution to the problem.

Tried also to add a `this->get_node_parameters_interface()` argument to match the signatures there, but got again errors 
```
        auto pub = rclcpp::create_publisher<std_msgs::msg::String>(
            this->get_node_parameters_interface(),
            this->get_node_topics_interface(),
            "my_topic",
             rclcpp::SensorDataQoS());
```

For what is worth, the following syntax works

```
       auto node_topics_interface = this->get_node_topics_interface();
       auto pub = rclcpp::create_publisher<std_msgs::msg::String>(
            this->get_node_parameters_interface(),
            node_topics_interface,
            "my_topic",
             rclcpp::SensorDataQoS());
```

fujitatomoya, 2022-02-02:
The compile error is

```
/root/ros2_ws/colcon_ws/install/rclcpp/include/rclcpp/create_publisher.hpp:94:1: note: candidate: ‘std::shared_ptr<PublisherT> rclcpp::create_publisher(NodeT&, const string&, const rclcpp::QoS&, const rclcpp::PublisherOptionsWithAllocator<AllocatorT>&) [with MessageT = std_msgs::msg::String_<std::allocator<void> >; AllocatorT = std::allocator<void>; PublisherT = rclcpp::Publisher<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> >; NodeT = std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>; std::string = std::__cxx11::basic_string<char>]’ <near match>
   94 | create_publisher(
      | ^~~~~~~~~~~~~~~~
/root/ros2_ws/colcon_ws/install/rclcpp/include/rclcpp/create_publisher.hpp:94:1: note:   conversion of argument 1 would be ill-formed:
/root/ros2_ws/colcon_ws/src/ros2/ros2_test_prover/prover_rclcpp/src/rclcpp_1876.cpp:21:44: error: cannot bind non-const lvalue reference of type ‘std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>&’ to an rvalue of type ‘rclcpp::node_interfaces::NodeTopicsInterface::SharedPtr’ {aka ‘std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>’}
   21 |             this->get_node_topics_interface(),
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
```

this explains well and can be expected. but what about using universal reference to support rvalue?

alsora, 2022-02-02:
Unfortunately, that exact same problem causes this compilation error in my real build, probably because the publisher is created inside a template class constructor.

```
In file included from ../../.conan/data/ros2-core/galactic/stable/package/d4b840d561753c2a8e60658de7e3c3c5b0743393/include/rclcpp/create_publisher.hpp:22,
                 from ../../.conan/data/ros2-core/galactic/stable/package/d4b840d561753c2a8e60658de7e3c3c5b0743393/include/rclcpp/node_impl.hpp:41,
                 from ../../.conan/data/ros2-core/galactic/stable/package/d4b840d561753c2a8e60658de7e3c3c5b0743393/include/rclcpp/node.hpp:1315,
                 from ../../.conan/data/ros2-core/galactic/stable/package/d4b840d561753c2a8e60658de7e3c3c5b0743393/include/rclcpp/executors/events_executor.hpp:30,
                 from ../../.conan/data/ros2-core/galactic/stable/package/d4b840d561753c2a8e60658de7e3c3c5b0743393/include/rclcpp/executors.hpp:21,
                 from ../../.conan/data/ros2-core/galactic/stable/package/d4b840d561753c2a8e60658de7e3c3c5b0743393/include/rclcpp/rclcpp.hpp:156,
                 from ../../my_objects.hpp:15,
                 from ../../my_node.cpp:10:
../../.conan/data/ros2-core/galactic/stable/package/d4b840d561753c2a8e60658de7e3c3c5b0743393/include/rclcpp/node_interfaces/get_node_topics_interface.hpp: In instantiation of ‘std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface> rclcpp::node_interfaces::get_node_topics_interface(NodeType&&) [with NodeType = const std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>&; typename std::enable_if<rcpputils::is_pointer<T>::value, int>::type <anonymous> = 0]’:
../../.conan/data/ros2-core/galactic/stable/package/d4b840d561753c2a8e60658de7e3c3c5b0743393/include/rclcpp/create_publisher.hpp:56:82:   required from ‘std::shared_ptr<PublisherT> rclcpp::detail::create_publisher(NodeParametersT&, NodeTopicsT&, const string&, const rclcpp::QoS&, const rclcpp::PublisherOptionsWithAllocator<AllocatorT>&) [with MessageT = std_msgs::msg::String_<std::allocator<void> >; AllocatorT = std::allocator<void>; PublisherT = rclcpp::Publisher<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> >; NodeParametersT = const std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>; NodeTopicsT = const std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>; std::__cxx11::string = std::__cxx11::basic_string<char>]’
../../.conan/data/ros2-core/galactic/stable/package/d4b840d561753c2a8e60658de7e3c3c5b0743393/include/rclcpp/create_publisher.hpp:103:68:   required from ‘std::shared_ptr<PublisherT> rclcpp::create_publisher(NodeT&, const string&, const rclcpp::QoS&, const rclcpp::PublisherOptionsWithAllocator<AllocatorT>&) [with MessageT = std_msgs::msg::String_<std::allocator<void> >; AllocatorT = std::allocator<void>; PublisherT = rclcpp::Publisher<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> >; NodeT = const std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>; std::__cxx11::string = std::__cxx11::basic_string<char>]’
../../my_objects.hpp:101:60:   required from ‘BaseObject<Msg>::BaseObject(const NodeInterfaces&, const string&, const string&, const rclcpp::QoS&, BaseObject<Msg>::republish_func_t) [with Msg = std_msgs::msg::String_<std::allocator<void> >; std::__cxx11::string = std::__cxx11::basic_string<char>; BaseObject<Msg>::republish_func_t = std::function<std::unique_ptr<std_msgs::msg::String_<std::allocator<void> >, std::default_delete<std_msgs::msg::String_<std::allocator<void> > > >(std::shared_ptr<const std_msgs::msg::String_<std::allocator<void> > >)>; typename Msg::UniquePtr = std::unique_ptr<std_msgs::msg::String_<std::allocator<void> >, std::default_delete<std_msgs::msg::String_<std::allocator<void> > > >; typename Msg::ConstSharedPtr = std::shared_ptr<const std_msgs::msg::String_<std::allocator<void> > >]’
/usr/include/c++/8/ext/new_allocator.h:136:4:   required from ‘void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = BaseObject<std_msgs::msg::String_<std::allocator<void> > >; _Args = {NodeInterfaces&, const char* const&, const char* const&, rclcpp::SensorDataQoS, std::_Bind<std::unique_ptr<std_msgs::msg::String_<std::allocator<void> >, std::default_delete<std_msgs::msg::String_<std::allocator<void> > > > (MyObject::*(MyObject*, std::_Placeholder<1>))(std::shared_ptr<const std_msgs::msg::String_<std::allocator<void> > >)>}; _Tp = BaseObject<std_msgs::msg::String_<std::allocator<void> > >]’
/usr/include/c++/8/bits/alloc_traits.h:475:4:   [ skipping 2 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/c++/8/bits/shared_ptr_base.h:677:16:   required from ‘std::__shared_count<_Lp>::__shared_count(_Tp*&, std::_Sp_alloc_shared_tag<_Alloc>, _Args&& ...) [with _Tp = BaseObject<std_msgs::msg::String_<std::allocator<void> > >; _Alloc = std::allocator<BaseObject<std_msgs::msg::String_<std::allocator<void> > > >; _Args = {NodeInterfaces&, const char* const&, const char* const&, rclcpp::SensorDataQoS, std::_Bind<std::unique_ptr<std_msgs::msg::String_<std::allocator<void> >, std::default_delete<std_msgs::msg::String_<std::allocator<void> > > > (MyObject::*(MyObject*, std::_Placeholder<1>))(std::shared_ptr<const std_msgs::msg::String_<std::allocator<void> > >)>}; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2]’
[... cut ...]
../../.conan/data/ros2-core/galactic/stable/package/d4b840d561753c2a8e60658de7e3c3c5b0743393/include/rclcpp/node_interfaces/get_node_topics_interface.hpp:53:1: note:   template argument deduction/substitution failed:
../../.conan/data/ros2-core/galactic/stable/package/d4b840d561753c2a8e60658de7e3c3c5b0743393/include/rclcpp/node_interfaces/get_node_topics_interface.hpp:50:26: error: no type named ‘type’ in ‘struct std::enable_if<false, int>’
   >::value, int>::type = 0
```

Yes, universal references should work here.
Is there a reason why is not already being used?

fujitatomoya, 2022-02-02:
> Is there a reason why is not already being used?

i am not sure though...sorry.