ros2/rclcpp #1882: Change default branch name to `main`

SteveMacenski, 2022-02-01:
We've had some extended discussions on Discourse about migrating all of the repos over to `main` from `master` as an already in the works change in terminology in the software engineering field. I think rclcpp should be the first of the core stack to start this transition since its the most user-facing layer. 

Are there any particular challenges with this? I imagine updating `repos` files and CI systems (@clalancette)

clalancette, 2022-02-04:
Given how central this repository is, we'll want to do some research on what things need to change.  At the least I think we need to consider:

1. Making a new branch on this repository
2. Modifying https://github.com/ros2/ros2/blob/master/ros2.repos
3. Modifying https://github.com/ros2-gbp/rclcpp-release/blob/master/tracks.yaml
4. Checking out the scripts at https://github.com/ros2/ci (these generally rely on ros2.repos, but we should double-check)
5. Checking out the configuration/scripts at https://github.com/ros2/ros_buildfarm_config
6. Checking out the tutorials on https://github.com/ros2/ros2_documentation
7. Checking out the tooling at https://github.com/ros-tooling/setup-ros and https://github.com/ros-tooling/action-ros-ci (maybe others in the ros-tooling org)
8. Checking out the "official" docker build process (I have no idea how this is done, so we'll need others to chime in here)
9. Making an announcement on Discourse to let downstream users know

That will give us a better idea of what needs to change.

alsora, 2022-02-04:
I don't know about how to do it in github, but could we keep both the main and master branch around, with the master being a mirror of main?
development will happen on main, but for a while, if someone has a link pointing to master, it will still work

SteveMacenski, 2022-02-04:
I think unlike other problems, keeping around `master` in any form doesn't really follow the spirit why it is we're trying to remove it

alsora, 2022-02-04:
I agree with you.
However a mirror would probably allow to at least change the default faster than in the other case.

SteveMacenski, 2022-02-04:
As a more practical matter, I think it would make sense to do a bulk operation in a month or two once we have a rolling sync on the new Ubuntu OS so we're not fighting issues with source locations at the same time as bootstrapping a new OS. 

Then if we bulk change all of the ROS 2 repos to `main`, we can do bulk operations on all of the items that Chris mentioned above and try to do a dry run of everything. Surely we'll miss something but it should be easy to track down if we're not doing 1-by-1 wack-a-mole.

christophebedard, 2022-02-04:
I think it would be helpful to have a full, comprehensive checklist for non-core maintainers (or, rather, maintainers of core & non-core packages that are not maintained by OR). It could probably go under https://github.com/ros2/ros2.

clalancette, 2022-06-28:
So we ended up making a 'rolling' branch on all repositories, including this one: https://discourse.ros.org/t/change-default-branch-name-to-rolling-in-ros-2-core/26009 .

Due to backwards compatibility concerns, we decided to keep the master branch around, with a bot that automatically synchronizes between rolling and master.  There is likely a long tail of CI/actions that depend on the old branch names, so we don't want to remove it and cause massive downstream breakage.  We can eventually consider removing that.

So I consider this done, and I'm going to close this out.  Feel free to reopen (or open a new issue) if you think otherwise.

closed 2022-06-28 by clalancette