ros2/rclcpp #1884: LifecycleNode::on_deactivate activate the publisher

vincentrou, 2022-02-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - rolling

#### Steps to reproduce issue
```
LNI::CallbackReturn PackageName::on_deactivate(const rclcpp_lifecycle::State& state)
{
  publisher_->on_deactivate();
  auto pub_activated_before = publisher_->is_activated();
  LifecycleNode::on_deactivate(state);
  auto pub_activated_after = publisher_->is_activated();
  return LNI::CallbackReturn::SUCCESS;
}
```
#### Expected behavior

In this snippet, `pub_activated_before` is false and `pub_activated_after` is true.

#### Actual behavior

In this snippet, `pub_activated_before` is false and `pub_activated_after` is false.

#### Additional information
It seems that `on_activate` and `on_deactivate` were inverted here
https://github.com/ros2/rclcpp/blob/3123f5a64376588d07800579cb36cbdc7ffe82ee/rclcpp_lifecycle/src/lifecycle_node_interface_impl.hpp#L532

fujitatomoya, 2022-02-05:
issue confirmed. this only happens if `ManagedEntityInterface` is controlled by `LifecycleNode`. i think we should add test cases to make sure that as well.

fujitatomoya, 2022-02-05:
@vincentrou https://github.com/ros2/rclcpp/pull/1885 should be able to fix the problem.

closed 2022-02-07 by fujitatomoya