ros2/rclcpp #1902: BUG [gazebo.rclcpp]: Error in destruction of rcl subscription handle: failed to remove subscriber

SpaceView, 2022-03-22:
## Bug report

**Required Info:**

- Operating System: Windows 10 + ROS2
- Installation type:   binary 

#### Steps to reproduce issue
use command window  "x64 Native Tools Command Prompt for VS 2019" 
    ref. https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html
    ref. https://ms-iot.github.io/ROSOnWindows/GettingStarted/UsingROSonWindows.html
Following the following link, just the project address is D:\rosprj\nav2_ws instead of C:\nav2_ws
    ref. https://ms-iot.github.io/ROSOnWindows/ros2/nav2.html

#### Expected behavior

#### Actual behavior
D:\rosprj\nav2_ws>ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
[INFO] [launch]: All log files can be found below C:\Users\tansh\.ros\log\2022-03-22-20-29-57-447560-LAPTOP-LHDHHCLV-15724
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_waffle.urdf
[INFO] [gzserver-1]: process started with pid [1836]
[INFO] [gzclient   -2]: process started with pid [3024]
[INFO] [robot_state_publisher.EXE-3]: process started with pid [18164]
[robot_state_publisher.EXE-3] [WARN] [1647952200.790917800] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[robot_state_publisher.EXE-3] Parsing robot urdf xml string.
[robot_state_publisher.EXE-3] Link base_link had 7 children
[robot_state_publisher.EXE-3] Link camera_link had 2 children
[robot_state_publisher.EXE-3] Link camera_depth_frame had 1 children
[robot_state_publisher.EXE-3] Link camera_depth_optical_frame had 0 children
[robot_state_publisher.EXE-3] Link camera_rgb_frame had 1 children
[robot_state_publisher.EXE-3] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher.EXE-3] Link caster_back_left_link had 0 children
[robot_state_publisher.EXE-3] Link caster_back_right_link had 0 children
[robot_state_publisher.EXE-3] Link imu_link had 0 children
[robot_state_publisher.EXE-3] Link base_scan had 0 children
[robot_state_publisher.EXE-3] Link wheel_left_link had 0 children
[robot_state_publisher.EXE-3] Link wheel_right_link had 0 children
[robot_state_publisher.EXE-3] [INFO] [1647952200.839995900] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher.EXE-3] [INFO] [1647952200.840049200] [robot_state_publisher]: got segment base_link
[robot_state_publisher.EXE-3] [INFO] [1647952200.840060000] [robot_state_publisher]: got segment base_scan
[robot_state_publisher.EXE-3] [INFO] [1647952200.840068900] [robot_state_publisher]: got segment camera_depth_frame
[robot_state_publisher.EXE-3] [INFO] [1647952200.840081700] [robot_state_publisher]: got segment camera_depth_optical_frame
[robot_state_publisher.EXE-3] [INFO] [1647952200.840091700] [robot_state_publisher]: got segment camera_link
[robot_state_publisher.EXE-3] [INFO] [1647952200.840100800] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher.EXE-3] [INFO] [1647952200.840109000] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher.EXE-3] [INFO] [1647952200.840117800] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher.EXE-3] [INFO] [1647952200.840126400] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher.EXE-3] [INFO] [1647952200.840134800] [robot_state_publisher]: got segment imu_link
[robot_state_publisher.EXE-3] [INFO] [1647952200.840144400] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher.EXE-3] [INFO] [1647952200.840153500] [robot_state_publisher]: got segment wheel_right_link
[gzserver-1] 2022-03-22 20:30:01.701 [PARTICIPANT Error] Type with the same name already exists:rosgraph_msgs::msg::dds_::Clock_ -> Function eprosima::fastrtps::ParticipantImpl::registerType
[gzserver-1] [foonathan::memory] Allocator foonathan::memory::new_allocator (at 0000000000000000) leaked 4544 bytes.
[gzserver-1] [foonathan::memory] Allocator foonathan::memory::heap_allocator (at 0000000000000000) leaked 29304 bytes.
[gzserver-1] [ERROR] [1647952202.512314700] [gazebo.rclcpp]: Error in destruction of rcl subscription handle: failed to remove subscriber, at C:\a\1\s\ws\src\rmw_fastrtps\rmw_fastrtps_shared_cpp\src\subscription.cpp:53, at C:\a\1\s\ws\src\rcl\rcl\src\rcl\subscription.c:243
[gzserver-1] [ERROR] [1647952202.512351600] [gazebo.rclcpp]: Error in destruction of rcl subscription handle: failed to remove subscriber, at C:\a\1\s\ws\src\rmw_fastrtps\rmw_fastrtps_shared_cpp\src\subscription.cpp:53, at C:\a\1\s\ws\src\rcl\rcl\src\rcl\subscription.c:243
[ERROR] [gzserver-1]: process has died [pid 1836, exit code 3489660927, cmd 'gzserver D:\rosprj\nav2_ws\install\turtlebot3_gazebo\share\turtlebot3_gazebo\worlds\turtlebot3_worlds/waffle.model                                                                      -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].

#### Additional Info
This is my first time try ROS2 on windows,

CursedRock17, 2023-10-09:
I don't know how still this is but looking at the first tutorial and the Nav2 tutorial they stated, this person should be using ROS 2 Foxy.

Also looking at the error message
> at C:\a\1\s\ws\src\rmw_fastrtps\rmw_fastrtps_shared_cpp\src\subscription.cpp:53,

could this be a problem for `rmw_fastrtps`? I checked in that repository, it hasn't been reported there yet.

clalancette, 2023-10-09:
Well, given that it has been a year with no response, I'm going to go ahead and close this one out.  If there is still a problem here, we can open a new issue and debug it there.

closed 2023-10-09 by clalancette