ros2/rclcpp #1905: yaml value converted to string parameter

christianrauch, 2022-03-26:
## Bug report

A yaml structure that is set as parameter value is stored as string.

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: binaries
- Version or commit hash: galactic
- Client library (if applicable): rclcpp

#### Steps to reproduce issue

1. declare a parameter `this->declare_parameter("bla");`, e.g. in the `set_and_get_parameters` demo example
2. run node: `ros2 run demo_nodes_cpp set_and_get_parameters`
3. create a configuration file `cfg.yaml`:
```yaml
/set_and_get_parameters:
  ros__parameters:
    bla:
    - x: 23
      y: 45
    - x: 55
      y: 123
```
4. `ros2 param load set_and_get_parameters cfg.yaml`
5. verify value: `ros2 param get set_and_get_parameters bla`

#### Expected behavior
The yaml structure should be properly represented as a list of dicts. The same goes for "arrays of arrays":
```yaml
bla:
- [23,45]
- [55,123]
```
which are also represented as a string.

#### Actual behavior
The value is interpreted as ordinary string: `String value is`

fujitatomoya, 2023-05-02:
> The yaml structure should be properly represented as a list of dicts. The same goes for "arrays of arrays":

is this supported? i think it takes `x` as string for `bla` parameter with it.

https://github.com/ros2/rclcpp/blob/b812790ee301d9aa536c79b9636736471701d1c3/rclcpp/include/rclcpp/parameter_value.hpp#L32-L44

> this->declare_parameter("bla");

Galactic is E.O.L, and this method is not supported on humble or later.

the following works,

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run prover_rclcpp rclcpp_1905 
[INFO] [1683068545.939686682] [testnode]: /testnode created...
[INFO] [1683068545.939748150] [testnode]: declared parameter named bla...
...

root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param load testnode test_params.yaml 
Set parameter bla.x successful
Set parameter bla.y successful
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get testnode bla.x
Integer values are: [23, 45]
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get testnode bla.y
Integer values are: [55, 123]
root@tomoyafujita:~/ros2_ws/colcon_ws# cat test_params.yaml
/testnode:
  ros__parameters:
    bla:
      x:
      - 23
      - 45
      y:
      - 55
      - 123
```

i might be mistaken, please correct me if i got anything wrong...

fujitatomoya, 2024-06-07:
this has been more than 2 years since last update, i will go ahead to close this. please feel free to reopen.

closed 2024-06-07 by fujitatomoya