ros2/rclcpp #1906: broken InvalidParameterValueException message

christianrauch, 2022-03-26:
## Bug report

When setting a parameter in the command line, which is outside of the limits, I get an error message that is not correctly formatted:
```
terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterValueException'
  what():  parameter 'ExposureTime' could not be set: Parameter {} doesn't comply with integer range.
```

I assume that `{}` is supposed to be something else, e.g. contain the value.

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: binary
- Version or commit hash: galactic
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
1. declare a parameter with bounds: `declare_parameter("ExposureTime", default_value, param_descr);` where `param_descr` is a `ParameterDescriptor` with integer bounds, e.g. (400 ... 20000).
2. initialise the parameter outside of the bounds, e.g. `--ros-args -p ExposureTime:=300`

#### Expected behavior
The exception should contain a useful error message.

#### Actual behavior
`Parameter {} doesn't comply with integer range.` does not make sense to me.

aprotyas, 2022-03-26:
It looks like what's happening is that the provided parameter value is first checked against the parameter descriptor, and only after a successful check is the rest of the descriptor populated with the name, type, and value information.

https://github.com/ros2/rclcpp/blob/d302e3c6280e7d6abf7a7cde8bca6498d2be7ddc/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp#L302-L320

So, when the error message is created, the name field in the parameter descriptor is empty.

I think the solution is to partially populate the parameter descriptor, at least with the parameter name, before the checks are made.

fujitatomoya, 2022-03-28:
> It looks like what's happening is that the provided parameter value is first checked against the parameter descriptor, and only after a successful check is the ParameterInfo structure populated with the name, type, and value information.

i think that this is a correct behavior, `ParameterInfo` is output for actual parameter information that can be accepted. (internally parameter value prevails to parameter descriptor.)

`Parameter {} doesn't comply with integer range.`, this happens only if parameter descriptor does not have effective name field by user application. (i think user application can store the different name compared to `name` argument for `declare_parameter`, but it will be overwritten after check process.) and i think this makes sense that it prints original descriptor name field with range check.

how about adjusting the message a bit internally? this can avoid the confusion?

```patch
diff --git a/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp b/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp
index 92d42bd8..3e7acea5 100644
--- a/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp
+++ b/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp
@@ -140,7 +140,11 @@ std::string
 format_range_reason(const std::string & name, const char * range_type)
 {
   std::ostringstream ss;
-  ss << "Parameter {" << name << "} doesn't comply with " << range_type << " range.";
+  if (!name.empty()) {
+    ss << "Parameter {" << name << "} doesn't comply with " << range_type << " range.";
+  } else {
+    ss << "Parameter doesn't comply with " << range_type << " range.";
+  }
   return ss.str();
 }

```

christianrauch, 2022-03-28:
> ```diff
> +    ss << "Parameter doesn't comply with " << range_type << " range.";
> ```

The overridden parameter name is known at runtime before the check. Why would we want to print a generic message that does not identify the parameter? It would be much more helpful if the name would be populated with the overridden parameter name that actually violated the ranges.

fujitatomoya, 2022-03-28:
> It would be much more helpful if the name would be populated with the overridden parameter name that actually violated the ranges.

```
terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterValueException'
  what():  parameter 'ExposureTime' could not be set: Parameter {} doesn't comply with integer range.
```

i think we can already see the actual parameter name 'ExposureTime' above?

christianrauch, 2022-03-28:
> i think we can already see the actual parameter name 'ExposureTime' above?

In this case yes. But this will also remove the parameter name when you set the parameter later (outside the initial overrides). In this case, you will get something like
```
$ ros2 param set /camera ExposureTime 300
Setting parameter failed: Parameter doesn't comply with integer range.
```
which is not helpful at all when you are setting multiple parameters.

aprotyas, 2022-03-28:
Thanks for the context about the ParameterInfo data structure @fujitatomoya. Without further information, I think resolving this is difficult because it's unclear to me whether an internal API modification is justified. Furthermore, I suspect the name field being printed is empty because the `NodeParameter` class is depending on the user fully populating the parameter descriptor (including the name field) correctly, and hence it does no additional checks along that line (IMO, it should stay that way).

Having said that, in `__check_parameters`, there are function-scoped parameter descriptor structures generated by searching the `ParameterInfo` map provided. In #1906, I've just populated this temporary structure with the parameter name information. This way, the change does not show in other data structures, but the name field is also not empty when used to generate error messages in the checker functions.

fujitatomoya, 2022-03-29:
@christianrauch thanks for your comment, i might miss something...

```
$ ros2 param set /camera ExposureTime 300
Setting parameter failed: Parameter doesn't comply with integer range.
```

it is really clear for me that `ExposureTime 300` does not comply with the range.

> which is not helpful at all when you are setting multiple parameters.

can you elaborate this, example would be appreciated.

christianrauch, 2022-03-29:
> can you elaborate this, example would be appreciated.

If you set 10 parameters and receive a single `Parameter doesn't comply with integer range.` how do you know which parameter is invalid?

Just ask yourself the following two questions:

1. What is the benefit of adding the parameter name explicitly to the error message?
2. What is the advantage of removing the parameter name and showing a generic error message?

Is this just about not willing to put the work into fixing a bug and living with a bad solution?

fujitatomoya, 2022-03-29:
> If you set 10 parameters and receive a single Parameter doesn't comply with integer range. how do you know which parameter is invalid?

i am not really clear how we can actually make this happen, so i did ask the example.

> Is this just about not willing to put the work into fixing a bug and living with a bad solution?

this comment is not constructive. i did not say that at all. just asking the information and questions to make sure. actually positive on this, see my comment here, https://github.com/ros2/rclcpp/pull/1908#discussion_r836948024

christianrauch, 2022-03-29:
> i am not really clear how we can actually make this happen, so i did ask the example.

You can define multiple parameter overrides on the command line, e.g. `--ros-args -p <param1>:=<value1> -p <param2>:=<value2>`. In such a case, you only get one message and if this does not contain either `param1` or `param2`, then you will not know which one has values outside the min/max limits.

> just asking the information and questions to make sure. actually positive on this

Sorry, I did not try to be rude. For me, this is clearly a bug. But now I see that the effect (faulty error message), not the cause (missing name in parameter info), is discussed to be "fixed". IMHO, this is wrong and not acceptable.

fujitatomoya, 2022-03-29:
> You can define multiple parameter overrides on the command line, e.g. --ros-args -p <param1>:=<value1> -p <param2>:=<value2>. In such a case, you only get one message and if this does not contain either param1 or param2, then you will not know which one has values outside the min/max limits.

thanks for the example.
in this case, there will be always one error message that hits the outside of the range in descriptor with parameter name. I confirmed the following with current mainline https://github.com/ros2/ros2/commit/aa0ace58f9905250bd79327b6e9729fb7df09416

```
        rcl_interfaces::msg::ParameterDescriptor range_descriptor1;
        range_descriptor1.name = "bar";
        range_descriptor1.type = rcl_interfaces::msg::ParameterType::PARAMETER_INTEGER;
        rcl_interfaces::msg::IntegerRange int_range1;
        int_range1.from_value = 0;
        int_range1.to_value = 10;
        int_range1.step = 1;
        range_descriptor1.integer_range.push_back(int_range1);

        rcl_interfaces::msg::ParameterDescriptor range_descriptor2;
        range_descriptor2.name = "ho";
        range_descriptor2.type = rcl_interfaces::msg::ParameterType::PARAMETER_INTEGER;
        rcl_interfaces::msg::IntegerRange int_range2;
        int_range2.from_value = 10;
        int_range2.to_value = 20;
        int_range2.step = 1;
        range_descriptor2.integer_range.push_back(int_range2);

        node->declare_parameter("foo", rclcpp::ParameterValue(5), range_descriptor1);
        node->declare_parameter("ge", rclcpp::ParameterValue(15), range_descriptor2);
        rclcpp::spin(node);
```

and we can get the following.

```bash
root@tomoyafujita:~/ros2_ws/master_ws# ros2 run rclcpp_1906 test_node --ros-args -p foo:=20 -p ge:=30
terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterValueException'
  what():  parameter 'foo' could not be set: Parameter {bar} doesn't comply with integer range.
[ros2run]: Aborted
root@tomoyafujita:~/ros2_ws/master_ws# ros2 run rclcpp_1906 test_node --ros-args -p foo:=10 -p ge:=30
terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterValueException'
  what():  parameter 'ge' could not be set: Parameter {ho} doesn't comply with integer range.
[ros2run]: Aborted
```

> But now I see that the effect (faulty error message), not the cause (missing name in parameter info), is discussed to be "fixed". IMHO, this is wrong and not acceptable.

so i was thinking that we can know which parameter is out of range, and we could just remove redundant parameter name and confusion if the descriptor name is empty with above easy patch. but that was just idea, probably https://github.com/ros2/rclcpp/pull/1908 makes more sense.

aprotyas, 2022-03-29:
> ```shell
>   what():  parameter 'foo' could not be set: Parameter {bar} doesn't comply with integer range.
> ```

Is this desired behavior? Or, should this behavior be allowed? Wouldn't this also cause some confusion? i.e. is the parameter name `foo` or is it `bar`? What if there's another named parameter `bar` and it's descriptor name field says `foo`? :smile: 

Regardless, I think we shouldn't override the user's data under the good faith assumption that they know what they're doing, so a simple patch like this in `__check_parameters` makes sense:

```cpp
if (descriptor.name.empty()) {
  descriptor.name = name;
}
```

fujitatomoya, 2022-03-29:
user application can know that name and type in the descriptor are ignored.

https://github.com/ros2/rclcpp/blob/eac006317692b7eff3aed671eadbff3d72cca098/rclcpp/include/rclcpp/node.hpp#L333-L335

but agree, the message would be confusing. it would be better that `Parameter Descriptor {bar} doesn't comply with integer range.`?

PR #1908 (2022-03-30): [NodeParameters] Set name in param info pre-check
In `__check_parameters`, there are function-scoped parameter
descriptor structures generated by searching the `ParameterInfo`
map provided from other declaratory functions. This PR populates
this temporary structure with the parameter name information.
This way, the change does not show in other data structures, but the
name field is also not empty when used to generate error messages in
`__check_parameter_value_in_range`.

See this [demo node](https://github.com/aprotyas/ros2_playground/tree/master/rclcpp_1906) for testing purposes.

Fixes #19 [... cut]
closed 2022-03-30 by fujitatomoya