ros2/rclcpp #1907: parameter_overrides() returns an empty vector

christianrauch, 2022-03-26:
## Bug report

The `parameter_overrides` method of `NodeOptions` returns an empty vector.

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: binary
- Version or commit hash: galactic
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
1. modify the `set_and_get_parameters` example to print the size of the `parameter_overrides` in the `SetAndGetParameters` constructor:
```cpp
std::cout << "overrides: " << options.parameter_overrides().size() << std::endl;
```
2. run the node with an override parameter: `ros2 run demo_nodes_cpp set_and_get_parameters --ros-args -p foo:=42`

#### Expected behavior

The parameter remapping (override) `foo:=42` should appear in the vector returned by the `parameter_overrides()`.

#### Actual behavior

`parameter_overrides()` returns an empty vector.

Treborsky, 2022-04-22:
Hi, can you give me a code example? I tried going through the source code, but found nothing that would indicate it's flawed.

Below is the function definition, if you want to take a look at it yourself.

https://github.com/ros2/rclcpp/blob/c24e485084f421292734ebf6b63f1a105a2c7602/rclcpp/src/rclcpp/node_options.cpp#L164-L174

Also this issue should be first submitted in demo_nodes_cpp repo

christianrauch, 2022-04-22:
> Hi, can you give me a code example?

Does adding `options.parameter_overrides().size()` to the `set_and_get_parameters` example not work for you?

> Also this issue should be first submitted in demo_nodes_cpp repo

Why? This is not an issue with the original demo. It's an issue with the API that doesn't do what it is supposed to do.

ivanpauno, 2022-04-22:
`NodeOptions` is used to configure the node with options created programatically, it doesn't allow you to read the options passed through the command line (that applies to `parameter_overrides`, or any of the other `NodeOptions` members).

After creating a `Node`, you can use `node->get_node_parameters_interface()->get_parameter_overrides()`.
See [1](https://github.com/ros2/rclcpp/blob/c24e485084f421292734ebf6b63f1a105a2c7602/rclcpp/include/rclcpp/node.hpp#L1171) and [2](https://github.com/ros2/rclcpp/blob/c24e485084f421292734ebf6b63f1a105a2c7602/rclcpp/include/rclcpp/node_interfaces/node_parameters.hpp#L185).

ivanpauno, 2022-04-22:
@christianrauch does that answer your question?

Treborsky, 2022-04-22:
> Does adding `options.parameter_overrides().size()` to the `set_and_get_parameters` example not work for you?
>
>
> Why? This is not an issue with the original demo. It's an issue with the API that [...]

I am no expert, I just had a quick look at the library code during a boring class, but I seems to me that the vector will be empty if you don't insert anything into it programatically. 

That would align with @ivanpauno 's comment. I'm sure he understands the topic more deeply.

christianrauch, 2022-04-23:
> I am no expert, I just had a quick look at the library code during a boring class, but it seems to me that the vector will be empty if you don't insert anything into it programatically.

Ok, but then the question would be why are the parameters not set? I expected that the `NodeOptions` contain the options that are provided to the node via command line, launch files, etc, This is why I opened this as a bug. Maybe some of the developers can chime in to explain if this is intentional and what the motivation behind this is.

Treborsky, 2022-04-25:
I found this in ros2 cli param api implementation: https://github.com/ros2/ros2cli/blob/c4b993d34aed110ef3c00a576927cb89b0ce87ec/ros2param/ros2param/api/__init__.py#L202-L218

I believe this is called when setting parameters from command line and it uses a node client, that executes a given request. In this particular case, it would call some sort of set params service on the node and that's how the params are set.

I found the exact monster of a function which sets the parameters, but it's 190 lines long, so I won't try and analyze it. You can find it under rclcpp/include/rclcpp/node_interfaces/node_parameters.cpp, line 592 set_parameters_atomically.

ivanpauno, 2022-04-25:
> Ok, but then the question would be why are the parameters not set? I expected that the NodeOptions contain the options that are provided to the node via command line, launch files, etc, This is why [...]

Yes, it's intentional.
As I commented above, `NodeOptions` is passed to a `Node` in order to programatically configure it.
The getters of `NodeOptions` only give access to what was previously set to that `NodeOptions` instance, and they're not meant to get the options passed through the command line.

If you want to get the parameter overrides that a `Node` used, including the ones passed through the command line, you can use `node->get_node_parameters_interface()->get_parameter_overrides()`.
See [1](https://github.com/ros2/rclcpp/blob/c24e485084f421292734ebf6b63f1a105a2c7602/rclcpp/include/rclcpp/node.hpp#L1171) and [2](https://github.com/ros2/rclcpp/blob/c24e485084f421292734ebf6b63f1a105a2c7602/rclcpp/include/rclcpp/node_interfaces/node_parameters.hpp#L185).

christianrauch, 2022-04-25:
> Yes, it's intentional.

What is the motivation behind this? I find this quite confusing that there are two ways to get the "parameter overrides". Why can there not only be one method that provides the gathered overrides (i.e. those from the command line and those from NodeOptions). Or why can the NodeOptions not just also contain the parameter overrides from the command line?

In any case, the API documentation should be updated with an explanation what the methods do exactly and why they behave like they behave currently.

ivanpauno, 2022-04-25:
> In any case, the API documentation should be updated with an explanation what the methods do exactly and why they behave like they behave currently.

The docs already mention the default values https://github.com/ros2/rclcpp/blob/c24e485084f421292734ebf6b63f1a105a2c7602/rclcpp/include/rclcpp/node_options.hpp#L43
If you have any other suggestion, please create a PR enhacing the docs.

> What is the motivation behind this?

Currently, you configure a node through:

- The `NodeOptions` instance passed to the `Node` constructor.
- The [arguments](https://github.com/ros2/rclcpp/blob/c24e485084f421292734ebf6b63f1a105a2c7602/rclcpp/include/rclcpp/node_options.hpp#L115) passed through `NodeOptions`
- The arguments passed to the [rclcpp::Context](https://github.com/ros2/rclcpp/blob/c24e485084f421292734ebf6b63f1a105a2c7602/rclcpp/include/rclcpp/context.hpp#L129-L130) passed [to the Node](https://github.com/ros2/rclcpp/blob/c24e485084f421292734ebf6b63f1a105a2c7602/rclcpp/include/rclcpp/node_options.hpp#L99) (if no context is specified, the default one is used which is initialized through `rclcpp::init()`).

You can get the resulting values with the getters provided by `Node`.

e.g. for parameter overrides, you can use `node->get_node_parameters_interface()->get_parameter_overrides()`.
See [1](https://github.com/ros2/rclcpp/blob/c24e485084f421292734ebf6b63f1a105a2c7602/rclcpp/include/rclcpp/node.hpp#L1171) and [2](https://github.com/ros2/rclcpp/blob/c24e485084f421292734ebf6b63f1a105a2c7602/rclcpp/include/rclcpp/node_interfaces/node_parameters.hpp#L185).

The fact that the API works like this is just the way it was developed, I would say.
e.g. something like `rclcpp::NodeOptions().parse_options_from_argc_argv(argc, argv).parameter_overrides()` would also be a reasonable API. But `NodeOptions` was thought just like a record of options that's passed to `Node` and does nothing, and `Node` does all the argument processing.

> Why can there not only be one method that provides the gathered overrides (i.e. those from the command line and those from NodeOptions)

There's one, see above in this comment.
The only caveat is that you can access them after a `Node` is created.
Ideally, we should've a way to access the parsed options from the command line before a node is created (this is actually possible, but a bit not nice to do currently. I can post a code snippet of how to do it if that's what you want).

> Or why can the NodeOptions not just also contain the parameter overrides from the command line?

It's just a record of options to be passed to node, using the named parameter idiom to make construction nicer.
I agree it would be nice to have a way of getting the equivalent `NodeOptions` parsed from the provided arguments.
Though that might create confusion, if the arguments are also passed to `Node`, so I'm not sure.