ros2/rclcpp #1917: GuardCondition should have a callback that gets handled by the executor

sloretz, 2022-04-07:
## Feature request

Discovered while looking at ros2/rclcpp#1916

#### Feature description

I think it would be useful to have a callback that gets run by the executor after the guard condition is triggered.
`rclpy` [already has this feature](https://github.com/ros2/rclpy/blob/c704619fc9666c5403a7d0373eb4795c94b721b1/rclpy/rclpy/guard_condition.py#L21).

This could be used to do work in the executor. One could use it to complete a future to work around ros2/rclcpp#1916.

You can sort of do this now with timers with a 0 second period by canceling them in the callback. The timer callback won't run again until it's reset, so resetting is sort of like calling `trigger()`).

#### Implementation considerations

The `GuardCondition` class has a callback that gets called when it's triggered, but this isn't suitable for this feature.
https://github.com/ros2/rclcpp/blob/85a7046ac39bfc0d787ed2bbe62574424e31b871/rclcpp/include/rclcpp/guard_condition.hpp#L112

This callback replaces the trigger implementation so that the `rcl_trigger_guard_condition` method never gets called, and the executor isn't woken. A different callback would be needed to implement this feature.

https://github.com/ros2/rclcpp/blob/85a7046ac39bfc0d787ed2bbe62574424e31b871/rclcpp/src/rclcpp/guard_condition.cpp#L79-L87

alsora, 2022-04-12:
I think that this is an error in the `guard_condition.cpp` code.

Calling `rcl_trigger_guard_condition` should be independent from whether there is an `on_trigger_callback_` or not.

wjwwood, 2022-05-12:
> This could be used to do work in the executor. One could use it to complete a future to work around https://github.com/ros2/rclcpp/issues/1916.

I'd rather see a `Executor::call_soon(<function or lambda>)` kind of function. (Ab)using guard conditions to accomplish this is no different really than doing the same with zero duration, one-off timers.

That being said, if users are to use a guard condition effectively, they will need to have an associated callback.