ros2/rclcpp #1924: async_send_request will not return if service node launches after client

carTloyal123, 2022-04-30:
- Operating System:
  - Windows 11, Docker with Ubuntu Focal, ROS2 Galactic Desktop
- Installation type:
  - Premade docker image, imagine from binary
- DDS implementation:
  - Default
- Client library (if applicable):
  - rclcpp

I seem to have stumbled on a bug but I am not entirely sure. I do not get a service response if my service server is launched after my client. Is this intended for some reason? Any help appreciated!
#### Steps to reproduce issue

Request being sent that never returns:
```
auto servo_config_req_ = std::make_shared<i2cpwm_ros2_msgs::srv::ServosConfig::Request>();
...
using ServiceResponseFuture = rclcpp::Client<i2cpwm_ros2_msgs::srv::ServosConfig>::SharedFuture;
auto response_received_callback = [this](ServiceResponseFuture future) {
    RCLCPP_INFO(rclcpp::get_logger("softsys_joy_node"), "Got config result: %d", future.get()->error);
};
auto servo_config_result = this->i2cpwm_srv_client_->async_send_request(servo_config_req_, response_received_callback);
```

I have searched for other similar answers but I am not sure anything addresses this specifically. If I have missed an issue, please feel free to guide me in that direction.

#### Expected behavior
- async_send_request will wait until a service is available and proceed to process the request then return the result.
#### Actual behavior
- async_send_request never returns if the service server is not available at time of calling the service.
#### Additional information
I am launching the two nodes with stock python launch files. Please let me know if I can provide further detail.

clalancette, 2022-05-02:
This is expected behavior.  If you really want to make sure that the server is there before making the call, you should call `wait_for_service` (like in https://docs.ros.org/en/galactic/Tutorials/Writing-A-Simple-Cpp-Service-And-Client.html?highlight=wait_for_service#write-the-client-node) before making the call.

Let me know if that answers your question.

carTloyal123, 2022-05-08:
That is exactly the solution! Thank you! Apologies for overlooking that section.

closed 2022-05-08 by carTloyal123