ros2/rclcpp #1925: Empty node creates asan AddressSanitizer: new-delete-type-mismatch abort on construction

fake-name, 2022-05-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04, Ubuntu 18.04 armhf (Tegra)
- Installation type:
  - Source
- Version or commit hash:
  - Install was via `rosinstall_generator`. Relevant revisions:
  - `rclcpp`: 2a5daafe0c12541904b72f19e4afa07c3052969d
  - `rcl`: fac8347437b3d8e1e38f8e1e6728ff03c638792a
  - I built this today, so the revisions should all be basically HEAD at 2022/05/03
- DDS implementation:
  - N/A
- Client library (if applicable):
  - `rclcpp`

#### Steps to reproduce issue

Compile the following node:

`main.cpp`
```

#include <rclcpp/rclcpp.hpp>

#include <iostream>

class TestNode : public rclcpp::Node
{
	private:

	public:

		TestNode()
		: Node("test_node")
		{

		}
		virtual ~TestNode()
		{

		}

};

int main(int argc, char * argv[])
{
	std::cout << "Main!" << std::endl;
	rclcpp::init(argc, argv);
	auto node = std::make_shared<TestNode>();

	std::cout << "Starting node. ROS2 OK: " << rclcpp::ok() << "." << std::endl;
	rclcpp::spin(node);

	rclcpp::shutdown();

	return 0;
}

```

`CMakeLists.txt`:
```
cmake_minimum_required(VERSION 3.8)
project(node_test)

# find dependencies
find_package(ament_cmake REQUIRED)
[... 103 lines cut ...]
    #25 0x7fd10749c44e in std::shared_ptr<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > > rclcpp::create_publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void>, rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface> >(std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&) (/opt/workspace/ros2/galactic/install/rclcpp/lib/librclcpp.so+0x6f944e)
    #26 0x7fd107493be0 in rclcpp::node_interfaces::NodeParameters::NodeParameters(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&, bool, bool, rclcpp::QoS const&, rclcpp::PublisherOptionsBase const&, bool, bool) (/opt/workspace/ros2/galactic/install/rclcpp/lib/librclcpp.so+0x6f0be0)
    #27 0x7fd107472216 in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) (/opt/workspace/ros2/galactic/install/rclcpp/lib/librclcpp.so+0x6cf216)
    #28 0x7fd10747158e in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) (/opt/workspace/ros2/galactic/install/rclcpp/lib/librclcpp.so+0x6ce58e)
    #29 0x56311caaf896 in TestNode::TestNode() (/home/durr/testws/build/node_test/test_node+0xa896)

SUMMARY: AddressSanitizer: new-delete-type-mismatch ../../../../src/libsanitizer/asan/asan_new_delete.cpp:172 in operator delete(void*, unsigned long)
==12240==HINT: if you don't care about these errors you may set ASAN_OPTIONS=new_delete_type_mismatch=0
==12240==ABORTING

```

The binary aborts during startup.

#### Expected behavior

Considering apparently asan is built as part of the standard CI process (https://github.com/ros2/ros2/issues/665), I'd expect to be able to instantiate about the minimum `rclcpp` node possible without abort being called. 

#### Actual behavior

`asan` triggers an abort. 

-----------------

I *think* this is harmless, but it makes using asan annoying, if nothing else.

fujitatomoya, 2022-05-11:
this can be observed on master https://github.com/ros2/ros2/commit/077c53b9a27100bcb4e3ec94549d5a9e6690fea5. but i am not sure why it deallocate the different type... any thoughts?

clalancette, 2022-05-16:
I wonder if this is related to https://github.com/ros2/rclcpp/pull/1324 .  That's been a long outstanding PR that we still haven't been able to get in.

fujitatomoya, 2022-05-17:
@clalancette thanks for the info, i will look into it.

tylerjw, 2022-06-06:
I see I created a duplicate issue for this here: https://github.com/ros2/rclcpp/issues/1948

My issue has a link to a minimal example that runs in a Github Actions CI job and demonstrates this bug. You will note that this failure does not exist in Galactic. I just discovered this after migrating a project from Galactic to Rolling that runs ASAN in CI.

fake-name, 2022-06-07:
> You will note that this failure does not exist in Galactic

I'm using galactic, and seeing the issue. I'm building from source, but specifying galactic to `rosinstall_generator`

tylerjw, 2022-06-07:
Maybe the older version of gcc on 20.04 just doesn't catch this error. What compiler are you using when building from source?

fake-name, 2022-06-07:
Gcc 10.3.