ros2/rclcpp #1935: Backport "Drop wrong template specialization (#1926)" to Humble?

tylerjw, 2022-05-19:
I'm sorry if this is the wrong place to ask this. Please let me know if I should have posted this somewhere else.

This fix has us building ros2 from the source right now and we would really appreciate it if it could make it into Humble.

As a follow-up to this, would it be possible to add a test to CI that builds ros2 using C++20. I understand why ros2 can't be upgraded to C++20 itself yet but it would be really appreciated if ROS wouldn't break when projects are using C++20 themselves.

clalancette, 2022-05-20:
> This fix has us building ros2 from the source right now and we would really appreciate it if it could make it into Humble.

I requested the bot to backport it into Humble.  I *think* it should be ABI safe, but we'll have to test that.  However, this will have to be for Patch Release 1 as we are too late for Humble release day.

> As a follow-up to this, would it be possible to add a test to CI that builds ros2 using C++20. I understand why ros2 can't be upgraded to C++20 itself yet but it would be really appreciated if ROS wouldn't break when projects are using C++20 themselves.

I don't think we should add an additional CI test at this time.  We're perfectly happy to review patches that fix C++ 20 issues (as was done in #1926), but C++ 20 isn't an officially supported combination.  So in that respect, it is the same as all other Tier-3 combinations where support is community/best-effort.

clalancette, 2022-06-09:
I'm actually going to close this, since we have the backport PR open.

closed 2022-06-09 by clalancette