ros2/rclcpp #1941: Lifecycle node trigger_transition() function caused transition publish failed

weibw-720, 2022-05-24:
## Question

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - foxy
- DDS implementation:
  - Fast-DDS
- Client library (if applicable):
  - rclcpp_lifecycle

#### Steps to reproduce issue

I just learned ros2 recently and am learning to use lifecycle node.
When I execute the ` trigger_transition(Transition::TRANSITION_DEACTIVATE); `in the timer callback function,I will get the following error report:
```
string capacity not greater than size
[ERROR] [1653388415.412117081] []: Unable to start transition 4 from current state deactivating: Could not publish transition: cannot publish data, at /tmp/binarydeb/ros-foxy-rmw-fastrtps-shared-cpp-1.3.0/src/rmw_publish.cpp:55, at /tmp/binarydeb/ros-foxy-rcl-1.1.13/src/rcl/publisher.c:319, at /tmp/binarydeb/ros-foxy-rcl-lifecycle-1.1.13/src/rcl_lifecycle.c:367

```
And my node state will always be deactivating, can no longer be changed
Here are some of my code:

```
rclcpp::TimerBase::SharedPtr m_timer = create_wall_timer(500ms, std::bind(&TestLifecycleNode::onTimerCb, this));
void TestLifecycleNode::onTimerCb() {
    auto nodes = this->get_node_graph_interface()->get_node_names();
    map<string, uint8_t>::iterator it = m_parents_state.begin();
    while (it != m_parents_state.end()) {
        string name =  it->first;
        uint8_t state = it->second;
        if (std::find(nodes.begin(), nodes.end(), name) == nodes.end()) {
            RCLCPP_INFO(get_logger(),
                        "Parent %s is not longer present, removing from parent nodes container",
                        name.c_str());
            it = m_parents_state.erase(it);
            if (state == State::PRIMARY_STATE_ACTIVE && m_parents_state.empty())
                trigger_transition(Transition::TRANSITION_DEACTIVATE);
        }
        else
            it++;
    }
}
```
But when I execute the ` trigger_transition(Transition::TRANSITION_DEACTIVATE); ` in the message callback function,It will be all right.Here are some of my code:
```
rclcpp::Subscription<NodeBinding>::SharedPtr m_binding_subscriber = 
                                create_subscription<NodeBinding>(
                                "test_binding",   rclcpp::QoS(1000).keep_all().transient_local().reliable(),
                                std::bind(&TestLifecycleNode::onNodeBindingCb, this, _1));
void TestLifecycleNode::onNodeBindingCb(const NodeBinding::SharedPtr msg) {
    if (msg->child_node == get_name() && m_parents_state.count(msg->parent_node) > 0) {
            uint8_t remover_state = m_parents_state[msg->parent_node];
            m_parents_state.erase(msg->parent_node);
            if(remover_state == State::PRIMARY_STATE_ACTIVE && m_parents_state.empty())
                trigger_transition(Transition::TRANSITION_DEACTIVATE);
    }
}

```

Above, I would like to ask what I did wrong? Or is this a bug?

I'm looking forward to someone's help. This problem has bothered me for a long time.Thanks

----
## Feature request

#### Feature description

#### Implementation considerations

alsora, 2022-05-24:
Yes, this is a known issue.
See this existing ticket for details https://github.com/ros2/rclcpp/issues/1880

weibw-720, 2022-05-24:
> Yes, this is a known issue.
> See this existing ticket for details #1880

Is this problem caused by dds or ros2? This problem will occur frequently, it will cause the life cyclenode to feel very bad to use.
How can I fix it? Is there any way I can avoid this problem? Does upgrading to ros2 humble fix it?

fujitatomoya, 2022-05-24:
> Is this problem caused by dds or ros2?

ROS 2 LifeCycle Node.

> Is there any way I can avoid this problem?

AFAIK, there is no work-around on this.

> Does upgrading to ros2 humble fix it?

No, mainline still has this problem.

weibw-720, 2022-05-25:
> > Is this problem caused by dds or ros2?
> 
> ROS 2 LifeCycle Node.
> 
> > Is there any way I can avoid this problem?
> 
> AFAIK, there is no work-around on this.
> 
> > Does upgrading to ros2 humble fix it?
> 
> No, mainline still has this problem.

So what can I do now to avoid this problem?Do you have a good way？

weibw-720, 2022-05-26:
@fujitatomoya @alsora 
I found that under the galactic version and the humble version, my program can run normally, only the foxy version fails.So I updated to the humble version now, which has avoided the problem, my bold guess is the problem of dds.

alsora, 2022-05-26:
I haven't used Humble yet, but I can say that I had this issue on Galactic (it's extremely rare, but it's still present)

fujitatomoya, 2022-05-26:
@weibw-720 good to hear! i am not sure how you got it work. i think technically there would be inconsistent primary state, so problem could happen in humble too, if i am not mistaken.

weibw-720, 2022-05-27:
@fujitatomoya @alsora 
Maybe my test case is too simple, and there are no recurring problems. I will continue with the complete test case and test it comprehensively, but I think if it can pass my test case, then it currently meets my usage scenario. , I can still accept it. At least this problem is bound to appear on the foxy version, and I haven't encountered it so far in the humble version.