ros2/rclcpp #1952: Support setting parameters atomically using ros2 service command line

deepanshubansal01, 2022-06-09:
## Feature request
It would be nice to support setting parameters atomically using `ros2 service call` command line.

#### Feature description
Currently, the `ros2 service call ` sets parameters in a non atomic fashion, meaning if one fails it wouldn't affect the other params being set. It would good to support an optional argument `--atomically` or something which could help in debugging and testing if params are required to be set in an atomic fashion using the command line. 

#### Implementation considerations
For setting the params atomically using `ros2 service call` we could leverage the already existing Node API service `rcl_interfaces::srv::SetParametersAtomically`.

alsora, 2022-06-10:
Hi @deepanshubansal01 I think that this issue would be better suited for the https://github.com/ros2/ros2cli repository.
That's where the code that implements `ros2 service call` or similar CLI commands live.

P.S. note that you can set parameters with `ros2 param set`

clalancette, 2022-06-10:
> Hi @deepanshubansal01 I think that this issue would be better suited for the https://github.com/ros2/ros2cli repository.
> That's where the code that implements `ros2 service call` or similar CLI commands live.

Agreed.  The underlying `rclcpp` and `rclpy` client libraries already has this functionality, we would just need to somehow expose it via the command-line.  I'm going to move this over to `ros2cli`.