ros2/rclcpp #1962: Issue in compliling a plugin inherited from gazebo_plugins/gazebo_ros_camera

Luckykantnayak, 2022-07-02:
## Bug report

**Required Info:**

- Operating System:Ubuntu 20.04
  - 
- Installation type: ROS2 Foxy
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):rclcpp
  - 
- packages required (if applicable):gazebo_plugins
#### Steps to reproduce issue

**light_sensor_plugin.hpp**
```
#ifndef GAZEBO_LIGHT_SENSOR__LIGHT_SENSOR_PLUGIN_HPP_
#define GAZEBO_LIGHT_SENSOR__LIGHT_SENSOR_PLUGIN_HPP_

#include "gazebo_light_sensor/visibility_control.h"
#include <gazebo_plugins/gazebo_ros_camera.hpp>
#include <string>
#include <rclcpp/rclcpp.hpp>
#include <sensor_msgs/msg/illuminance.hpp>
namespace gazebo
{

  class LightSensorPlugin : public gazebo_plugins::GazeboRosCamera
  {
    public:
      LightSensorPlugin();

      virtual ~LightSensorPlugin();
      void Load(gazebo::sensors::SensorPtr _parent, sdf::ElementPtr _sdf);
    protected:  void OnNewFrame(const unsigned char *_image,
        unsigned int _width, unsigned int _height,
        unsigned int _depth, const std::string &_format) override;
    std::shared_ptr<rclcpp::Node> ros_node_ = rclcpp::Node::make_shared("light_sensor_plugin");
    rclcpp::Publisher<sensor_msgs::msg::Illuminance>::SharedPtr sensor_pub_;

    double _fov;
    double _range;
  };

}  // namespace gazebo_light_sensor

#endif  // GAZEBO_LIGHT_SENSOR__LIGHT_SENSOR_PLUGIN_HPP_
```
**light_sensor_plugin.cpp**
```
#include "gazebo_light_sensor/light_sensor_plugin.hpp"
#include <gazebo/common/Plugin.hh>
#include <string>

#include <gazebo/sensors/Sensor.hh>
#include <gazebo/sensors/CameraSensor.hh>
#include <gazebo/sensors/SensorTypes.hh>
#include <sensor_msgs/msg/illuminance.hpp>

namespace gazebo
{
//GZ_REGISTER_SENSOR_PLUGIN(LightSensorPlugin)
LightSensorPlugin::LightSensorPlugin()
:_fov(6),_range(10)
{

}

LightSensorPlugin::~LightSensorPlugin()
{
  RCLCPP_DEBUG_STREAM(ros_node_->get_logger(), "Camera unloaded");
}
void LightSensorPlugin::Load(gazebo::sensors::SensorPtr _parent, sdf::ElementPtr _sdf) {
  gazebo_plugins::GazeboRosCamera::Load(_parent,_sdf);
  if(!rclcpp::ok())
  {
    RCLCPP_FATAL_STREAM(ros_node_->get_logger(),"A ROS node for Gazebo has not been initialized, unable to load plugin. "
        << "Load the Gazebo system plugin 'libgazebo_ros_api_plugin.so' in the gazebo_ros package)");
    return;
  }

  sensor_pub_ = ros_node_->create_publisher<sensor_msgs::msg::Illuminance>("lightSensor",1);

}
void LightSensorPlugin::OnNewFrame(const unsigned char *_image,
    unsigned int _width, unsigned int _height,
    unsigned int _depth, const std::string &_format) {

    sensor_msgs::msg::Illuminance msg;
    msg.header.stamp = rclcpp::Clock().now();
    msg.header.frame_id = "";
    //msg.header.seq = seq;

    int startingPix = _width * ( (int)(_height/2) - (int)( _fov/2)) - (int)(_fov/2);

    double illum = 0;
    for (int i=0; i<_fov ; ++i)
    {
      int index = startingPix + i*_width;
      for (int j=0; j<_fov ; ++j)
        illum += _image[index+j];
    }

    msg.illuminance = illum/(_fov*_fov);
    msg.variance = 0.0;

    sensor_pub_->publish(msg);
   }

}  // namespace gazebo_light_sensor
```

#### Error while building package
[image]

#### Additional information

----

CursedRock17, 2023-04-30:
Did [this](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1404) update solve the problem because jacobperron proposed a fix two weeks after both openings of the issue in both this project and ros-simulation/gazebo_ros_pkgs

clalancette, 2024-01-19:
Yeah, I'm going to close this out as likely fixed.  If it isn't, please feel free to reopen.

closed 2024-01-19 by clalancette